esp8266/Arduino #4519: Unable to connect to esp8266 when esp work in AP_STA mode and AP have hidden SSID.

ufm, 2018-03-15:
Continue for issue #3625
The same trouble in SDK version 2.2.1(cfd48f3).
Unable to connect to esp8266 when esp work in AP_STA mode and AP have hidden SSID.

sezeryalcin, 2020-02-12:
Is this still open for Arduino Core 2.6.3?
I just upgraded from 2.5.2 and cannot connect to hidden wifi!

devyte, 2020-10-14:
@Erriez given that you're currently sort of playing with hidden SSIDs, could I ask for your help in testing whether this can be reproduced? The case is in #3625.

Erriez, 2020-10-14:
@sezeryalcin Are you using the WiFi multi? Which version of this repository are you using?

@devyte Correct: I implemented hidden SSID's for Multi WiFi. I used the following improved WiFi scan test sketch:

```
#include <ESP8266WiFi.h>

// Connect to (hidden) STA
#define WIFI_SSID ""
#define WIFI_PASS ""

void runWiFiScan()
{
    String ssid;
    int32_t rssi;
    uint8_t encryptionType;
    uint8_t* bssid;
    int32_t channel;
    bool hidden;
    int scanResult;

    Serial.println(F("Starting WiFi scan..."));
    
    scanResult = WiFi.scanNetworks(/*async=*/false, /*hidden=*/true);

    Serial.printf(PSTR("%d networks found:\n"), scanResult);

    // Print unsorted scan results
    for (int8_t i = 0; i < scanResult; i++) {
        WiFi.getNetworkInfo(i, ssid, encryptionType, rssi, bssid, channel, hidden);

        Serial.printf(PSTR(" %d: [CH %02d] [%02X:%02X:%02X:%02X:%02X:%02X] %ddBm %c %c %s\n"),
                      i,
                      channel,
                      bssid[0], bssid[1], bssid[2],
                      bssid[3], bssid[4], bssid[5],
                      rssi,
                      (encryptionType == ENC_TYPE_NONE) ? ' ' : '*',
                      hidden ? 'H' : 'V',
                      ssid.c_str());
        delay(0);
    }
}

void connectWiFi()
{
    Serial.printf(PSTR("Connecting to %s"), WIFI_SSID);
    WiFi.begin(WIFI_SSID, WIFI_PASS);
  
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(F("."));
    }
    Serial.println("");
  
    Serial.println(F("WiFi connected. IP address: "));
    Serial.println(WiFi.localIP());
}

void setup() 
{
    Serial.begin(115200);
    Serial.println(F("\nESP8266 hidden WiFi test\n"));
    Serial.println(system_get_sdk_version());

    // Set in station mode
    WiFi.mode(WIFI_STA);

    // Run WiFi scan
    runWiFiScan();

    // Connect to (hidden) WiFi
    connectWiFi();
}

void loop() {
}
```

My output is:

```
21:55:53.608 -> ESP8266 hidden WiFi test
21:55:53.608 -> 
21:55:53.608 -> 2.2.1(cfd48f3)
21:55:53.608 -> Starting WiFi scan...
21:55:55.806 -> 7 networks found:
21:55:55.806 ->  0: [CH 01] [18:35:D1:09:6C:21] -86dBm * V XXXXXX
21:55:55.806 ->  1: [CH 04] [FA:8F:CA:98:58:20] -84dBm   H 
21:55:55.806 ->  2: [CH 06] [E0:28:6D:2F:4C:68] -69dBm * V XXXXXX
21:55:55.806 ->  3: [CH 01] [3A:35:D1:09:6C:21] -86dBm * V XXXXXX
21:55:55.806 ->  4: [CH 11] [CA:0E:14:34:0E:E0] -76dBm * V XXXXXX
21:55:55.806 ->  5: [CH 11] [54:BE:53:89:4E:8F] -92dBm * V XXXXXX
21:55:55.806 ->  6: [CH 11] [C8:0E:14:34:0E:E0] -74dBm * H 
21:55:55.839 -> Connecting to dtr..........
21:56:01.528 -> WiFi connected. IP address: 
21:56:01.528 -> 10.0.0.240
```

The H indicates a hidden WiFi network. Please note that the WiFi access point must be 2.4GHz and accept unknown MAC addresses.

Can you run this sketch and post the output here?

Would you like to create MR for this improved sketch?

devyte, 2020-10-14:
@Erriez I believe the point here would be to set up an esp in AP mode with a hidden ssid, then on a different esp try your above improved sketch.

Erriez, 2020-10-15:
@devyte I am not able to connect from my Android cellphone to an ESP8266 with SoftAP `hidden=true`. It works with `hidden=false`.

Example: `ESP8266 | WifiAccessPoint`, replace:
```c++
WiFi.softAP(ssid, password, /*hidden=*/ true);
```