esp8266/Arduino #4521: Feature Request:  Make _getServiceTxt(char *name, char *proto) public

chadham, 2018-03-15:
Need to be able to query serviceTextRecord attributes, method is currently private.

Thanks.

devyte, 2018-03-16:
Are you aware of pr #3107 ?

chadham, 2018-03-16:
Was not. Seems like it would be simpler to make the existing method public, no?

mblythe86, 2018-03-21:
I think you misunderstand what _getServiceTxt is doing.  That method is used by the responder to send any Txt records you've added with addServiceTxt to others on the network (i.e. when someone else does a query).  That's all you'd be able to see from the method...what you've already added.

Without #3107, Txt records on responses to your queries are simply discarded.

chadham, 2018-03-21:
Im looking for a way to get the txt records for a discovered service... (not for txt records I added to the service Im broadcasting)

mblythe86, 2018-06-20:
> Im looking for a way to get the txt records for a discovered service... (not for txt records I added to the service Im broadcasting)

Ok, then you need #3107.  Just making _getServiceTxt pulic won't get you what you want.

devyte, 2018-12-05:
Closing in view of #5442 with a full rewrite of mdns.

closed 2018-12-05 by devyte