esp8266/Arduino #4522: WiFi stop working on updating to > 2.3.0, but works fine on <= 2.3.0

eshvatskyi, 2018-03-15:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				NodeMCU 1.0 (ESP-12E Module) (Compatible)
- Core Version:			[2.4.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			qio
- Flash Size: 		        4MB
- lwip Variant:                  v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:              115200

### Problem Description

WiFi doesn't work at all.
If I run any simple sketch (scan, softAP or STA) on version > 2.3.0 WIFI doesn't work.
All that I get, in a log, probe request events.

But everything fine on 2.3.0.

No issues with compiling and upload or crystal frequency I don't have.

Please advise what it could be. 
I've tried everything, stuck with this for a week.

SDK:2.2.1(cfd48f3)/Core:2.4.1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)

devyte, 2018-03-16:
Did you try erasing flash before flasing your sketch built with 2.4.1?

eshvatskyi, 2018-03-16:
fully a few times :), also tried to switch on both sides a few times. Effect that same

ekt-, 2018-03-16:
Same here. ESP8266 LoLin NodeMCU v.3
Rolling back to 2.3.0, wifi works fine

bill-orange, 2018-03-16:
I am using WEMOS Mini Pros and Huzzah Feathers and I have not seen this problem. Have you tried selecting different lwip variants?

ekt-, 2018-03-17:
I've tried
v2 Lower Memory
v2 Higher Bandwidth
v1.4 Higher Bandwidth

with identical results

bill-orange, 2018-03-17:
@ekt-  @eshvatskyi  Unfortunately,I don't own any NODEMCUs.  The one I purchased a few months ago died in flaming glory when the serial chip failed!

I set the Aurduino IDE to NODEMCU and loaded the NTPClient example into WEMOS Mini Pro and it worked fine,  I am not sure what you have going on here.  Perhaps you should post your code along with a debug log.  Maybe you are using syntax in your WiFi setup that is not accepted in the new board manager but parsed through the compiler on the old board manager okay.

ekt-, 2018-03-17:
with the code below
- 2.3.0 it connects and a ip is obtained
- 2.41 never connects

`#include <ESP8266WiFi.h>

void setup()
{
  Serial.begin(115200);
  Serial.println();

  WiFi.begin(ssid, password);

  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.print("Connected, IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() 
{
}
`

ekt-, 2018-03-17:
That said, this Lolin boards are pretty garbage. For once, they mount a 40hz crystal, instead of a 26hz one. In 2.4.1 there is a flag which allows me to specify the crystal freq, but just to say this things are not that standard.
So it might be another quirk (like the crystal thing) which has been exacerbated in this latest version of the library.
So just to say that if it's only me with this boards, don't waste any more time.

Maybe it's time for me to switch and try some lua programming :)

eshvatskyi, 2018-03-19:
AT commands BIN works fine, SDK 2.2.1, 16.0 AT firmware.

eshvatskyi, 2018-03-20:
AT commands BIN works fine, SDK 2.2.1, 16.0 AT firmware.

> AT+GMR
> AT version:1.6.0.0(Feb  3 2018 12:00:06)
> SDK version:2.2.1(f42c330)
> compile time:Feb 12 2018 16:31:26
> Bin version(Wroom 02):1.6.1
> OK

AT firmware work really file, I could scan, connect and broadcast

any suggestions to try?

igrr, 2018-03-20:
Does scanning (WiFiScan example) still work for you with 2.4.1, or is it broken as well?

If you set up a SoftAP on the ESP, can you connect to it using a phone/computer?

eshvatskyi, 2018-03-20:
no example from 2.4.1 works.

bill-orange, 2018-03-20:
Looking for a pattern here. What hardware are you using. NODEMCU?

eshvatskyi, 2018-03-20:
yes NODEMCU - compatible.

WereCatf, 2018-03-20:
Interestingly, I have zero issues using my LoLin V3 Nodemcu-boards (with the CH240G USB-TTL-chip and ESP-12E module) with latest git, connects to WiFi just wine and the WiFiScan-example works as well.

devyte, 2018-03-20:
@eshvatskyi I also can't reproduce any connection issues with any of my Nodemcu boards on latest git.
What does "compatible" mean? is it custom hardware designed by yourself? a copy/clone bought somewhere shady? something else?

igrr, 2018-03-21:
Hmm, if none of the examples work (not even scan), then it looks like a PHY configuration issue. But that is pretty hard to debug without being able to reproduce this. Ordered a bunch of NodeMCU knock-offs on Taobao, will test when they arrive.

eshvatskyi, 2018-03-21:
I've tried to use platformio.
everything works on 1.5 release, but 1.6 release that includes 2.4.0 version won't :(

minida28, 2018-03-21:
@eshvatskyi Did you try DIO mode?
I think I have same hardware with yours (nodemcu v1.0), it wont even boot if flashed in QIO mode, only DIO mode works. I can even use its pin 9 and pin 10.
But I am on fairly recent Git and also newer than 2.4.1.

kreso975, 2018-03-23:
Hi, last night i updated to 2.4.1 and WiFi problem show up. I'm playing with ESP8266-01. But only problem with connecting was with router Linksys EA6900. With 2 other TP-Link routers everything worked normal. I did tried Router restart, different settings etc - did not work. When in debug mode, i notice message saying Check your static ip! 

 wifi_station_set_hostname( wifi_hostname );      
 WiFi.hostname( wifi_hostname );                     
 WiFi.mode( WIFI_STA );
 WiFi.config(_ip, _dns, _gw, _sn);                  
 WiFi.begin( wifi_ssid, wifi_password );
 MDNS.begin(wifi_hostname);

marcelstoer, 2018-04-14:
I can add little new information other than confirming that I'm likely seeing the same issue. While I can _connect_ to all WiFis in the vicinity at least one of them will eventually freeze the ESP8266 stack after a while.

The application is regularly [pulling in data from Weather Underground](https://github.com/ThingPulse/esp8266-weather-station/blob/master/src/WundergroundHourly.cpp#L75-L86) and (deep-)sleeps in between. After a couple of intervals (non deterministic) reading from the client causes the whole system to freeze when connected to an Apple Time Capsule AirPort. However, when connected to an iPhone hotspot all is well. When things go wrong there's no crash, no information on the serial monitor, no nothing - just an unrecoverable freeze.

marcelstoer, 2018-04-29:
FWIW another customer reported this at ThingPulse/espaper-weatherstation#12. He/she uses one of our [ePaper Weather Station](https://thingpulse.com/product/espaper-plus-kit-wifi-epaper-display/)s which run off an ESP8266 Wroom 2 MCU (same I tested with).

Since I can reliably reproduce it my self with an Apple Time Capsule AirPort AP I'll see if I manage to produce meaningful debug info.

devyte, 2018-07-13:
@eshvatskyi @marcelstoer several fixes have been merged since this was opened, especially into the lwip glue. Is this issue still valid in latest git?

devyte, 2018-11-12:
@eshvatskyi @marcelstoer ping?

marcelstoer, 2018-11-12:
I haven't been seeing this with 2.4.2.

eshvatskyi, 2018-11-12:
I'm going to check this in the nearest time.
Looks like you could close this issue.
I will reopen it in case of need. 
Thanks

devyte, 2018-11-12:
Thanks. Closing.

closed 2018-11-12 by devyte

ibaranov-cp, 2018-11-17:
This still happens unfortunately...