esp8266/Arduino #4526: Calling pure virtual method causes Panic, reboots ESP8266

chadham, 2018-03-16:
Version 2.4.1, NODEMCU

Calling a pure virtual method causes the following error 
Panic /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/abi.cpp:52 __cxa_pure_virtual

If I provide a dummy implementation and remove the =0 (ie. not make it a pure virtual function, it works fine.

Stack:
3fff2000:  3ffe9f3b 3fff3ad0 3fff2030 40100690  
3fff2010:  3fff3dc4 3fff3ad0 3fff3f04 4021a2cc  
3fff2020:  3ffe9f3b 3fff3ad0 3fff3f04 4020b3c2  
3fff2030:  00000000 00000000 00000000 401004d8  
3fff2040:  3fff3ad0 00000000 3fff2070 40207200  
3fff2050:  3fff3dec 00000001 00000001 3ffe8ba1  
3fff2060:  3fff3ad0 00000000 3fff3f04 4020b8ad  
3fff2070:  00000000 00000000 00000000 4010053d  
3fff2080:  3fff3dac 37363534 401070f0 66656463  
3fff2090:  3fff3ac4 00000200 3fff2294 4021bcc5  
3fff20a0:  00000010 3fff2140 3fff2140 40219060  
3fff20b0:  3fff3f04 3fff3f24 3fff2294 4020bb60  
3fff20c0:  00000010 00000010 3fff2140 4021988d  
3fff20d0:  3fff3f24 00000000 3fff2140 40219a21  
3fff20e0:  3fff2100 3fff214c 3fff2140 4021c1e0  
3fff20f0:  3ffe9456 00000006 3fff214c 40219bbf  
3fff2100:  3fff1164 000002ca 000002ca 4010020c  
3fff2110:  3fff3f24 00000000 3fff214c 40100690  
3fff2120:  3fff3f24 00000000 3fff3f24 3ffe8ba1  
3fff2130:  3fff3f24 00000000 3fff3f04 4020bd50  
3fff2140:  00000000 00000000 00000000 00000000  
3fff2150:  00000000 00000000 3fff21a0 00000320  
3fff2160:  00000001 00000001 3fff3ed4 00000000  
3fff2170:  00000000 3fff0af0 3fff21a0 00000005  
3fff2180:  3fff3f04 3fff0af0 00000001 40207611  
3fff2190:  00000000 00000005 3fff0af0 40209ad9  
3fff21a0:  3fff3ed4 0000000f 00000008 3fff3eec  
3fff21b0:  0000000f 00000001 3fff21cc 3fff21a0  
3fff21c0:  3fff0e10 3fff3ed4 3fff210a 401070e0  
3fff21d0:  00000000 00000000 00000100 3fff0814  
3fff21e0:  00000000 3ffe90c6 3fff0af0 40209bd8  
3fff21f0:  00000000 00000000 00000000 4020fa99  
3fff2200:  3fffdad0 3fff07e0 3fff0af0 4020a30c  
3fff2210:  00000000 00000000 00000000 00000000  
3fff2220:  00000000 00000000 00000000 00000000  
3fff2230:  00000000 feefeffe feefeffe feefeffe  
3fff2240:  feefeffe feefeffe feefeffe 3fff1248  
3fff2250:  3fffdad0 00000000 3fff1241 4021a3a4  
3fff2260:  feefeffe feefeffe 3fff1250 40100710

devyte, 2018-03-16:
Closing due to lack of info. Please see issue POLICY doc and issue template for more details.

closed 2018-03-16 by devyte

igrr, 2018-03-17:
@chadham Just curious, what behaviour do you expect when calling a pure virtual function?

chadham, 2018-03-17:
I found my error.

I was inadvertently calling a pure virtual function from the base objects constructor ... ( I wasn’t aware that in C++ the bass constructor is called first and then the derived constructor. Obviously then there was no virtual function to call until the derived class is constructed)

I changed it so the pure virtual function is called from the derived classes constructor (after the base class is done constructing) and it works fine.