esp8266/Arduino #4528: UDP malfunctions

marecl, 2018-03-16:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12E]
- Core Version:			[2.4.1]
- Development Env:		[Arduino IDEr]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[n/a]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [256000] (serial upload only)

### Problem Description
So I've got NTPClient example (with little changes) and in this version UDP packet is not sent or not received. It never reaches the point where packet is parsed correctly.  
Default NTPClient example on the other hand works. Partially. One request (for every 5-6 sent) is valid.
Rollback to 2.4.0-rc1 solved problem and I've always got my NTP packet.
I have *not* tried UDP'ing local machines to check if it works or dosen't.  

In case I suck in describing things:
2.4.1: udp malfunctions (NTPClient)
2.4.0-rc1: everything works fine (also NTPClient)

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {
Serial.begin(9600);
Serial.println(updateNTP);
}

void loop() {
//something
}

uint32_t updateNTP(char* ntpServerName) {
  WiFiUDP udp;
  IPAddress timeServerIP;
  udp.begin(2390);
  WiFi.hostByName(ntpServerName, timeServerIP);
  Serial.println(F("Sending NTP packet..."));

  byte packetBuffer[48];
  memset(packetBuffer, 0, 48);
  packetBuffer[0] = 0b11100011;
  packetBuffer[1] = 0;
  packetBuffer[2] = 6;
  packetBuffer[3] = 0xEC;
  packetBuffer[12] = 49;
  packetBuffer[13] = 0x4E;
  packetBuffer[14] = 49;
  packetBuffer[15] = 52;

  int cb = udp.parsePacket();

  uint8_t i = 0;
  while (!cb && i++ < 10) { //Exits usually after 4-5th iteration but it's invalid
    udp.beginPacket(timeServerIP, 123);
    udp.write(packetBuffer, 48);
    udp.endPacket();
    udp.flush();
    delay(1000);
    cb = udp.parsePacket();
  }

  if (i < 10 && cb) {
    udp.read(packetBuffer, 48);
    unsigned long highWord = word(packetBuffer[40], packetBuffer[41]);
    unsigned long lowWord = word(packetBuffer[42], packetBuffer[43]);
    unsigned long secsSince1900 = highWord << 16 | lowWord;
    secsSince1900 = secsSince1900 - 2208988800UL;
    secsSince1900 += settings.timezone * 3600;
    Serial.println(secsSince1900);
    return secsSince1900;
  } else return false;
}
```

d-a-v, 2018-03-18:
Add this after ```WiFi.begin``` and before NTP
```
while (WiFi.status() != WL_CONNECTED)
{
  Serial.print('.');
  delay(250);
}
```

devyte, 2018-03-18:
Plus a timeout mechanism, to avoid infinite loop in case of wrong password or whatever.

marecl, 2018-03-23:
I said, this is simplified. It's here only for context purposes.  

Once again in case  You forgot to read: I'm taking about existing issue that degrades UDP reliability because in 2.4.0-rc1 UDP was working properly. Later updates were making UDP unstable and rendering it useless because getting packet is a lottery.

5chufti, 2018-03-24:
once I got your sketch compiling and working (example code should at least compile for curtesy...) I can not confirm your problem at all. I allways get an instant and correct ntp timestamp as answer, using both lwip v2 and precompiled v1.4 variants.

```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

uint32_t updateNTP(char* ntpServerName) {
  WiFiUDP udp;
  IPAddress timeServerIP;
  udp.begin(2390);
  WiFi.hostByName(ntpServerName, timeServerIP);
  Serial.println(F("Sending NTP packet..."));

  byte packetBuffer[48];
  memset(packetBuffer, 0, 48);
  packetBuffer[0] = 0b11100011;
  packetBuffer[1] = 0;
  packetBuffer[2] = 6;
  packetBuffer[3] = 0xEC;
  packetBuffer[12] = 49;
  packetBuffer[13] = 0x4E;
  packetBuffer[14] = 49;
  packetBuffer[15] = 52;

  int cb = udp.parsePacket();

  uint8_t i = 0;
  while (!cb && i++ < 10) { //Exits usually after 4-5th iteration but it's invalid
    udp.beginPacket(timeServerIP, 123);
    udp.write(packetBuffer, 48);
    udp.endPacket();
    udp.flush();
    delay(1000);
    Serial.print(".");
    cb = udp.parsePacket();
  }

  if (i < 10 && cb) {
    udp.read(packetBuffer, 48);
    unsigned long highWord = word(packetBuffer[40], packetBuffer[41]);
    unsigned long lowWord = word(packetBuffer[42], packetBuffer[43]);
    unsigned long secsSince1900 = highWord << 16 | lowWord;
    secsSince1900 = secsSince1900 - 2208988800UL;
    Serial.println(secsSince1900);
    return secsSince1900;
  } else return false;
}

void setup() {
  Serial.begin(74880);
  WiFi.mode(WIFI_STA);
  WiFi.begin("mySSID", "myPASSPHRASE");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print('.');
    delay(250);
  }
  Serial.println("WiFi connected");
  Serial.println(updateNTP("pool.ntp.org"));
}

void loop() {
  //something
}
```

devyte, 2018-03-24:
Thanks @5chufti !
Closing as can't reproduce.

closed 2018-03-24 by devyte

marecl, 2018-04-23:
Please re-open. I used NTPClient example again using all LwIP variants. 
As I said version 2.4.0-rc1 worked fine with no packet loss. Now, again - I want this issue to be solved or get a workaround because my project *partially relies* on NTP server.

Version: latest github (d582cab)

I dont know how @5chufti couldnt reproduce as this issue occurs with US, EU and even local (PL) NTP servers.

````
WiFi connected  
IP address:   
192.168.2.84  
Starting UDP  
Local port: 2390  
sending NTP packet...  
no packet yet  
sending NTP packet...  
no packet yet  
sending NTP packet...  
no packet yet  
sending NTP packet...  
no packet yet  
sending NTP packet...  
no packet yet  
sending NTP packet...  
no packet yet  
sending NTP packet...  
no packet yet  
````