esp8266/Arduino #4531: EEPROM data read fails

namila007, 2018-03-17:
#### Platform

- Hardware:					[nodemcu amica ESP8266MOD]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|]
- Operating System:		[Windows|]

### Settings in IDE

- Module:  			|Nodemcu]
- Flash Mode:			[qio|dio|other]?
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description
hi, in my code i did a simple thing. i get the data from jSon object and write them to the EEPROM. once i write them to eeprom i can read the relevant data from that writing function ``handleFinishWizard()``  (i have put reading functions here n there to find where is the problem) .When i write data and finish it i can read the data.once its over when the program moved to the loop, eeprom data are corrupted and it doesnt display anything. i attached the output in debug message. whats the problem here?

to simulate this problem, once the sketch uploaded,do the following;
1. send POST request to `192.168.4.1:8080/api/config`
set header `aplication/json' and body with;
``{
	"ssid":"SOMETHING",
	"password": "SOMETHING",
	"link":"sdfdfs"
}``

2.GET request to `192.168.4.1:8080//api/finish`

thanks,
namila

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
code in github [CODE LINK](https://github.com/namila007/mediot-hardware/blob/dev_print/main/main.ino)

### Debug Messages

```
add 1
aid 1
station: c0:ee:fb:ed:3b:0e join, AID = 1
value on EEPROM ADD: 
OnePlus 3T
test123@abc
test.com

value on EEPROM ADD: 
OnePlus 3T
test123@abc
test.com

value on EEPROM ADD: 
OnePlus 3T
test123@abc
test.com

value on EEPROM ADD: 
OnePlus 3T
test123@abc
test.com

value on EEPROM ADD: 
OnePlus 3T
test123@abc
test.com

value on EEPROM saveBefore: 
⸮?

⸮?

value on EEPROM saveBefore: 
⸮?

⸮?

value on EEPROM saveBefore: 
⸮?

⸮?

value on EEPROM saveBefore: 
⸮?

⸮?

value on EEPROM saveBefore: 
⸮?

⸮?

value on EEPROM save: 

value on EEPROM save: 

value on EEPROM save: 

```

WereCatf, 2018-03-17:
Plenty of user-errors. ```
void setupCredentials(const char *ssid, const char *password, const char* link){
--SNIP--
EEPROM.put(addr,ssid);
```
This isn't how you store a character-string into EEPROM: you're storing the *memory-address* pointing to the string, not the string itself. Of course it's not going to work.

devyte, 2018-03-17:
Closing due to user error.

closed 2018-03-17 by devyte

namila007, 2018-03-17:
@WereCatf  thank you for pointing out that mistake.thanks 👍