esp8266/Arduino #4533: ESP8266 Servo Library

vitecd, 2018-03-18:
#### Platform

- Hardware:					[ESP-07]
- Core Version:			[2.4.1]
- Development Env:		[Arduino IDE 1.8.5]
- Operating System:		[Windows 7 x64]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[512K]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I've tried to using ESP as a WiFi locker, unfortunately servo (SG90) opening the locker then object attaching in Setup section. I've tried to change DEFAULT_PULSE_WIDTH parameter in Servo.h, but it doesn't have any effect. How I can change init position?

### https://ruvideos.org/3uirR6NHxsg-wifi-%D0%B7%D0%B0%D0%BC%D0%BE%D0%BA-%D0%BE%D1%82%D0%BA%D1%80%D1%8B%D1%82%D0%B8%D0%B5-%D0%B4%D0%B2%D0%B5%D1%80%D0%B5%D0%B9-%D0%BF%D1%80%D0%B8-%D0%BF%D1%80%D0%B8%D0%B1%D0%BB%D0%B8%D0%B6%D0%B5%D0%BD%D0%B8%D0%B8-%D0%BD%D0%B0-esp8266.html Sketch

code been changed and attached bellow 

`#include <ESP8266WiFi.h>
#include <Servo.h>

const char *ssid     = "BugLocker";
const char *password = "*****"; // 12345678
const unsigned char mac1[]    = {****};
const unsigned char mac2[]    = {****};
const unsigned char mac3[]    = {****};

//unsigned long previousMillis = 0;
//const long interval = 7000;
//const int SleepTime = 7;

bool ConnectFlag = false;
Servo myservo;  // create servo object to control a servo 
                // twelve servo objects can be created on most boards

WiFiEventHandler stationConnectedHandler; // событие - соединение
WiFiEventHandler stationDisconnectedHandler; // событие - разъединение

void setup() {
//  Serial.begin(115200);
  lock();
  myservo.attach(5);  // attaches the servo on GIO5 to the servo object 
  pinMode(4, OUTPUT); // реле на GPIO13 (D7)
  //  digitalWrite(13, HIGH); // начальное значение
  WiFi.persistent(false); // бережем флеш-память, не перезаписываем данные подключения, если они не изменились с прошлой загрузки
  WiFi.mode(WIFI_AP); // Режим точки доступа
  WiFi.softAP(ssid, password);
  tone(4, 600, 100);

  stationConnectedHandler    = WiFi.onSoftAPModeStationConnected   (&onStationConnected);    // при соединении переходим к функции
  stationDisconnectedHandler = WiFi.onSoftAPModeStationDisconnected(&onStationDisconnected); // при разъединении переходим к функции
}

void loop() {
//  unsigned long currentMillis = millis();
//  if (currentMillis - previousMillis >= interval && ConnectFlag == false) {
//    previousMillis = currentMillis;
//    Serial.println("Sleep "+String(SleepTime)+" Sec.");
//    ESP.deepSleep(SleepTime * 1000000, WAKE_RFCAL);
//  } 
}

void lock() {
  myservo.write(160);
}

void unlock () {
  myservo.write(65);
}

void onStationConnected(const WiFiEventSoftAPModeStationConnected& evt) { // при коннекте - включаем реле
//  Serial.print("ENTERED: ");

  if ( memcmp(evt.mac, mac1, 6) == 0) {
//  Serial.print("god1o: ");
    unlock();
    tone(4, 1800, 100);
}
  if ( memcmp(evt.mac, mac2, 6) == 0) {
//  Serial.print("good2: ");
    unlock();
    tone(4, 1800, 100);
  }
  if ( memcmp(evt.mac, mac3, 6) == 0) {
//  Serial.print("good3: ");
    unlock();
    tone(4, 1800, 100);
  }
  
  Serial.print("not good: ");
//  digitalWrite(13, LOW);
  ConnectFlag = true;
//  Serial.println(macToString(evt.mac)); // МАС-адрес в порт
}

void onStationDisconnected(const WiFiEventSoftAPModeStationDisconnected& evt) { // при дисконнекте - выключаем реле
//  Serial.print("DISCONNECTED: ");
  lock();
  //    digitalWrite(13, HIGH);
  ConnectFlag = false;
//  Serial.println(macToString(evt.mac)); // МАС-адрес в порт
  tone(4, 200, 100);
}

String macToString(const unsigned char* mac) { // Получаем МАС-адрес клиента в виде строки (может пригодиться при идентификации)
  char buf[20];
  snprintf(buf, sizeof(buf), "%02x:%02x:%02x:%02x:%02x:%02x",
           mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
  return String(buf);
}`

vitecd, 2018-03-18:
additional:
I've tried to use same code on Arduino UNO, changed the DEFAULT_PULSE_WIDTH and it has effected. Browsing ESP code, unable to found why it not working :(

earlephilhower, 2018-03-29:
I'm sitting here watching the Servo demo app move a SG90 back and forth just fine on an WeMos D1 Mini.  Runs like a champ.  I've tried both 80 and 160Mhz, both run fine.  You might have better luck @ www.esp8266.com asking about this, but I'm not seeing any issues w/the Arduino libs.

devyte, 2018-03-29:
Closing as can't reproduce.

closed 2018-03-29 by devyte