esp8266/Arduino #4535: Error compiling ESP8266WiFi.h: 'struct station_config' has no member named 'threshold'

Freddyice31, 2018-03-19:
----------------------------- Delete below -----------------------------
When attempting to compile your WiFi Scan example several errors fail at compile indicating 'struct station_config' has no member named 'threshold'

Wifi scan example code: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiScan/WiFiScan.ino

----------------------------- Delete above -----------------------------

### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [ X] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:				ESP-01
- Core Version:			Latest commit f9110f5  3 hours ago (19 March 2018)
- Development Env:		Platformio
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 		1MB
- lwip Variant:   v1.4
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

I'm getting errors trying to compile ESP8266WiFi.h

For main.c, I'm using the code from your WiFi scanner example.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

// Code stolen from example https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiScan/WiFiScan.ino
#include <Arduino.h>
#include "ESP8266WiFi.h"

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  Serial.println("Setup done");
}

void loop() {
  Serial.println("scan start");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  if (n == 0) {
    Serial.println("no networks found");
  } else {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i) {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE) ? " " : "*");
      delay(10);
    }
  }
  Serial.println("");

  // Wait a bit before scanning again
  delay(5000);
}

```

### Debug Messages
lib\ESP8266WiFi\src\ESP8266WiFiSTA.cpp: In member function 'wl_status_t ESP8266WiFiSTAClass::begin(const char*, const char*, int32_t, const uint8_t*, bool)':
lib\ESP8266WiFi\src\ESP8266WiFiSTA.cpp:129:10: error: 'struct station_config' has no member named 'threshold'
conf.threshold.rssi = -127;
^
lib\ESP8266WiFi\src\ESP8266WiFiSTA.cpp:132:10: error: 'struct station_config' has no member named 'threshold'

conf.threshold.authmode = AUTH_OPEN;
^
lib\ESP8266WiFi\src\ESP8266WiFiSTA.cpp: In member function 'bool ESP8266WiFiSTAClass::getAutoReconnect()':
lib\ESP8266WiFi\src\ESP8266WiFiSTA.cpp:364:46: error: 'wifi_station_get_reconnect_policy' was not declared in this scope
return wifi_station_get_reconnect_policy();
^
*** [.pioenvs\esp01_1m\libe13\ESP8266WiFi\ESP8266WiFiSTA.cpp.o] Error 1

devyte, 2018-03-19:
@Freddyice31 I can't reproduce this in the IDE. It sounds like an installation issue with platformio. 
The conf.threshold.rssi member is new to SDK 2.2, so if the error pops up, it's because you're somehow trying to build a recent core with an earlier SDK.
Closing.

closed 2018-03-19 by devyte

leobel96, 2018-03-26:
Same issue here. @Freddyice31 have you resolved it?

Freddyice31, 2018-04-02:
Sorry for delay. I have not solved it. All packages in Atom/Platformio come back as "up-to-date".

Freddyice31, 2018-04-02:
I got a successful compile by changing my platform.ini 'platform' line. Haven't tested to see if it works yet.
It now says:

[env:esp01_1m]
;platform = espressif8266   ; commented out for posterity to determine why
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = esp01_1m
framework = arduino

Freddyice31, 2018-04-02:
Sorry for the delay; I hadn't had a chance to revisit this until this morning.
Yes. I got it to work by editing the platform.ini file. I changed the 'platform' value. The file now looks like this:
[env:esp01_1m]
;platform = espressif8266    ; commented out for posterity to determine why
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = esp01_1m
framework = arduino

leobel96, 2018-04-04:
Perfect, thank you.