esp8266/Arduino #4538: ESP8266 - exception 29 wdt reset

SiESP, 2018-03-20:
**Platform**

- Hardware:				NodeMCUv1
- Development Env:		Arduino 1.8.5
- Operating System:		MacOS

**Settings in IDE**

- Module:  			Nodemcu 1.0
- Flash Mode:			dio
- Flash Size: 		        4MB (1M SPIFFS)
- lwip Variant:                 v2 Lower Memory
- Reset Method:		nodemcu
- CPU Frequency:		80Mhz
- Upload Using:		USB/SERIAL
- Upload Speed:            115200

**Problem Description**

Keep getting exception 29 due to wdt reset when using wifi in sketches.

The board seems to be fine for non-wifi based sketches.

Have tested this with several basic wifi sketches and all fail after a few seconds.

I have searched through here and tried new firmware, erasing the flash in multiple ways (esptool and through Arduino IDE), using the latest Arduino IDE version, library updates etc but nothing seems to solve this.

Have used delay() in the sketches but that doesn't seen to solve this either. Even the example wifi sketches fail.

**Sketch**

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <WebSocketsServer.h>
#include <Hash.h>

// Replace with your network credentials
const char* ssid = "XXX";
const char* password = "XXX";

WebSocketsServer webSocket = WebSocketsServer(81);
ESP8266WebServer server(80);  //instantiate server at port 80 (http port)

String page = "";
int LEDPin = BUILTIN_LED;

void setup(void){
  //the HTML of the web page
  page = "<h1>Simple NodeMCU Web Server</h1><p><a href=\"LEDOn\"><button>ON</button></a>&nbsp;<a href=\"LEDOff\"><button>OFF</button></a></p>";
  //make the LED pin output and initially turned off
  pinMode(LEDPin, OUTPUT);
  digitalWrite(LEDPin, HIGH);

  delay(1000);

  Serial.begin(115200);
  WiFi.begin(ssid, password); //begin WiFi connection
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  server.on("/", [](){
    server.send(200, "text/html", page);
  });

  server.on("/LEDOn", [](){
    server.send(200, "text/html", page);
    digitalWrite(LEDPin, LOW);
    delay(1000);
  });
  server.on("/LEDOff", [](){
    server.send(200, "text/html", page);
    digitalWrite(LEDPin, HIGH);
    delay(1000); 
  });

  server.begin();
  webSocket.begin();
  webSocket.onEvent(webSocketEvent);

  Serial.println("Web server started!");
  }

void loop(void){
  webSocket.loop();
  server.handleClient();
  if (Serial.available() > 0){
    char c[] = {(char)Serial.read()};
    webSocket.broadcastTXT(c, sizeof(c));
    }
  }

void webSocketEvent(uint8_t num, WStype_t type, uint8_t * payload, size_t length){
  if (type == WStype_TEXT){
  for(int i = 0; i < length; i++) Serial.print((char) payload[i]);
  Serial.println();
  }
}

```

**Exception in serial console**

```
Exception (29):
epc1=0x4020eb84 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffcd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe70:  3fff0b80 3fff0b50 00000000 402161b3  
3ffffe80:  00000000 00000000 0000001f 402126b7  
3ffffe90:  3ffef520 00000336 00000336 fffffffb  
3ffffea0:  3fff0a50 00000000 3fff0a4c 4020ec35  
3ffffeb0:  40208037 3ffe92bc 3fff0a4c 402121a8  
3ffffec0:  00000000 00000007 3fff1f1c 40211c35  
3ffffed0:  3ffe92bc 3ffe92bc 3fff0a4c 402121d6  
3ffffee0:  00000000 3fff0a4c 3fff1f1c 402146f8  
3ffffef0:  fa00000a 00000000 00000020 4020f084  
3fffff00:  00000000 3fff0a4c 3fff1f1c 4021474b  
3fffff10:  40210068 3fff06b0 3fff0b74 40100690  
3fffff20:  3ffee870 3ffe93c4 00000032 3fff06ac  
3fffff30:  40210068 3fff06b0 3ffe9250 40210076  
3fffff40:  40104d7e 012011bf 00000001 40210134  
3fffff50:  3ffe9250 3ffe93c4 012011bf 00000000  
3fffff60:  4022fab4 3ffee848 3ffee870 60000600  
3fffff70:  04ccc872 3ffee870 3ffee848 4022fac1  
3fffff80:  4022fb06 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffee898 3fffdad0 3ffef604 4020802f  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

**Exception decoder output**

```
0x402161b3: ip_chksum_pseudo at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/inet_chksum.c line 542
0x402126b7: udp_sendto_if_src at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/udp.c line 463
0x4020ec35: netif_sta_status_callback at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 303
0x40208037: loop_task at /Users/Simon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 57
0x402121a8: netif_set_ipaddr at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/netif.c line 711
0x40211c35: pbuf_free_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/pbuf.c line 1306
0x402121d6: netif_set_netmask at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/netif.c line 711
:  (inlined by) netif_set_addr_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/netif.c line 359
0x402146f8: dhcp_release_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/dhcp.c line 1314
0x4020f084: do_memp_malloc_pool at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c line 231
0x4021474b: dhcp_coarse_tmr at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/dhcp.c line 423
0x40210068: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 165
0x40100690: free at /Users/Simon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/umm_malloc/umm_malloc.c line 1739
0x40210068: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 165
0x40210076: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 171
0x40104d7e: wdt_feed at ?? line ?
0x40210134: sys_check_timeouts at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 351
0x4022fab4: ets_timer_handler_isr at ?? line ?
0x4022fac1: ets_timer_handler_isr at ?? line ?
0x4022fb06: ets_timer_handler_isr at ?? line ?
0x4020802f: loop_task at /Users/Simon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 57
```

Any suggestions gratefully received and appreciated as I have come to the end of my knowledge and abilities after 3 days of trying to sort this out.

WereCatf, 2018-03-20:
Does adding `WiFi.mode(WIFI_STA);` before WiFi.begin() change anything?

SiESP, 2018-03-20:
Now get a different exception 29 error

```
Exception (29):
epc1=0x4020eb8c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffcd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe70:  3fff0b70 00000000 00000000 402161bb  
3ffffe80:  00000000 00000000 0000001f 402126bf  
3ffffe90:  3ffef520 00000094 00000094 fffffffb  
3ffffea0:  3fff0a50 00000000 3fff0a4c 4020ec3d  
3ffffeb0:  40208023 3ffe92bc 3fff0a4c 402121b0  
3ffffec0:  00000000 00000007 3fff1f1c 40211c3d  
3ffffed0:  3ffe92bc 3ffe92bc 3fff0a4c 402121de  
3ffffee0:  00000000 3fff0a4c 3fff1f1c 40214700  
3ffffef0:  fa00000a 3ffe9eb0 3ffe9ec3 4020f08c  
3fffff00:  00000000 3fff0a4c 3fff1f1c 40214753  
3fffff10:  40210070 3fff06b0 3fff0b74 40100690  
3fffff20:  4022b2ae 00000000 00000034 3fff06ac  
3fffff30:  40210070 3fff06b0 3ffe9250 4021007e  
3fffff40:  40104d7e 01201147 00000017 4021013c  
3fffff50:  3ffe9250 3ffe93c4 01201147 00000000  
3fffff60:  4022fabc 3ffee848 3ffee870 60000600  
3fffff70:  0399dba8 3ffee870 3ffee848 4022fac9  
3fffff80:  4022fb0e 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffee898 3fffdad0 3ffef604 40208037  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
```

Decoded stack
```
Decoding 23 results
0x402161bb: ip_chksum_pseudo at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/inet_chksum.c line 542
0x402126bf: udp_sendto_if_src at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/udp.c line 463
0x4020ec3d: netif_sta_status_callback at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 303
0x402121b0: netif_set_ipaddr at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/netif.c line 711
0x40211c3d: pbuf_free_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/pbuf.c line 1306
0x402121de: netif_set_netmask at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/netif.c line 711
:  (inlined by) netif_set_addr_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/netif.c line 359
0x40214700: dhcp_release_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/dhcp.c line 1314
0x40214753: dhcp_coarse_tmr at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/dhcp.c line 423
0x40210070: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 165
0x40100690: free at /Users/Simon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/umm_malloc/umm_malloc.c line 1739
0x4010100f: ppCalFrameTimes at ?? line ?
0x40210070: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 165
0x4021007e: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 171
0x40104d7e: wdt_feed at ?? line ?
0x4021013c: sys_check_timeouts at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 351
0x4022fabc: ets_timer_handler_isr at ?? line ?
0x4022fac9: ets_timer_handler_isr at ?? line ?
0x4022fb0e: ets_timer_handler_isr at ?? line ?
0x40208037: loop_task at /Users/Simon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 57
```

SiESP, 2018-03-20:
Just tried a different sketch without websocket and get another different exception 29

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <OneWire.h>
#include <DallasTemperature.h>

// Replace with your network credentials
const char* ssid = "BTHub6-87PR_2GEXT";
const char* password = "i4dh7gMdnPmK";
ESP8266WebServer server(80);  //instantiate server at port 80 (http port)

// Data wire is conntec to the digital pin 2
#define ONE_WIRE_BUS D2

// Setup a oneWire instance to communicate with any OneWire devices
OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature sensor 
DallasTemperature sensors(&oneWire);

String page = "";
String text = "";
double data;

void setup(void){
  delay(1000);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password); //begin WiFi connection
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
  delay(500);
  Serial.print(".");
  }

  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  server.on("/data.txt", [](){
    text = (String)data;
  server.send(200, "text/html", text);
  });
  server.on("/", [](){
    page = "<h1>Sensor to Node MCU Web Server</h1><h1>Data:</h1> <h1 id=\"data\">""</h1>\r\n";
    page += "<script>\r\n";
    page += "var x = setInterval(function() {loadData(\"data.txt\",updateData)}, 1000);\r\n";
    page += "function loadData(url, callback){\r\n";
    page += "var xhttp = new XMLHttpRequest();\r\n";
    page += "xhttp.onreadystatechange = function(){\r\n";
    page += " if(this.readyState == 4 && this.status == 200){\r\n";
    page += " callback.apply(xhttp);\r\n";
    page += " }\r\n";
    page += "};\r\n";
    page += "xhttp.open(\"GET\", url, true);\r\n";
    page += "xhttp.send();\r\n";
    page += "}\r\n";
    page += "function updateData(){\r\n";
    page += " document.getElementById(\"data\").innerHTML = this.responseText;\r\n";
    page += "}\r\n";
    page += "</script>\r\n";
  server.send(200, "text/html", page);
});

[... 51 lines cut ...]

Decoded output:
```
Decoding 22 results
0x4020ec58: tcp_abort at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp.c line 1690
0x4020dcc5: netif_sta_status_callback at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 303
0x40211238: netif_set_ipaddr at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/netif.c line 711
0x40210cc5: pbuf_free_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/pbuf.c line 1306
0x40211266: netif_set_netmask at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/netif.c line 711
:  (inlined by) netif_set_addr_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/netif.c line 359
0x40213788: dhcp_release_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/dhcp.c line 1314
0x4020e114: do_memp_malloc_pool at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c line 231
0x402137db: dhcp_coarse_tmr at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/dhcp.c line 423
0x4020f0f8: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 165
0x40100690: free at /Users/Simon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/umm_malloc/umm_malloc.c line 1739
0x4020f074: tcpip_tcp_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 126
0x4020f0f8: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 165
0x4020f106: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 171
0x40104d6b: wdt_feed at ?? line ?
0x4020f1c4: sys_check_timeouts at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 351
0x4022d7bc: ets_timer_handler_isr at ?? line ?
0x4022d7c9: ets_timer_handler_isr at ?? line ?
0x4022d80e: ets_timer_handler_isr at ?? line ?
0x402072ef: loop_task at /Users/Simon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 57
```

WereCatf, 2018-03-20:
Just out of curiosity, why don't you try flashing the WiFiScan.ino.bin I just compiled ( https://www.dropbox.com/s/1h37h96xl3cr1we/WiFiScan.ino.bin?dl=0 ) and seeing if it works? If it does, then it's probably a problem with your setup, since it works fine here on my end.

PS. I don't know how to flash a precompiled binary under Mac, but under Windows `esptool.exe -vv -cd nodemcu -cb 921600 -cp COM3 -ca 0x00000 -cf WiFiScan.ino.bin` should do the trick.

SiESP, 2018-03-20:
I can try - is this an Arduino sketch or binary or something? Apologies for my lack of knowledge.

To help me decipher the esptool cmd line instruction can you tell me what the -vv -cd -cd -cp -ca and -cf flags cover?

I don't have esptool tool running on a Windows machine at the moment unfortunately.

WereCatf, 2018-03-21:
It's the WiFiScan-sketch that I compiled, nothing fancier than that. Also, I don't know what the flags are, I just copied the command that the Arduino IDE uses to flash the binary after compiling it (You can see it by enabling the File->Preferences->Show verbose output during upload)

SiESP, 2018-03-21:
Will check it out - if I can translate the flags from esptool.exe to esptool.py on the Mac I will give it a try.

Do I need the .bin file or can you send me the .ino sketch and I can upload that directly?

Apologies for my lack of knowledge!

Simon

Dynamite-7, 2018-04-05:
Hey everyone..
         I'm getting this in serial monitor of arduino...  please suggest some solutions, I'm unable to check my desired output.

Exception (9):
epc1=0x402027ae epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000001f depc=0x00000000

ctx: cont 
sp: 3ffefe40 end: 3fff0090 offset: 01a0

>>>stack>>>
3ffeffe0:  0000004b 00000050 3ffeed48 402027a8  
3ffefff0:  3ffe8cbc 3ffeefa0 3ffe8cb8 a0640534  
3fff0000:  3ffe8994 3ffeefa0 0000004b 402032cc  
3fff0010:  40104e5a 3ffeefa0 3ffeed48 402021cd  
3fff0020:  00000000 00000000 00000000 402032a1  
3fff0030:  3ffe8cbc 3ffeed6c 3fff15b4 0000003f  
3fff0040:  00000038 3ffef060 3ffeefa0 402032cc  
3fff0050:  3fff068c 40202578 3ffeefa0 3ffef060  
3fff0060:  3fffdad0 00000000 3ffef058 402022c3  
3fff0070:  3fffdad0 00000000 3ffef058 402038a0  
3fff0080:  feefeffe feefeffe 3ffef070 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

Please help ASAP..!!!

Thanks.

Pablo2048, 2018-04-05:
Yeah, another ignoring rules/policy (aka too lazy to read), but 'help me ASAP' guy :-( Did you at least try exception decoder?

devyte, 2018-07-13:
Several fixes have gone in, especially into the lwip glue code, since this was opened. Is this issue still valid in latest git?

ulisses-santos, 2018-11-01:
Hi guys!

I'm using esp8266 in my application like a server TCP, and when it received RST package of client TCP, it's resetting... I'm sending below the image of communication.

[https://www.dropbox.com/s/5sec8a1kn358tv9/esp8266_resetting.PNG?dl=0](url)

Why is it resetting?

Thanks very much

devyte, 2018-11-14:
Closing due to age and lack of feedback.

closed 2018-11-14 by devyte