esp8266/Arduino #4540: upload works but does not run sketch

dwc56, 2018-03-21:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] 

### Problem Description

Detailed problem description goes here.

when i upload any sketch the upload works but the sketch does not run even blink does not work 

if i press reset it says:

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0xef
csum 0xef
csum err
ets_main.c 

### Debug Messages

dwc56, 2018-03-24:
it is not a power issue because i was powering it from a external 3.3v 3a power supply

RudyFiero, 2018-03-24:
If you have other modules that work then why are you opening an issue on this? Do you have any valid reason to suspect the ESP8266-Arduino framework?

dwc56, 2018-03-24:
i wanted to know if it was a known problem that it did not work for some modules or if it was a hardware problem with the module because i have 2 that do not work from different sellers

devyte, 2018-03-24:
@dwc56 if you're using a prebuilt ESP12 board like the nodemcu or wemos d1, then it's eithher a bad board, bad serial drivers, bad IDE settings, or bad installation.
If you're using a standalone ESP12, plus custom hardware, then it's most likely bad wiring, bad power supply (you seem to have already ruled out that last), or error with entering flash mode.
At this time, there are no known flashing issues with any board.
I suggest discussing at a community forum to figure out your particular problem.
Closing.

closed 2018-03-24 by devyte