esp8266/Arduino #4543: ESP8266 WiFi.scanNetworks returns 0...always

Normtot, 2018-03-21:
My new nodeMCU board (purchased March 2018), having no additional wiring other than the USB to a PC, will not respond with any wifi networks upon running a WiFi.scanNetworks call. The board is surrounded by many networks as per the results of my phone sitting next to it. Other sketches developed on the Arduino IDE for serial communications work well, as do sketches on the GPIO pins with sensors, but the WiFi does not work to minimally see networks.

I've tried flashing it, using capacitors for the power supply, grounding some pins, etc. This nodeMCU from HiLetGo with AI-THINKER ESP8266 just won't "see" any networks. Attempting to log into a specific network with WiFi.begin obviously doesn't work either because it doesn't see the network.

Can someone tell me if there is an even more basic test than WiFi.scanNetworks to run to determine if the WiFi portion of this chip is working?

Deimos1994, 2018-03-21:
Does soft AP works?
[Soft AP example](http://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/soft-access-point-examples.html)

Normtot, 2018-03-21:
My phone doesn't see the soft AP of the ESP8266.  I thought I would focus on the client portion of the ESP8266.

Deimos1994, 2018-03-21:
> My new nodeMCU board
Do you have other ESP8266? and if so do they work?
When using soft AP what do you see in serial "Ready" or "Failed!"?

Normtot, 2018-03-21:
I have no other ESP8266 boards.  When the sketch is run, some commands output some diagnostic results, such as mode: STA+AP, PHY mode: N, bcn 100, etc.  This is a sketch from Arduino IDE.  I suspect that Ready or Failed are AT command responses?  What type of network would I see from my phone if the soft AP was working?

Deimos1994, 2018-03-21:
```cpp
#include <ESP8266WiFi.h>

void setup()
{
  Serial.begin(115200);
  Serial.println();

  Serial.print("Setting soft-AP ... ");
  boolean result = WiFi.softAP("ESPsoftAP_01", "pass-to-soft-AP");
  if(result == true)
  {
    Serial.println("Ready");
  }
  else
  {
    Serial.println("Failed!");
  }
}

void loop()
{
  Serial.printf("Stations connected = %d\n", WiFi.softAPgetStationNum());
  delay(3000);
}
```
If you are using this example then you will see a network with a name ESPsoftAP_01
In the serial you should see this, if soft AP started successfully:
```
Setting soft-AP ... Ready
Stations connected = 0
Stations connected = 0
```
If soft ap start is not successful then you will see Failed! instead of Ready

Normtot, 2018-03-21:
Some success!  Thanks.  The code above returns "Ready".  So something is working.  However, it is returning zero stations connected, even having my phone's wi-fi next to it.  Also, my phone doesn't see the wi-fi network ESPsoftAP_01.

Deimos1994, 2018-03-21:
If it returns `Setting soft-AP ... Ready` it means you should see the wi-fi network ESPsoftAP_01 on your phone, and if you don't see it, then it means we have no idea what is the problem. Maybe it is a hardware issue, but I am not expert.

Normtot, 2018-03-21:
Thanks.  That's what I was afraid of.  :)

devyte, 2018-03-21:
Note: struct scan_config is inited with:
```
    struct scan_config config;
    memset(&config, 0, sizeof(config));
```

With SDK 2.2, scan_config struct received new data members:
```
struct scan_config {
    uint8 *ssid;    // Note: ssid == NULL, don't filter ssid.
    uint8 *bssid;    // Note: bssid == NULL, don't filter bssid.
    uint8 channel;    // Note: channel == 0, scan all channels, otherwise scan set channel.
    uint8 show_hidden;    // Note: show_hidden == 1, can get hidden ssid routers' info.
    wifi_scan_type_t scan_type; // scan type, active or passive    
    wifi_scan_time_t scan_time; // scan time per channel     // <--- new, struct has min and max times, both are now set to 0
};
```

It is currently unknown whether setting both min and max to 0 makes sense or is well-behaved.

devyte, 2020-10-14:
I have WiFi.scanNetworks working in my own app, and I believe @Erriez also just did a test with scanning while implementing #7629. That means that the current core has scan working correctly. 
Closing due to can't reproduce.

closed 2020-10-14 by devyte