esp8266/Arduino #4546: analogWriteFreq() depending on CPU frequency && only works -1 times

Metaln00b, 2018-03-21:
### Platform
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Wemos D1 mini]
- CPU Frequency:		[80Mhz|160MHz]

### Problem Description

analogWriteFreq() depending on CPU frequency.
The frequency of the buzzer at 80 MHz is lower than at 160 MHz.
In addition, the buzzer only sounds 3 times instead of 4 times

### Sketch

```c

void buzzer(uint8_t _pin, unsigned int frequency, unsigned long duration)
{
    pinMode (_pin, OUTPUT );
    analogWriteFreq(frequency);
    analogWrite(_pin,500);
    delay(duration);
    analogWrite(_pin,0);
}

void buzzer_attention(void)
{
    buzzer(5,2637,100);
    delay(133);
    buzzer(5,2637,100);
    delay(133);
    buzzer(5,2637,100);
    delay(133);
    buzzer(5,2637,100);
    delay(133);
}
void setup() {
    // put your setup code here, to run once:
    buzzer_attention();
}

void loop() {
    // put your main code here, to run repeatedly:
    ESP.wdtFeed();
}

```

devyte, 2018-07-13:
@Metaln00b several fixes have been merged since this was opened, including a waveform generator. Please retest with latest git.

Metaln00b, 2018-10-12:
> 
> 
> @Metaln00b several fixes have been merged since this was opened, including a waveform generator. Please retest with latest git.

Works great now =)

closed 2018-10-12 by Metaln00b