esp8266/Arduino #4548: MDNS stop responding for some time after I send data.

Tizana, 2018-03-22:
Hi,
My code is very simple, I use my esp8266 as a server which should wait for a client connection to read data from it and transmit it to the Arduino via the serial.
I use also the mDNS to detect the IP addresses of the ESP8266 inside the network and here's the code :

```
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <EEPROM.h>

.....
WiFiServer server(80);

void setup() {
  Serial.begin(115200);
  delay(10); 

  if (!MDNS.begin("Device 1")) { //
    while (1) {  delay(1000); }
  }
  MDNS.addService("SmartDimmerx2", "tcp", 80);

  WiFi.mode(WIFI_STA);
  delay(4000);
  // Check if WiFi is already connected and if not, 
  if (WiFi.status() != WL_CONNECTED){
    Serial.println("Disconnected");
    ConnecStat_Toarduino = false;
  }
  else {
    Serial.println("Connected");
    ConnecStat_Toarduino = true;
  }
  // Start the server
  server.begin();

}

 
void loop() {
  MDNS.update();

  if (WiFi.status() != WL_CONNECTED) {
    Serial.println("Disconnected");
    ConnecStat_Toarduino = false;
    delay(100);
  }
  else if ((WiFi.status() == WL_CONNECTED) && (ConnecStat_Toarduino = false)) {
    Serial.println("Connected");
    ConnecStat_Toarduino = true;
  }

  if (Serial.available() > 0) {
    String incomingChars = Serial.readStringUntil('\n');
    if(incomingChars.indexOf("WPS") != -1) {
     WPS_fnct();
    }
  }
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    time_cnx = millis();
    return;
  }

  // Wait until the client sends some data

  while (!client.available()) {
    if ((millis() - time_cnx)>2000) 
    {return;}
    delay(1);
  }

  // Read the first line of the request
  String req = client.readStringUntil('\r');

  client.flush();
  client.stop();
  
  // Match the request
  int val;
  if (req.indexOf("V=") != -1)
  {
    Serial.print(req.substring(2,5));
    Serial.println(req.substring(7,11));
     //Serial.println(char(req.substring(8,10).toInt()));
  }
  
  else {
    client.stop();
    return;
  }
}
```

I use my phone app as a client and the code work perfectly, I scan for the “_SmartDimmerx2._tcp.local.” and the phone detect the esp with its IP addresses.
I was able to send data to the  ESP and it receive it correctly and then send it back to arduino through serial, 

The problem was after I send continuously some data to the ESP and then i try again to scan for the mDNS the ESP didn't respond. The ESP still not responding to the mDNS scan for like 30 seconds/1 minutes and after that it become discoverable again. 
In the meantime when the mDNS is not responding the ESP still able to receive data from the phone and send it through the serial without any problem and display the correct data. So the esp is not bugged.

I feel like something is blocking the mDNS service when I send a lot of data from the phone. Did I made something wrong in my code ?
When the mDns is not responding I tried to scan with Zeroconf android  app  and it didn't respond too until after like one minute.

Tizana, 2018-03-23:
so i tried to activate the DEBUG in order to see what's wrong,
so i add #define DEBUG_ESP_MDNS_ERR in the beginning of the file and activate the debugging for serial inside the IDE and here's what i got.

1- when i first start the ESP i got these kind of lines that popping every like 2 seconds :

```
:urn 33
:urd 4, 33, 13
:urd 4, 33, 18
:urd 5, 33, 23
:urn 283

```

2- and every time i try to search for esp with my phone using the (ZeroconfResolver/mdns ) these line pop up, and my phone was able to find the esp8266 

```
:urn 43
:urd 14, 43, 13
:urd 4, 43, 28
:urd 5, 43, 33

```
3- then when i send a series of data with my phone to the esp i got these kind of lines :

```
WS:ac
:rn 12
:ref 1
WS:av
:ref 2
:ur 2
:c0 1, 12
:ur 1
:close
WS:dis
:del
WS:ac
:rn 12
:ref 1
WS:av
:ref 2
:ur 2
:c0 1, 12
:ur 1
:close
WS:dis
:del
.........
```

so i use a slider in my phone and every time i change the value of it, it should send this value to the ESP
[image]

4. after sending some data i try again to search for the esp8266 using the (ZeroconfResolver/mdns ) and the esp didn't respond and the phone wasn't able to detect it, and here's what i got in the debug : problem

```
:urn 43
:urd 14, 43, 13
:urd 4, 43, 28
:urd 5, 43, 33
failed _reservefailed _reservefailed _reservefailed _reservefailed _reservefailed _reservefailed _reservefailed _reservefailed _reservefailed _reservefailed _reservefailed .........
```

so to get the esp discoverable again i can either restart the esp or i can wait for like 30 seconds to 2 minutes and the ESP will be discoverable again.

i search for this error 'failed _reserve' and i found it here inside the UdpContext.h
https://github.com/esp8266/Arduino/blob/461c922586564b8b8b1ece7e5591531b38dbfb87/libraries/ESP8266WiFi/src/include/UdpContext.h#L270

can anyone please explain me why this problem is triggered ? it's look like the tx buffer is full ? what should i do then ?
i'm sorry i'm a noob in network things , i'm only a electronic ing.

thx

devyte, 2018-03-23:
Closing as duplicate of #4557 .

closed 2018-03-23 by devyte