esp8266/Arduino #4557: failed _reserve error with mDNS

Tizana, 2018-03-23:
hi,

my ESP work as a TCP server which should listen to the client, read data and transmit it to an ARDUINO through the serial. 

The ESP use the mDNS in order to be discoverable by the client.

the entire program work perfectly, the client is able to discover the ESP server and determine its IP address. also the ESP receive all the data from the client and transmit it to the ARDUINO through the serial.

but when i try to send a lot amount of data from the client to the ESP , like 200 data in approximately 2 seconds (200 datas means 200 connections with the ESP), and then i try to discover the ESP using the Zeroconf the ESP return me this error **failed _reserve** and the client wasn't able to resolve the hostname.

i looked for this error and i found it inside the **UdpContext.h** header file :
https://github.com/esp8266/Arduino/blob/461c922586564b8b8b1ece7e5591531b38dbfb87/libraries/ESP8266WiFi/src/include/UdpContext.h#L270

the ESP still not able to be discoverable for like 1 to 2 minutes and give always this type of error then magically after that this error disappear.
while the ESP give me this error i was able to send data from the client to the ESP without any problem.

here's the code :

```
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <EEPROM.h>

.....
WiFiServer server(80);

void setup() {
  Serial.begin(115200);
  delay(10); 

  if (!MDNS.begin("Device 1")) { //
    while (1) {  delay(1000); }
  }
  MDNS.addService("SmartDimmerx2", "tcp", 80);

  WiFi.mode(WIFI_STA);
  delay(4000);
  // Check if WiFi is already connected and if not, 
  if (WiFi.status() != WL_CONNECTED){
    Serial.println("Disconnected");
    ConnecStat_Toarduino = false;
  }
  else {
    Serial.println("Connected");
    ConnecStat_Toarduino = true;
  }
  // Start the server
  server.begin();

}

 
void loop() {
  MDNS.update();

  if (WiFi.status() != WL_CONNECTED) {
    Serial.println("Disconnected");
    ConnecStat_Toarduino = false;
    delay(100);
  }
  else if ((WiFi.status() == WL_CONNECTED) && (ConnecStat_Toarduino = false)) {
    Serial.println("Connected");
    ConnecStat_Toarduino = true;
  }

  if (Serial.available() > 0) {
    String incomingChars = Serial.readStringUntil('\n');
    if(incomingChars.indexOf("WPS") != -1) {
     WPS_fnct();
    }
  }
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    time_cnx = millis();
    return;
  }

  // Wait until the client sends some data

  while (!client.available()) {
    if ((millis() - time_cnx)>2000) 
    {return;}
    delay(1);
  }

  // Read the first line of the request
  String req = client.readStringUntil('\r');

  client.flush();
  client.stop();
  
  // Match the request
  int val;
  if (req.indexOf("V=") != -1)
  {
    Serial.print(req.substring(2,5));
    Serial.println(req.substring(7,11));
     //Serial.println(char(req.substring(8,10).toInt()));
  }
  
  else {
    client.stop();
    return;
  }
}
```

apparently this is a problem of buffer reservation ?? am'i right ?  but in my main program i use **client.flush()** every connection to freed memory space ? so why this happen ?

d-a-v, 2018-03-23:
check in the FAQ for tcpCleanup()

devyte, 2018-03-23:
@Tizana said:
> 200 data in approximately 2 seconds (200 datas means 200 connections...)

yeah, that's not gonna work. A socket is kept in TIME_WAIT state for a while after closing. This is by TCP spec, it happens inside lwip, and is not specific to the code in this repo (similar happens on a real PC). The previous comment and doc section in the FAQ reference a way to work around the issue by aborting those TIME_WAITs (there is some risk involved), so that you can recover from it.
Your design is flawed. You should open a connection and keep it open, or use (potentially lossy) UDP instead of TCP. Alternatively, don't open connections so frequently: use a time-poll with millis(), and open a connection every e.g.: 30s.
Closing due to user error.

closed 2018-03-23 by devyte

Tizana, 2018-03-24:
so i add this instruction `Serial.println( ESP.getFreeHeap(),DEC);` in the bottom of the loop and realized that my memory is decreasing so fast until reaching a level like 700 and since that it will work slowly and the mDNS won't respond.

1. i tried with tcpCleanup() function according to the FAQ http://arduino-esp8266.readthedocs.io/en/latest/faq/readme.html

```
// no need for #include
struct tcp_pcb;
extern struct tcp_pcb* tcp_tw_pcbs;
extern "C" void tcp_abort (struct tcp_pcb* pcb);

void tcpCleanup (void) {
  while (tcp_tw_pcbs)
    tcp_abort(tcp_tw_pcbs);
}
```
and then i call tcpCleanup(); in the bottom of the loop and the memory still decresing, i tried to call tcpCleanup(); in the begining of the loop and the same happening, memory leaks.

2. if i open the connection every 30s as you told me, in this case other clients can't connect to the ESP for 30 second because the esp is already connected, but in my application multiple client should be able to connect to ESP.

and even if i tried your solution i wasn't able to keep the connection with the client, because when the loop function return  the 'client' object is detroyed.
i tried to remove all the client.close() and declared WiFiClient client; outside the loop so the client become a global variable and won't be destroyed after every loop but the ESP still disconnect automatically after every loop.

```
#include .......
......
WiFiClient client;

void setup() {
.....
}

void loop() {
.....
client = server.available();
......
}
```

how can i keep the connection after every loop ?

d-a-v, 2018-03-26:
A leak introduced in 2.4.1 has been fixed, you should try with 2.4.0 or, better with sources from git.