esp8266/Arduino #4558: scanNetworks with Async = true

mjs513, 2018-03-24:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:			Link Node D1
- Core Version:			3/20/18
- Development Env:		Arduino IDE
- Operating System:	Windows 10

### Settings in IDE

- Module:  			Link Node D1 (yes I modified the boards.txt with this version since it is not listed used wemos d1 (retired)
- Flash Mode:	        [qio]
- Flash Size: 		[4MB]
- lwip Variant:          v2 Lower Memory
- Reset Method:	ck
- Flash Frequency:	[80Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [921600] (serial upload only)

### Problem Description

While testing I came across a issue with WiFi.scanNetworks(1), async = true.  If you run with this option set it returns:
```
scan start
scan done
-1 networks found
```
If you run it with just WiFi.scanNetworks() it returns the available networks just fine.

Testing was done using the example sketch only.  The only change was the options used for WiFi.scanNetworks();  Please not this is not a major issue for me but just thought I should report it.

### [MCVE] Sketch

```cpp

/*
    This sketch demonstrates how to scan WiFi networks.
    The API is almost the same as with the WiFi Shield library,
    the most obvious difference being the different file you need to include:
*/
#include "ESP8266WiFi.h"

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  Serial.println("Setup done");
}

void loop() {
  Serial.println("scan start");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks(1);
  Serial.println("scan done");
  if (n == 0) {
    Serial.println("no networks found");
  } else {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i) {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE) ? " " : "*");
      delay(10);
    }
  }
  Serial.println("");

  // Wait a bit before scanning again
  delay(5000);
}
```

Pablo2048, 2018-03-24:
IMO it's correct, because you set async mode so you have to wait until async scan (in the background) is done. Best example I've seen for async scan is here https://github.com/me-no-dev/ESPAsyncWebServer#scanning-for-available-wifi-networks

mjs513, 2018-03-24:
Thanks Pablo2048.  Didn't know that before.  Going to give the example a try and see what happens.

devyte, 2018-03-30:
@mjs513 did the example work? If so,  please close this.

mjs513, 2018-03-31:
Going to try it now but going to close anyway now that I understand the issue

closed 2018-03-31 by mjs513

DaeMonSxy, 2024-12-10:
Same issue here, I assume esp8266 async scan has some issues. simply said-not working
Found 0.

```debug("[wifi] Scanning...");

    // Perform an asynchronous network scan
    WiFi.scanNetworksAsync([](int numNetworks) {
        int numCredentials = sizeof(wifiCredentials) / sizeof(wifiCredentials[0]);

        for (int i = 0; i < numNetworks; ++i) {
            char ssid[32];
            WiFi.SSID(i).toCharArray(ssid, sizeof(ssid)); // Get SSID as char array
            int rssi = WiFi.RSSI(i);                      // Get RSSI

            for (int j = 0; j < numCredentials; ++j) {
                if (strcmp(ssid, wifiCredentials[j].ssid) == 0) {
                    wifiCredentials[j].rssi = rssi2quality(rssi); // Update the RSSI in the credentials array
                    break;                                        // Exit the inner loop once a match is found
                }
            }
        }```

Pablo2048, 2024-12-10:
It works, you are doing something wrong.