esp8266/Arduino #4559: time.c and sntp.c doubt

David8266, 2018-03-24:
Hi,
I am compiling a code on my ESP-12E to handle the heating of my house.
A fundamental variable is time.
I used the time.c function to get the time from NTP servers, as shown in the code below, and everything works properly. Every second the correct time and date is printed.
But I have a doubt about the method with which the time is maintained, and looking at the code contained in the library I honestly could not make it clear.
In practice I would like to know if every time the time parameter is requested it is always read by the ntp servers, or if there is an internal timer/ counter that add 1 seconds each time, leaving the update with the servers only when the ConfigTime function is called.
Thanks in advance for your answers.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <time.h>
#include <ESP8266WiFi.h>

unsigned int timezone = 1;
unsigned int dst = 1;
struct tm timeinfo;

void setup() {

  Serial.begin(115200);

    while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  configTime(timezone * 3600, dst * 3600, "1.it.pool.ntp.org", "2.it.pool.ntp.org", "3.it.pool.ntp.org");

  while (!time(nullptr)) {
    Serial.print(".");
    delay(1000);
  }
  Serial.println("");

  delay(10000);

  time_t now = time(nullptr);
  Serial.println(ctime(&now));
  
}

void loop() {
  
  time_t now = time(nullptr);
  gmtime_r(&now, &timeinfo);
  Serial.println(tm_hour(&now));
  delay(1000);
}

```

Deimos1994, 2018-03-24:
> In practice I would like to know if every time the time parameter is requested it is always read by the ntp servers, or if there is an internal timer/ counter that add 1 seconds each time, leaving the update with the servers only when the ConfigTime function is called.

Just turn your router off (or disconnect internet cable) and test it yourself.

David8266, 2018-03-24:
Yes I tried and it continues to work properly.
But looking in time library seems that there is an if condition that discriminate if ntp servers are available or not, and in this case continue to manually increase the time variable.
But I'm not so expert to read that code... for this reason I prefer to ask someone.

5chufti, 2018-03-24:
depends on whioch lwip version you use.
in general the first read is done with dhcp answer and following updates are done regularly.
you can read up on that topic in #1679 and issues referenced by/within this issue

d-a-v, 2018-03-26:
Once sntp is started (which is automatic with lwip2/STA and a dhcp server providing an NTP server address), NTP server is requested at start then every hour from inside lwIP/sntp.
Offline or not, Time is always "manually" updated, and synced when NTP data are received.

David8266, 2018-03-26:
I'm sorry for the improper thread.
However I thank you very much for the help.
I understood the function, but just one last question. Is it possible to change the ntp update interval?

Deimos1994, 2018-03-26:
@David8266 
As far as I understand by default it updates every hour.
You can change with this :
`#define SNTP_UPDATE_DELAY           3600000`
look here: https://github.com/esp8266/Arduino/blob/ae13809c8184300aab9e3f09ef23af23d936b7ee/tools/sdk/lwip/src/core/sntp.c
(I am also not an expert, but I think it is correct way)

d-a-v, 2018-03-26:
This delay is defined by ```SNTP_UPDATE_DELAY```.
It is currently not defined in our lwIP options, so it defaults to 3600000 (ms).
If you need to change it, 
* add your #define in ```tools/sdk/lwip/include/lwipopts.h``` for lwIP-v1.4
  then use the lwIP variant "v1.4 compile from sources"
* add it in ```./tools/sdk/lwip2/builder/glue-lwip/arduino/lwipopts.h``` for lwip-v2
  then recompile lwip2 with ```make install``` in tools/sdk/lwip2

devyte, 2018-03-30:
This is not an issue. Closing.

closed 2018-03-30 by devyte