esp8266/Arduino #4563: Stack overflow in basic sketch

martinius96, 2018-03-25:
Hello there, i am using NodeMCU v3 Lolin with Arduino core 1.3.0. I want to send basic data do webserver over HTTPS. I have sketch with DHT22 for humidity, temperature and BMP280 for temperature and atmo pressure. 

If i turn it only (offline) serial monitor reading, everything okay. Problem is when i use ESP8266wifi.h and WifiClientSecure.h. After code come to loop, it sign stack overflow and restart board (without any boot mode etc... only exception with stack overflow).

Now code is working only from global variables, setup to loop and there it restarts and send stack overflow.. In setup i am connecting to wifi. It is sucessful and then code come to loop, restart with stack overflow..

Any solutions?
Code: 
`#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h> //KNIZNICA pre HTTPS spojenia
#include "Adafruit_BMP280.h"         //bmp280 kniznica s upravou na 0x76 adresu
#include "DHT.h"
#define DHTPIN 2     // what digital pin we're connected to
#define DHTTYPE DHT22   // DHT 22  (AM2302), AM2321
DHT dht(DHTPIN, DHTTYPE);
Adafruit_BMP280 bmp; //inicializacia BMP senzora
const char * ssid = "moj-xxxxxxxx-2929"; //meno wifi siete
const char * password = "xxxx"; //Heslo na wifi siet
const char * host = "sadsda.xxx.sk"; //bez https a www
const int httpsPort = 443; //https port zabezpeceny prenos
const char * fingerprint = "‎13 9f 87 1d b1 85 be e6 bd 73 c1 8d 04 63 58 99 f0 32 43 92"; //odtlacok HTTPS certifikatu v SHA1 formate
void setup() {          
  bmp.begin();        //start snimaca BMP
      dht.begin();              
  Serial.begin(9600);    //SPUSTENIE SERIOVEJ LINKY --UART-- NA CITACIU RYCHLOST 9600
  while (!Serial) {
    ;                                        
  }
 
 WiFi.begin(ssid, password); //pripoj sa na wifi siet s heslom
  while (WiFi.status() != WL_CONNECTED) { //pokial sa nepripojime na wifi opakuj pripajanie a spustaj funkcie pre ovladanie v offline rezime
    delay(500);
    Serial.println(".");
  }
  Serial.println("");
  Serial.println("WiFi pripojene"); //uspesne pripojenie na wifi siet
  Serial.println("IP adresa: ");
  Serial.println(WiFi.localIP()); // pridelena IP adresa pre dosku
 
}
 
void loop() {                                      //ZACIATOK SLUCKY
  delay(5000);
 if (WiFi.status() != WL_CONNECTED) {
odoslat_data();  
    WiFi.begin(ssid, password);
  } else {
    odoslat_data(); 
  }
  }
  
void odoslat_data(){
   WiFiClientSecure client;
  if (client.verify(fingerprint, host)) {} else {}
  if (client.connect(host, httpsPort)) {
    String teplota1 = String(dht.readTemperature());
    String teplota2 = String(bmp.readTemperature());
    String vlhkost = String(dht.readHumidity());
    String tlak = String((bmp.readPressure() / 100) + 103,855);
 
    String url = "/podstranky/system/nodemcu/zapishodnoty.php?teplota1=" + teplota1 + "&teplota2=" + teplota2 + "&vlhkost=" + vlhkost + "&tlak=" + tlak; //--------------------------------------------------------------------------------DOPLN LINK
    client.print(String("GET ") + url + " HTTP/1.1\r\n" + "Host: " + host + "\r\n" + "User-Agent: NodeMCU\r\n" + "Connection: close\r\n\r\n");
    Serial.println("Odoslane teploty do db:");
     Serial.println("Atmosfericky tlak: ");           //SERIOVY VYPIS TEXT O TLAKU VZDUCHU
    Serial.println((bmp.readPressure() / 100) + 103,855); //SERIOVY VYPIS STAVU RELATIVNEHO TLAKU 30,... je konstanta pre nadmorsku vysku, ktora sa prirata k teplote. (Použite pri nadmorskej do 1000m nadmorska vyska/8,3 tuto hodnotu napiste namiesto 30,...)
    Serial.println("Teplota: "); 
    Serial.println(bmp.readTemperature());
    Serial.println("Vlhkost: "); 
    Serial.println(dht.readHumidity());
        Serial.println("Teplota DHT: "); 
    Serial.println(dht.readTemperature());    
  } else if (!client.connect(host, httpsPort)) {
    Serial.println("Neuspesne pripojenie pre odoslanie teplot do DB - offline rezim aktivny");
  }
  }`

ERROR in serial monitor:
`WiFi pripojene
IP adresa: 
192.168.2.24
 
Exception (0):
epc1=0x30303030 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
 
ctx: cont 
sp: 3fff0df0 end: 3fff1040 offset: 01a0
 
>>>stack>>>
3fff0f90:  31333032 30303532 30303030 30303030  
3fff0fa0:  30303030 30303030 30303030 30303030  
3fff0fb0:  30303030 30303030 30303030 3fff0000  
3fff0fc0:  3fffdad0 0000000c 3ffeff5c 40204fe9  
3fff0fd0:  40105ec0 002c07f7 3ffefe98 00000000  
3fff0fe0:  3fff5584 0000005f 0000005c 3fff556c  
3fff0ff0:  0000000f 00000005 3fff5554 0000000f  
3fff1000:  00000005 3fff20c4 0000000f 00000005  
3fff1010:  3fffdad0 00000000 3ffefdd0 4020277c  
3fff1020:  feefeffe 00000000 3fff0010 4020596c  
3fff1030:  feefeffe feefeffe 3fff0020 40100718  
<<<stack<<<`

5chufti, 2018-03-25:
Did you check your post?
Do you think someone is going to have a look at that garbled mess?

martinius96, 2018-03-25:
Formatting code on github isn't my problem.. I solved problem... 
There is problem with:  String tlak = String((bmp.readPressure() / 100) + 103,855); 
There is coma, not dot... That caused this problem. Closed.

closed 2018-03-25 by martinius96