esp8266/Arduino #4569: WiFiClientSecure.connect crash after several POST requests

telliottosceola, 2018-03-26:
I have an application where post requests are made to an HTTPS server repeatedly.  After several requests, generally 100-200 or so my application crashes.  I have debugging enabled for SSL so I can see what is going on there.  It seems to crash right after Server Hello Done(14) every time.  I have also Decoded the exception.  Here is my code which I have simplified as much as possible:

```
#include <WiFiServerSecure.h>
#include <WiFiClientSecure.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WiFiUdp.h>
#include <ESP8266WiFiType.h>
#include <ESP8266WiFiAP.h>
#include <WiFiClient.h>
#include <WiFiServer.h>
#include <ESP8266WiFiScan.h>
#include <ESP8266WiFiGeneric.h>
#include <ESP8266WiFiSTA.h>

unsigned long wifiConnectTimeout = 5000;
WiFiClientSecure client;

const char* host = "univagohealthcare.com";
int httpsPort = 443;
const char* fingerprint = "D9 B3 32 51 E5 2A 45 A7 1F 1D 93 E8 07 75 DF 7A 04 63 6E E6";

char* httpRequestContent = "POST /api/circuit_reboot_check HTTP/1.0\r\n"
    "User-Agent: MatrixSSL/ 3.7.2-OPEN \r\n"
    "Host: univagohealthcare.com\r\n"
    "Accept: */*\r\n"
    "Content-Type: text/html\r\n"
    "Content-Length: 132\r\n\r\n{\"event_type\":\"INTERVAL\",\"event_reason\":\"Check In\",\"device_id\":\"e5:a1:36:43:4f:e0\",\"ssid\":\"Travis-WiFi\",\"input\":\"open\",\"interval\":5}";

int requests = 0;

void setup() {
  // put your setup code here, to run once:
  WiFi.begin("Travis-WiFi", "2Washington99");
  Serial.begin(115200);
  Serial.print("Connecting");
  
}

void loop() {

  while(WiFi.status() != WL_CONNECTED){
    Serial.print('.');
    delay(500);
  }
  Serial.print("Requests: ");
  Serial.println(requests);
  
  // put your main code here, to run repeatedly:
  if(WiFi.status() == WL_CONNECTED){
    Serial.println("Sending request");
    if(!client.connect(host, httpsPort)){
      Serial.println("server connection failed");
      delay(1000);
      return;
    }
    if(!client.verify(fingerprint, host)){
      Serial.println("Cannot verify Server");
      delay(1000);
      return;
    }
    client.print(httpRequestContent);
    while(client.connected()){
      if(client.available()){
        String line = client.readStringUntil('\n');
        Serial.println(line);
      }
    }
    requests++;
[... 282 lines cut ...]
3fff04f0:  3fff1c9c 3fff5a8c 402049e0 3ffef5a0  
3fff0500:  00003a98 00003a98 00003a98 4020136f  
3fff0510:  3ffef304 00000000 3fff9984 3ffef568  
3fff0520:  000001bb 3ffe8b8e 3ffef304 40203e07  
3fff0530:  000001bb 3ffef304 3ffe8b8e 3ffef568  
3fff0540:  000001bb 3ffef304 3ffe8b8e 40203ead  
3fff0550:  40107478 6200bcc0 40107478 6200bcc0  
3fff0560:  3ffef300 3ffef324 3ffef568 402021e9  
3fff0570:  00000000 00000000 00000000 feefeffe  
3fff0580:  3ffef304 0000001c 3ffef568 402041b0  
3fff0590:  00000000 00000000 00000001 3ffef594  
3fff05a0:  3fffdad0 00000000 3ffef58c 40204a2c  
3fff05b0:  feefeffe feefeffe 3ffef5a0 40100a2c  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

Any insight into the possible causes of these crashes would be greatly appreciated.  I just cannot imagine what is causing these crashes.

debsahu, 2018-03-31:
I see something like this as well. A few WiFiClientSecure request go through but 3rd or 4th one I see the same error as you. If I convert the server to not use SSL, WiFiClient sends same data without issues. Usually you will be asked to enable "SSL" debugging in Arduino IDE and post debugging results here as well.

jo-russ, 2018-04-01:
Same here. 
There's somewhere a memory leak in connection handling routines. 
Tested all boards > 2.0.0 It works correctly only in v2.2.0 though, so consider downgrading until this is solved.

telliottosceola, 2018-04-02:
Hi @jo-russ,

Which library are you referring to?  WiFiClientSecure?  I show my installed version of that library is 1.0.0 in the Arduino IDE.

@debsahu,
I had SSL debugging enabled and have posted the output above at the time of the crash.  

Thanks everyone.

jo-russ, 2018-04-02:
@telliottosceola 
Not the library itself, the esp8266 arduino board / bundle. 
With version 2.2.0 my code is running flawlessly for days now (free heap size stabilized) while on 2.3.x and 2.4.x it gets eaten away and in the end crashes.
IIRC with version 2.1.x it managed to make 2 consecutive (and that's probably why it worked) connections then for any other attempt - failure.

telliottosceola, 2018-04-11:
Hi @jo-russ 
Just tried running on 2.2.  My code posted above instantly fails on client.connect. Any ideas?  I'll post the debug logs soon.  Do you see anything in the code above that would cause that?

jo-russ, 2018-04-12:
@telliottosceola Can't see anything out of ordinary. The other difference I see from what I have is "Connection: close" header in request and httpsPort as "const int"
I'd try to remove all unnecessary included libraries and see if this helps. Does the httpsclient example works for you?
Check up on memory available, try disabling ssl debugging etc...

bhaskar-anil429, 2018-04-12:
@earlephilhower tried the new latest git version as well, still its crashing after several requests.
@comino thank you for looking into this. let us now if you find any thing.

earlephilhower, 2018-04-12:
(looks like I posted this in the PR and not the issue before, reposting)

I'm not seeing any leak per connection using a modified version of the script posted at the top here (to point to my public web server, and do GET instead of POST, but nothing materially different). I'm running with debug level "None", not even the larger mem free option of NoAssert+NDEBUG///
````
#include <WiFiServerSecure.h>
#include <WiFiClientSecure.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WiFiUdp.h>
#include <ESP8266WiFiType.h>
#include <ESP8266WiFiAP.h>
#include <WiFiClient.h>
#include <WiFiServer.h>
#include <ESP8266WiFiScan.h>
#include <ESP8266WiFiGeneric.h>
#include <ESP8266WiFiSTA.h>

unsigned long wifiConnectTimeout = 5000;
WiFiClientSecure client;

const char* host = "ziplabel.com";
int httpsPort = 443;
const char* fingerprint = "37 00 53 F1 10 64 F2 41 99 E6 84 07 C5 71 3F 3D 53 25 91 1E";

char* httpRequestContent = "GET /index.html HTTP/1.0\r\n"
    "User-Agent: MatrixSSL/ 3.7.2-OPEN \r\n"
    "Host: ziplabel.com\r\n"
    "Accept: */*\r\n"
    "\r\n";

int requests = 0;

void setup() {
  // put your setup code here, to run once:
  WiFi.begin("*", "*");
  Serial.begin(115200);
  Serial.print("Connecting");

}

void loop() {
  Serial.printf("Free: %d\n", ESP.getFreeHeap());
  while(WiFi.status() != WL_CONNECTED){
    Serial.print('.');
    delay(500);
  }
  Serial.print("Requests: ");
  Serial.println(requests);

  // put your main code here, to run repeatedly:
  if(WiFi.status() == WL_CONNECTED){
    Serial.println("Sending request");
    if(!client.connect(host, httpsPort)){
      Serial.println("server connection failed");
      delay(1000);
      return;
    }
    if(!client.verify(fingerprint, host)){
      Serial.println("Cannot verify Server");
      delay(1000);
      return;
    }
    client.print(httpRequestContent);
    while(client.connected()){
      if(client.available()){
        String line = client.readStringUntil('\n');
        Serial.println(line);
      }
    }
    requests++;
  }
}
````
Mem free stabilized at 31056 bytes after 2 or 3 cycles and is now at ~400 loops without any change.
````
</BODY>
</HTML>
Free: 31056
Requests: 164
Sending request
HTTP/1.1 200 OK
Date: Thu, 12 Apr 2018 15:25:04 GMT
Server: Apache/2.4.6 (CentOS) OpenSSL/1.0.2k-fips PHP/5.4.16
Last-Modified: Tue, 01 
````

bhaskar-anil429, 2018-04-12:
it just crashed after 288 request. 
i am going to run a new instance and will print the memory info.
 so fat it made 8 requests
started with free -- 21992
memory after 8 requests -- 21600
memory after 25 requests -- 20600

`{"result":"on"}
Free: 21992
Requests: 1
Sending request
HTTP/1.1 200 OK

Content-Type: application/json; charset=utf-8

Connection: close

Status: 200 OK

Cache-Control: no-cache, no-store

Strict-Transport-Security: max-age=15552000; includeSubDomains

Pragma: no-cache

X-XSS-Protection: 1; mode=block

X-Request-Id: f5999b09-d58b-4a9f-b364-1e9de325ceeb

X-Runtime: 0.005893

X-Frame-Options: SAMEORIGIN

X-Content-Type-Options: nosniff

Expires: Fri, 01 Jan 1990 00:00:00 GMT

Date: Thu, 12 Apr 2018 15:52:36 GMT

X-Powered-By: Phusion Passenger 5.1.4

Server: nginx/1.10.3 + Phusion Passenger 5.1.4

`

`{"result":"on"}
Free: 20600
Requests: 25
Sending request
HTTP/1.1 200 OK

Content-Type: application/json; charset=utf-8

Connection: close

Status: 200 OK

Cache-Control: no-cache, no-store

Strict-Transport-Security: max-age=15552000; includeSubDomains

Pragma: no-cache

X-XSS-Protection: 1; mode=block

X-Request-Id: dff3c8ca-0f61-41fc-b384-de8a4c2935d2

X-Runtime: 0.007524

X-Frame-Options: SAMEORIGIN

X-Content-Type-Options: nosniff

Expires: Fri, 01 Jan 1990 00:00:00 GMT

Date: Thu, 12 Apr 2018 15:58:57 GMT

X-Powered-By: Phusion Passenger 5.1.4

Server: nginx/1.10.3 + Phusion Passenger 5.1.4`

telliottosceola, 2018-04-12:
I'm running the same test here as well.  I slightly modified the code so the http response does not print to the serial monitor(just commented out Serial.println(line);
I also added the print for getFreeHeap().  I appear to be loosing exactly 56 bytes per request.  It is very consistent.  Here is the current version of code I am running followed by my serial monitor output thus far.

```
#include <WiFiClientSecure.h>
#include <ESP8266WiFi.h>

unsigned long wifiConnectTimeout = 5000;
WiFiClientSecure client;

const char* host = "univagohealthcare.com";
int httpsPort = 443;
const char* fingerprint = "CE 01 11 1A A1 2E 18 65 F5 9D 65 3B C3 3B A0 73 7F 79 85 B9";

char* httpRequestContent = "POST /api/circuit_reboot_check HTTP/1.0\r\n"
    "User-Agent: MatrixSSL/ 3.7.2-OPEN \r\n"
    "Host: univagohealthcare.com\r\n"
    "Accept: */*\r\n"
    "Content-Type: text/html\r\n"
    "Content-Length: 132\r\n\r\n{\"event_type\":\"INTERVAL\",\"event_reason\":\"Check In\",\"device_id\":\"e5:a1:36:43:4f:e0\",\"ssid\":\"Travis-WiFi\",\"input\":\"open\",\"interval\":5}";

int requests = 0;

void setup() {
  // put your setup code here, to run once:
  WiFi.begin("Travis-WiFi", "2Washington99");
  Serial.begin(115200);
  Serial.print("Connecting");
  
}

void loop() {
  Serial.printf("Free: %d\n", ESP.getFreeHeap());

  while(WiFi.status() != WL_CONNECTED){
    Serial.print('.');
    delay(500);
  }
  Serial.print("Requests: ");
  Serial.println(requests);
  
  // put your main code here, to run repeatedly:
  if(WiFi.status() == WL_CONNECTED){
    Serial.println("Sending request");
    if(!client.connect(host, httpsPort)){
      Serial.println("server connection failed");
      delay(1000);
      return;
    }
    if(!client.verify(fingerprint, host)){
      Serial.println("Cannot verify Server");
      delay(1000);
      return;
    }
    client.print(httpRequestContent);
    while(client.connected()){
      if(client.available()){
        String line = client.readStringUntil('\n');
        //Serial.println(line);
      }
    }
    requests++;
  }
}
```
Serial monitor log thus far

```
Requests: 0
Sending request
Free: 22040
Requests: 1
Sending request
Free: 21984
Requests: 2
Sending request
Free: 21928
Requests: 3
Sending request
Free: 21872
Requests: 4
Sending request
Free: 21816
Requests: 5
Sending request
Free: 21760
Requests: 6
Sending request
Free: 21704
Requests: 7
Sending request
Free: 21648
Requests: 8
Sending request
Free: 21592
Requests: 9
Sending request
Free: 21536
Requests: 10
Sending request
Free: 21480
Requests: 11
Sending request
Free: 21424
Requests: 12
Sending request
Free: 21368
Requests: 13
Sending request
Free: 21312
Requests: 14
Sending request
Free: 21256
Requests: 15
Sending request
Free: 21200
Requests: 16
Sending request
Free: 21144
Requests: 17
Sending request
Free: 21088
Requests: 18
Sending request
Free: 21032
Requests: 19
Sending request
Free: 20976
Requests: 20
Sending request
Free: 20920
Requests: 21
Sending request
Free: 20864
Requests: 22
Sending request
Free: 20808
Requests: 23
Sending request
Free: 20752
Requests: 24
Sending request
```

telliottosceola, 2018-04-12:
Also @earlephilhower I want to say thank you very much for your input thus far.  You are more than welcome to run a test using the exact code I have posted above so it makes post requests to our server.  This way you can run a test identical to ours.  Only difference would be your hardware.

earlephilhower, 2018-04-12:
Your first 2 lines say what's happening:
````
Requests: 0
Sending request
Free: 22040
````

So the `GET` is well behaved and closes the connection, so the next loop the memory is free (hence 30KB avail).  Seems like the `PUT` isn't closing the connection (probably no `Content-Length` in the reply and the server is trying to keep the link open with a `Connection: keep-alive` (hence 20KB avail).

For a quick fix I suggest moving the WiFiClientSecure object into the loop() context which should ensure it's closed at the end of a loop (or calling `client.stop()`).  The longer fix is to see what happens to HTTPS **and** HTTP connections when you call connect() on an object that already has a connection open.

telliottosceola, 2018-04-12:
Hi @earlephilhower 
Success!  I actually already added client.stop to the end of the loop after the server before you posted.  I had also added client.flush() as well although never in my experience has flush ever done anything for anyone.
The magic key was declaring the WiFiClientSecure object inside the loop.  I honestly have no idea how or why that would make a difference but I took your word for it and it works.  Here is my current code:
```
#include <WiFiClientSecure.h>
#include <ESP8266WiFi.h>

unsigned long wifiConnectTimeout = 5000;

const char* host = "univagohealthcare.com";
int httpsPort = 443;
const char* fingerprint = "CE 01 11 1A A1 2E 18 65 F5 9D 65 3B C3 3B A0 73 7F 79 85 B9";

char* httpRequestContent = "POST /api/circuit_reboot_check HTTP/1.0\r\n"
    "User-Agent: MatrixSSL/ 3.7.2-OPEN \r\n"
    "Host: univagohealthcare.com\r\n"
    "Accept: */*\r\n"
    "Content-Type: text/html\r\n"
    "Content-Length: 132\r\n\r\n{\"event_type\":\"INTERVAL\",\"event_reason\":\"Check In\",\"device_id\":\"e5:a1:36:43:4f:e0\",\"ssid\":\"Travis-WiFi\",\"input\":\"open\",\"interval\":5}";

int requests = 0;

void setup() {
  // put your setup code here, to run once:
  WiFi.begin("Travis-WiFi", "2Washington99");
  Serial.begin(115200);
  Serial.print("Connecting");
  
}

void loop() {
  WiFiClientSecure client;
  while(WiFi.status() != WL_CONNECTED){
    Serial.print('.');
    delay(500);
  }
  Serial.println();
  int startHeap = ESP.getFreeHeap();
  int heap = startHeap;
  Serial.printf("Free at start of request %i: %d\n", requests, ESP.getFreeHeap());
  
  Serial.printf("Free after WiFi check: %d, lost: %i\n", ESP.getFreeHeap(), heap - ESP.getFreeHeap());
  heap = ESP.getFreeHeap();
  
  // put your main code here, to run repeatedly:
  if(WiFi.status() == WL_CONNECTED){
    if(!client.connect(host, httpsPort)){
      Serial.println("server connection failed");
      delay(1000);
      return;
    }
    Serial.printf("Free after client.connect check: %d, lost: %i\n", ESP.getFreeHeap(), heap - ESP.getFreeHeap());
    heap = ESP.getFreeHeap();
    if(!client.verify(fingerprint, host)){
      Serial.println("Cannot verify Server");
      delay(1000);
      return;
    }
    Serial.printf("Free after client.verify check: %d, lost: %i\n", ESP.getFreeHeap(), heap - ESP.getFreeHeap());
    heap = ESP.getFreeHeap();
    client.print(httpRequestContent);
    Serial.printf("Free after client.print: %d, lost: %i\n", ESP.getFreeHeap(), heap - ESP.getFreeHeap());
    heap = ESP.getFreeHeap();
    int index = 0;
    while(client.connected()){
      if(client.available()){
        String line = client.readStringUntil('\n');
        Serial.printf("Free after read line %i: %d, lost: %i\n", index, ESP.getFreeHeap(), heap - ESP.getFreeHeap());
        heap = ESP.getFreeHeap();
        index++;
[... 478 lines cut ...]
Heap Memory lost: 20184 

Free at start of request 17: 41920
Free after WiFi check: 41920, lost: 0
Free after client.connect check: 21544, lost: 20376
Free after client.verify check: 21544, lost: 0
Free after client.print: 20872, lost: 672
Free after read line 0: 21136, lost: -264
Free after read line 1: 21400, lost: -264
Free after read line 2: 21528, lost: -128
Free after read line 3: 21544, lost: -16
Free after read line 4: 21512, lost: 32
Free after read line 5: 21496, lost: 16
Free after read line 6: 21528, lost: -32
Free after read line 7: 21512, lost: 16
Free after read line 8: 21496, lost: 16
Free after read line 9: 21528, lost: -32
Free after read line 10: 21528, lost: 0
Free after read line 11: 21512, lost: 16
Free after read line 12: 21512, lost: 0
Free after read line 13: 21512, lost: 0
Free after read line 14: 21512, lost: 0
Free after read line 15: 21512, lost: 0
Free after read line 16: 21544, lost: -32
```

telliottosceola, 2018-04-12:
Just cleaned up the code and removed everything that was not required.  Here is the minimal code that works successfully:

```
#include <WiFiClientSecure.h>
#include <ESP8266WiFi.h>

unsigned long wifiConnectTimeout = 5000;

const char* host = "univagohealthcare.com";
int httpsPort = 443;
const char* fingerprint = "CE 01 11 1A A1 2E 18 65 F5 9D 65 3B C3 3B A0 73 7F 79 85 B9";

char* httpRequestContent = "POST /api/circuit_reboot_check HTTP/1.0\r\n"
    "User-Agent: MatrixSSL/ 3.7.2-OPEN \r\n"
    "Host: univagohealthcare.com\r\n"
    "Accept: */*\r\n"
    "Content-Type: text/html\r\n"
    "Content-Length: 132\r\n\r\n{\"event_type\":\"INTERVAL\",\"event_reason\":\"Check In\",\"device_id\":\"e5:a1:36:43:4f:e0\",\"ssid\":\"Travis-WiFi\",\"input\":\"open\",\"interval\":5}";

int requests = 0;

void setup() {
  // put your setup code here, to run once:
  WiFi.begin("Travis-WiFi", "2Washington99");
  Serial.begin(115200);
  Serial.print("Connecting");
  
}

void loop() {
  WiFiClientSecure client;
  while(WiFi.status() != WL_CONNECTED){
    Serial.print('.');
    delay(500);
  }
  Serial.println();
  int startHeap = ESP.getFreeHeap();

  Serial.printf("Free at start of request %i: %d\n", requests, startHeap);
  
  // put your main code here, to run repeatedly:
  if(WiFi.status() == WL_CONNECTED){
    if(!client.connect(host, httpsPort)){
      Serial.println("server connection failed");
      delay(1000);
      return;
    }

    if(!client.verify(fingerprint, host)){
      Serial.println("Cannot verify Server");
      delay(1000);
      return;
    }

    client.print(httpRequestContent);

    while(client.connected()){
      if(client.available()){
        String line = client.readStringUntil('\n');
      }
    }
    requests++;
  }
  
  int finishHeap = ESP.getFreeHeap();
  Serial.printf("Free at end of request %i: %d\n", requests, finishHeap);
  Serial.printf("Heap Memory lost: %i \n", startHeap - finishHeap);
  Serial.println();
}
```

And the serial monitor log:

```
Connecting..
Free at start of request 0: 42224
Free at end of request 1: 22040
Heap Memory lost: 20184 

Free at start of request 1: 42224
Free at end of request 2: 22040
Heap Memory lost: 20184 

Free at start of request 2: 42224
Free at end of request 3: 22040
Heap Memory lost: 20184 

Free at start of request 3: 42224
Free at end of request 4: 22040
Heap Memory lost: 20184 

Free at start of request 4: 42224
Free at end of request 5: 22040
Heap Memory lost: 20184 

Free at start of request 5: 42224
Free at end of request 6: 22040
Heap Memory lost: 20184 

Free at start of request 6: 42224
```

telliottosceola, 2018-04-12:
@earlephilhower,

Well...  After letting the code run for a bit I found that I am loosing 48 bytes from heap memory "every once in a while".  It does not happen after a consistent number of requests it seems to be random at the moment.  In this test the first time memory was lost was after the first 32 requests, so the programmer in me quickly scrolled down to the 64th request expecting to find loss in memory there as well but no memory had been lost at that point.  The next time 48 bytes were lost from heap was after request 158.  Then there was a loss at 210, another at 225, another at 267, etc.

Now this is a vast improvement from before but if the module is loosing heap memory then eventually it WILL crash.

I just added client.stop at the end of the loop and will let that run now for testing.  So far it lost 48 bytes of memory after request number 5 so I am still seeing it.  Somehow 48 bytes are not being freed.  Other than declaring the WiFiClientSecure object in the loop so it leaves scope every cycle to free memory and calling client.stop at the end of the loop for good measure is there anything else you can think I should try?

If you want me to post most recent code update please let me know and I'll be happy to.

earlephilhower, 2018-04-12:
I'm on git head (well, git head as of 2-3 days ago) but I'm unable to make your original test app lose memory now, connecting to your server.  Can you verify that it fails consistently for you?  I see ~22KB (22136) free on each loop, with the # constant for loops 2-13.  All I did was update the fingerprint, which I guess changed since your original posting.

I was also unable to see a memory leak when I took the last code and moved the WiFiClientSecure out to globals and off the stack.  

There will be +/- a few bytes, unrelated to the WiFiClient*, simply due to LWIP and it's connection cache/received buffers.  If you can leave your app running and capture the free over a longer period (maybe let it try overnight, capture Serial output to a file, but at least 15mins or so) that would help to see if it's really lost or a temporary thing.

I also edited your last code sample to remove your WiFi password.

sumitkumarjaiswal, 2018-04-13:
Using 2.4.1 - Can any one help me what should i do, because WiFiClientSecure client
client.connect(host, 443) drops the stack memory and after some time it crashes.  see below logs. Kindly reply..

Before connecting to https url
--------------------------------------
Free heap:- 32304
unmodified stack   = 3168
current free stack = 3824

Just After 1st connection with https url 
--------------------------------------------
Free heap:- 18088
unmodified stack   = 1536
current free stack = 3824

After 2nd connection with https url
-------------------------------------------
Free heap:- 37152
unmodified stack   = 1488
current free stack = 3856

After 3rd connection with https url
----------------------------------------------
Free heap:- 37152
unmodified stack   = 1348
current free stack = 3808

Code to display head and stack
----------------------------------------
Serial.print ("Free heap:- ");
  Serial.println(ESP.getFreeHeap ());
  Serial.printf("unmodified stack   = %4d\n", cont_get_free_stack(&g_cont));
  register uint32_t *sp asm("a1");
  Serial.printf("current free stack = %4d\n", 4 * (sp - g_cont.stack));

telliottosceola, 2018-04-13:
@sumitkumarjaiswal 
Can you post your full code?

devyte, 2018-05-29:
BearSSL is merged in #4273 , with alternate BearSSL::WiFi* classes. Although axtls-based classes are still available and even the default, they are planned for deprecation and then retirement, hence won't be fixed. Any issues with BearSSL-based classes should be reported in new issues.
Finally, several issues have been fixed recently regarding connection stability.
Closing.

closed 2018-05-29 by devyte