esp8266/Arduino #4576: uart_esp8266-12E

7hanhphat, 2018-03-27:
I have a small problem about Serial (esp8266_12E), as follows:

Use one (ESP8266_12E) to communicate with two devices (a) and (b).
Serial_ (a) = 9600_baud.
Serial_ (b) = 57600_baud.
I want (esp8266_12E) to use a Serial for two devices (a), (b).

tttapa, 2018-03-27:
You can't use the RX pin of UART1 (GPIO8) because it's used in the flash chip SPI interface.

7hanhphat, 2018-03-27:
Thank you for the help. I want to use UART0 for two communication speeds, or use another solution.

tttapa, 2018-03-27:
This the issue section for the ESP8266 Arduino Core.  
It's meant to report bugs etc., not for help with your project.  

If you want general help or a discussion, please use the [Arduino forum](http://forum.arduino.cc/), [r/arduino](https://www.reddit.com/r/arduino/) or [r/esp8266](https://www.reddit.com/r/esp8266).  
If you have a specific question, try [Stack Exchange](https://arduino.stackexchange.com/).

devyte, 2018-03-27:
Closing due to non-compliance. See issue POLICY doc for more details.

closed 2018-03-27 by devyte