esp8266/Arduino #4579: Where can I find more information about timer1 and timer2?

AlfonsMittelmeyer, 2018-03-27:
It was told, that PWM and tone don't work at the same time. It was told also, that tone can only make sound for one pin. But I implemented an example for PWM and tone at the same time and tone with several channels in stereo. I could offer a library with functions for PWM and tone at the same time and tone with several channels. But another solution could be to do a very small change in the core libraries. PWM uses timer1 and tone uses timer1, which causes a problem. But why shouldn't timer2 be used for tone? My question is, what for is timer2 used, because it does exist seemingly. Ich found these registers:

```
//Timer 1 Registers (23bit CountDown Timer)
#define T1L  ESP8266_REG(0x600) //Load Value (Starting Value of Counter) 23bit (0-8388607)
#define T1V  ESP8266_REG(0x604) //(RO) Current Value
#define T1C  ESP8266_REG(0x608) //Control Register
#define T1I  ESP8266_REG(0x60C) //Interrupt Status Register (1bit) write to clear
//edge interrupt enable register
#define TEIE 	ESP8266_DREG(0x04)
#define TEIE1	0x02 //bit for timer 1

//Timer 2 Registers (32bit CountUp Timer)
#define T2L  ESP8266_REG(0x620) //Load Value (Starting Value of Counter)
#define T2V  ESP8266_REG(0x624) //(RO) Current Value
#define T2C  ESP8266_REG(0x628) //Control Register
#define T2I  ESP8266_REG(0x62C) //Interrupt Status Register (1bit) write to clear
#define T2A  ESP8266_REG(0x630) //Alarm Value
```
see: https://gitlab.com/alfredynho/AgroSis/blob/c57c4e89215b2dea50f59849f788a3e241b7211c/hardware/esp8266com/esp8266/cores/esp8266/esp8266_peri.h

But it wouldn't matter, if timer2would be reserved for internal functionalities und shouldn't be used for implementation purposes. My thought is, to implement virtual timers, which are based on only one hardware timer. So PWM could use the virtual timer timer1 and tone the vitual timer timer2. For doing so, I would like to get more information about the timers and the registers. Does somebody know, where I may find more information?

devyte, 2018-03-27:
@AlfonsMittelmeyer not really the place to ask this. However, in view of your work in #4349 , and givent that it's getting a bit crowded there, let's keep this issue open as a side discussion.
Have you seen the [Espressif Technical Reference](https://www.espressif.com/sites/default/files/documentation/esp8266-technical_reference_en.pdf)? the appendix has a small reference to the timer registers.

AlfonsMittelmeyer, 2018-03-27:
Many thanks. Yes, I had seen the technical reference, but had missed, that I would find some information in the appendix.

One question more. There are the interrupt types edge and level. I know, what this would mean for an IO pin. But what does this mean for a timer?

igrr, 2018-03-27:
Level and edge refer to the CPU interrupt source type. Edge interrupts cause the corresponding intstatus bit to be set, if the interrupt is not masked. This happens each time interrupt triggers. With an edge interrupt, software does not need to clear the interrupt status.
When a level interrupt gets triggered, the interrupt line from the peripheral to the CPU stays high until the software clears the interrupt status. CPU will keep setting corresponding bit of intstatus until this line goes low (unless interrupt is masked).

AlfonsMittelmeyer, 2018-04-11:
In ets_sys.h I saw, that TIM_EDGE shall be used:

```
#define ETS_SLC_INUM        1
#define ETS_SDIO_INUM       1
#define ETS_SPI_INUM        2
#define ETS_GPIO_INUM       4
#define ETS_UART_INUM       5
#define ETS_UART1_INUM      5
#define ETS_CCOMPARE0_INUM  6
#define ETS_SOFT_INUM       7
#define ETS_WDT_INUM        8
#define ETS_FRC_TIMER1_INUM 9  /* use edge*/
```

So, why it's allowed in core_esp8266_timer.c to set TIM_LEVEL and why is the interupt state cleared in timer1_isr_handler via T1I = 0? If somebody would wish to use TIM_LEVEL, wouldn't he also wish to clear the interrupt state himself? If the timer1_isr_handler clears it, then would there be any difference between edge and level (if level would work)?

And there is the edge interrupt disabled in core_esp8266_timer.c in case of TIM_EDGE and TIM_SINGLE:

`if ((T1C & ((1 << TCAR) | (1 << TCIT))) == 0) TEIE &= ~TEIE1;//edge int disable
`

End because of this edge has to be enabled in timer1_write:

`if ((T1C & (1 << TCIT)) == 0) TEIE |= TEIE1;//edge int enable`

Is there a reason to do so? If the edge interrupt wouldn't be disabled it wouldn't be needed to enable it, which saves time and some lines of code.

I enabled the edge interrupt only once and this was sufficient. Wouldn't it be sufficient?

devyte, 2018-12-02:
This discussion seems to have run its course. Closing.

closed 2018-12-02 by devyte