esp8266/Arduino #4581: TCP Communication problems in new versions

xhdix, 2018-03-27:
I don't know what made this happen. And what changes have you made In the new versions.
The problem is getting data from Android. (android code:  DOS.writeChars("commandSample") )
Each character transfer also adds a ```null``` character.

I solved this issue in my program.
In this way:

```cpp

  for(i = 0; i < MAX_SRV_CLIENTS; i++){
    if (serverClient[i] && serverClient[i].connected()){
      if(serverClient[i].available()){
        while(serverClient[i].available()) { 
          char inChar = (char)serverClient[i].read();

		  if (inChar != '\0') {        //  I added this line to solve the problem.. *************

			  tlinstr += inChar;
			  if (inChar == '\n') {
				tlntstrCmplt = true;
			  }

                 }
        }
      
      }
      
    }
  }
  
  

```

Maybe this solution will reduce debug time for somebody's. :)

mittaldivya01, 2018-04-01:
Hello, can you please guide on how can we connect to an IP address.. 
Thank you!!

devyte, 2018-07-13:
Issue template ignored, none of the required info provided, no MCVE sketch.
Closing.

closed 2018-07-13 by devyte