esp8266/Arduino #4583: The program is not running above ESP8266 community ver 2.3.0

Kartel0878, 2018-03-27:
Basic _Infos_

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:			[ESP-12 nodemcu v3]
- Core Version:			[I'm sorry, I do not know.]
- Development Env:		[Arduino IDE|]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Nodemcu|]
- Flash Mode:			[winbond 25q32jvsiq]
- Flash Size: 		        [Setup:4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:               [115200] (serial upload only)

### Problem Description
Sorry friends for my English. I use translator.
Additional libraries are used here:
#include <ESP8266WebServer.h>   rev. 1.0.0
#include <FS.h> file system. The full version of the code is located below the googl link
how to create the markup: https://github.com/esp8266/arduino-esp8266fs-plugin

I'm trying to transfer a number from a web page to esp8266.The page itself is loaded into the file system of the microcontroller.
This code works perfectly ESP8266 community ver 2.3.0 : https://drive.google.com/file/d/1sdbqcQnuWRPOcZFbHhFBEv1_lbu4WcxJ/view?usp=sharing
But does not want to work on version 2.4.0 and higher : https://drive.google.com/open?id=1-mSh-83beacVOaz9kY4vtMaLkgKhKqFF
Link to the entire project with the html page and js file system:
https://drive.google.com/open?id=1rGwss5GXduifTnqWZPJe-8HXDIJrQhyV

I'm sorry,but this is the minimum possible code:  
-------------------

Sketch:
```
#include <ESP8266WiFi.h>        
#include <ESP8266WebServer.h>   
#include <ESP8266SSDP.h>       
#include <FS.h> 

IPAddress apIP(192, 168, 4, 1);

// Web интерфейс для устройства
ESP8266WebServer HTTP(80);

File fsUploadFile;

String _ssid     = "Network_ProJect_EVA"; 
String _password = ""; 
String _ssidAP = "WiFi";   
String _passwordAP = ""; 

void setup() {
  Serial.begin(115200);
  Serial.println("");
  Serial.println("Start 4-FS");
  FS_init();
  Serial.println("Start 1-WIFI");
  //Запускаем WIFI
  WIFIinit();
  //Настраиваем и запускаем HTTP интерфейс
  Serial.println("Start 2-WebServer");
  HTTP_init();

}
void loop() {
  HTTP.handleClient();
  delay(1);
}
```
```
void HTTP_init(void) {
  	// start HTTP server
  	HTTP.begin();
  	HTTP.on("/test",testNumber);
  		

}
void testNumber(){
 	String num =HTTP.arg("state");
  int n = HTTP.arg("state").toInt(); 
 	Serial.println(num);
  Serial.println(n);
 }

```
==========================
page code ltml:
```
<!DOCTYPE html>
<html>
<head>
	<meta charset="utf-8"/>
	<title>FAROS</title>
	<link rel="stylesheet" type="text/css" href="style.css">
	<script src="/script.js"></script>
</head>
<body>	<!--head:-->
	<br> <br> 
	<div class="header">Pro</div><br>   
	<div class="header" id="ject">Ject</div>
	<div id="eva">EVA</div>
	<br>
<!--main:-->
	Тестовое число:
	<p> <input class="formtext" type="text" name="name" maxlength="3" id="testNumber"> </p> <br>
	<input name="bt" value="отправить" type="button" id="buttonout" onclick="butfun()">
</body>
</html>
```

code JavaScript:
```
function butfun() {
	var num = document.getElementById("testNumber").value;
	var request =new XMLHttpRequest();
	request.open("POST","/test",true);
	request.send('state=' + num);
	alert(num);
}

```
```

### Debug Messages

```
Debug messages go here
```

devyte, 2018-03-28:
@Kartel0878 The above code is not complete, there are function definitions missing, and asking to go into google drive to look at the entire project is expecting too much. The requirement is to provide an MCVE sketch, and at the very least your example is not complete.
In addition, in the template you said you don't know which core version, and then you talk about 2.3 and 2.4 and above. So which is it?
There are known issues with core 2.4.0 and WiFiServer/Client. I suggest trying 2.4.1, or better yet, latest git. The instructions for installing latest git are on readthedocs.
I'm closing this due to lack of info. If you think you can fill in the missing info and improve the example, go ahead and do so, and I'll consider reopening.

closed 2018-03-28 by devyte

Kartel0878, 2018-03-28:
@devyte  The code is very large and hard to read.On Google drive the code is more structured and there prepared a separate folder data with html page and js for loading into the file system. Sorry to confuse you, I corrected the text and added the code. The problem is that this code works fine with ESP8266 community ver 2.3.0 . I get a number from a web page.But in ESP8266 community ver 2.4.0 and 2.4.1, this code doesn't work. I want to note that in all versions I was able to take the number from the microcontroller on the web page,this part of the code I did not upload because it works. But I was able to transfer the number from the web page to the microcontroller only in version 2.3.0. I would have worked in ESP8266 community ver 2.3.0, but IT's very slow and less stable. Thank you for your patience.