esp8266/Arduino #4585: SPIFFS.end() does not free memory

petrkr, 2018-03-28:
When do SPIFFS.begin() it will consume some like 2kB of memory, but SPIFFS.end() will not release it. That is kinda problem for low-memory environment as for my application these 2kB is difference between SSL handshake can be done later or not.

Tested SDK: 2.3.0, 2.4.1 from Arduino IDE
Tested ESP_12E

Test code:
```
// Filesystem
#include "FS.h"

void printHeap() {
  Serial.print(F("Heap size: "));
  Serial.println(ESP.getFreeHeap());
}

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println();
  Serial.println("Setup begin");
  Serial.print(F("SDK: ")); Serial.println(ESP.getSdkVersion());
  Serial.print(F("Core: ")); Serial.println(ESP.getCoreVersion());
  
  printHeap();
  
  Serial.println("SPIFF BEGIN");
  SPIFFS.begin();
  printHeap();

  Serial.println("SPIFF END");
  SPIFFS.end();
  printHeap();

  Serial.println("Setup end");
}

void loop() {
  Serial.println("Loooop");
  Serial.println("SPIFF BEGIN");
  SPIFFS.begin();
  printHeap();
  
  delay(1000);
  
  Serial.println("SPIFF END");
  SPIFFS.end();
  printHeap();

}
```

Sample output:
```
Setup begin
SDK: 1.5.3(aec24ac9)
Core: 2_3_0
Heap size: 47072
SPIFF BEGIN
Heap size: 44960
SPIFF END
Heap size: 44960
Setup end
Loooop
SPIFF BEGIN
Heap size: 44960

Setup begin
SDK: 2.2.1(cfd48f3)
Core: 2_4_1
Heap size: 47160
SPIFF BEGIN
Heap size: 44984
SPIFF END
Heap size: 44984
Setup end
Loooop
SPIFF BEGIN
Heap size: 44984
SPIFF END
Heap size: 44296
Loooop
SPIFF BEGIN
Heap size: 44296
SPIFF END
Heap size: 44224
```

Eszartek, 2018-04-03:
It appears SPIFFS doesn't have the functionality to free the cache and buffers it allocates.
For a quick comparison, I disabled caching in SPIFFS and it only used 744 bytes for its structures, so that may be an option for you.

```
Loooop                                                                          
Heap size: 43056                                                                
SPIFF BEGIN                                                                     
Heap size: 42312                                                                
SPIFF END                                                                       
Heap size: 42312 
```
Maybe someone who understands the allocation code can write a few lines that free it all in SPIFFS.end. 

Since SPIFFS is an external project to this, you may want to inquire at https://github.com/pellepl/spiffs .

I also waited 30 seconds in the loop() before running the test, as internal wifi/OS bits tend allocate memory for the first 10 seconds at boot time. Its best to have the heap stable before measuring.

petrkr, 2018-04-03:
Can you please provide me how to disable cache? I tried set 
`#define SPIFFS_CACHE                    0`
in spiff_config.h, but I am not able to compile it after it, because some code appear to need it.

```
In file included from /home/petrkr/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/spiffs_api.cpp:24:0:
/home/petrkr/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/spiffs_api.h: In member function 'bool SPIFFSImpl::_tryMount()':
/home/petrkr/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/spiffs_api.h:219:78: error: 'SPIFFS_buffer_bytes_for_cache' was not declared in this scope
         size_t cacheBufSize = SPIFFS_buffer_bytes_for_cache(&_fs, _maxOpenFds);
                                                                              ^
```

SDK 2.3.1, becase 2.4.1 randomly disconnects from wifi after like 2 to 4 hours and can not be used for long term usage.

Eszartek, 2018-04-03:
I got that error as well, so I just changed that line to:

```
//size_t cacheBufSize = SPIFFS_buffer_bytes_for_cache(&_fs, _maxOpenFds);
size_t cacheBufSize = 0;
```
I can't be sure this change won't cause any functionality issues, but it appears to work for me so far.

petrkr, 2018-04-03:
Seems like it can not open file for read or write at all.. But for append yes, strange...

Eszartek, 2018-04-04:
For me the no-cache is working perfectly and I got a bunch of free heap as well. 

Unless one of the maintainers plan on implementing a cache/buffer freeing feature, I'd say SPIFFS is working as designed and that a pull request should be done by someone who can implement it. I'll look into doing this if no one else does, as I'm now interested in having SPIFFS clean it self up better for the use case you mention, as well as a properly functioning no-cache option.

Eszartek, 2018-04-05:
I setup a fork at https://github.com/Eszartek/Arduino to work on implementing the freeing of SPIFFS memory resources as well as a "no cache" option for those folks that are memory contrained.

PR #5651 (2019-01-25): Release buffer/cache on SPIFFS.end()
Releases the buffers allocated by the SPIFFS object when/if SPIFFS::end()
is called.

Fixes #4585
closed 2019-01-25 by earlephilhower