esp8266/Arduino #4586: WiFiClient connection failed

michaelmnich, 2018-03-29:
Hi i was trying to make one of yours tutorial:

http://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/client-examples.html

still Geting: 

[Connecting to http://localhost/ ... connection failed!]

[Connecting to http://localhost/ ... connection failed!]

[Connecting to http://localhost/ ... connection failed!]

[Connecting to http://localhost/ ... connection failed!]

I tried on 2 different network and the same situation.

Code:
`#include <ESP8266WiFi.h>

const char* ssid = "Ewaaa&Spolka";
const char* password = "$qazxsw123#cze";

const char* host = "http://localhost/";

void setup()
{
  Serial.begin(9600);
  Serial.println();

  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" connected");
}

void loop()
{
  WiFiClient client;

  Serial.printf("\n[Connecting to %s ... ", host);
  if (client.connect(host, 80))
  {
    Serial.println("connected]");

    Serial.println("[Sending a request]");
    client.print(String("GET /") + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Connection: close\r\n" +
                 "\r\n"
                );

    Serial.println("[Response:]");
    while (client.connected())
    {
      if (client.available())
      {
        String line = client.readStringUntil('\n');
        Serial.println(line);
      }
    }
    client.stop();
    Serial.println("\n[Disconnected]");
  }
  else
  {
    Serial.println("connection failed!]");
    client.stop();
  }
  delay(5000);
}`

devyte, 2018-03-29:
What exactly does "localhost" mean in your network?
The example you refer to shows "www.example.com", meaning you're supposed to replace that address with a real world address. Unless your local network happens to have a server called "localhost", your code obviously won't work.
Closing due to user error.

closed 2018-03-29 by devyte

michaelmnich, 2018-03-29:
What user error?

localhost mean that i have locally set up serwer.

devyte, 2018-03-29:
And do you have a dns server on your local network, where "localhost" is mapped to an IP address?

d-a-v, 2018-03-29:
@michaelmnich what you should know is that localhost is always "self-address" = 127.0.0.1 on whatever host it is referenced from, including the ESP. Meaning that for your ESP, localhost or 127.0.0.1 is itself. Your sketch is trying to reach itself on port 80 which you did not have opened.

You want/need to use the IP address or name of your PC (not localhost nor 127.0.0.1 on it but the other address usually 192.168.x.x) which is running the web server in order for the ESP to access to it.

michaelmnich, 2018-03-29:
Ty you are right localhost is occupied. I fix my self for that solution dont know why :P ill check your solution this evening  :) ty