esp8266/Arduino #4588: Exception (29): epc1=0x4020ce26 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

Sofrus, 2018-03-29:
#### Platform

- Hardware:				[Wemos D1]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Wemos D1 R1]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [57600]

### Problem Description

The Sketch working few minutes and then get Exception (29)

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
```
#include <ESP8266WiFi.h>
//#include <WiFiClientSecure.h>
#include <ESP8266TelegramBOT.h>
#include <Servo.h>

// Initialize Wifi connection to the router
char ssid[] = "BANKEX";              // your network SSID (name)
char password[] = "13941394";                              // your network key

// Initialize Telegram BOT
#define BOTtoken "499448646:AAGNEEAN2EbPgALj53ulE5dgiSIEw2MWmCc"  //token of TestBOT
#define BOTname "@BankexGatekeeperBot"
#define BOTusername "@BankexGatekeeperBot"
TelegramBOT bot(BOTtoken, BOTname, BOTusername);

int Bot_mtbs = 2000; //mean time between scan messages
long Bot_lasttime;   //last time messages' scan has been done

Servo myservo;  // create servo object to control a servo
// twelve servo objects can be created on most boards

void Bot_EchoMessages() {
  yield();
  for (int i = 1; i < bot.message[0][0].toInt() + 1; i++)      {
    delay(300);
    //bot.sendMessage(bot.message[i][4], bot.message[i][5], "");
    delay(300);
    if (bot.message[i][5] == "open")
      delay(300);

    {
      Rotate_Servo();   // rotate servo, blocking bot
    }
  {
      bot.sendMessage(bot.message[i][4], "Opening", "");
    }

  }

  bot.message[0][0] = "";   // All messages have been replied - reset new messages
}

void Rotate_Servo()
{
  myservo.write(150);
  delay(300);

  myservo.write(90);

  delay(500);

  //  int pos;
  //
  //  for (pos = 0; pos <= 87; pos += 1) // goes from 0 degrees to 90 degrees
  //  { // in steps of 1 degree
  //    myservo.write(pos);              // tell servo to go to position in variable 'pos'
  //    delay(15);                       // waits 15ms for the servo to reach the position
  //  }
  //  for (pos = 87; pos >= 0; pos -= 1)  // goes from 90 degrees to 0 degrees
  //  {
  //    myservo.write(pos);              // tell servo to go to position in variable 'pos'
  //    //delay(1);                       // waits 15ms for the servo to reach the position
  //  }
}

void setup() {
  yield();
  myservo.write(90);
  myservo.attach(2);  // attaches the servo on GIO2 to the servo object

  Serial.begin(57600);
  delay(3000);

  // attempt to connect to Wifi network:
  Serial.print("Connecting Wifi: ");
  Serial.println(ssid);
  while (WiFi.begin(ssid, password) != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  IPAddress ip = WiFi.localIP();
  Serial.println(ip);
  //delay(1);
  bot.begin();
  delay(1);
  // launch Bot functionalities
}

void loop() {
  // yield();
  if (millis() > Bot_lasttime + Bot_mtbs)  {
    // delay(1);
    // bot.getUpdates("–1");

    // delay(1);
    bot.getUpdates(bot.message[0][1]);   // launch API GetUpdates up to xxx message
    Bot_EchoMessages();   // reply to message with Echo
    Bot_lasttime = millis();

  }
}
```
### Debug Messages

```
Decoding stack results
0x40219c9e: more_comps at crypto/bigint.c line 1072
0x40219e98: alloc at crypto/bigint.c line 1106
0x40100670: realloc at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/umm_malloc/umm_malloc.c line 1641
0x4021a1b1: regular_multiply at crypto/bigint.c line 822
0x4020546c: optimistic_yield(uint32_t) at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 111
0x4021aceb: bi_barrett at crypto/bigint.c line 1289
0x4021af40: bi_mod_power at crypto/bigint.c line 1414
0x40219d44: trim at crypto/bigint.c line 1197
0x4021c0c9: RSA_encrypt at crypto/rsa.c line 254
0x40216dc4: send_client_key_xchg at ssl/tls1_clnt.c line 409
0x40217259: do_clnt_handshake at ssl/tls1_clnt.c line 123
0x402144c8: mem_free at core/mem.c line 152
0x40210031: pbuf_free_LWIP2 at core/pbuf.c line 777
0x4020358c: ClientContext::read(char*, unsigned int) at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src/include/ClientContext.h line 494
0x4020542c: __yield() at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 96
0x40216bcc: basic_read at ssl/tls1.c line 1607
0x40216fa8: do_client_connect at ssl/tls1_clnt.c line 168
0x401004d8: malloc at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/umm_malloc/umm_malloc.c line 1668
0x402033e0: SSLContext::_delete_shared_SSL(SSL_*) at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 115
0x40216d34: ssl_read at ssl/tls1.c line 314
0x40203bc1: SSLContext::connect(ClientContext*, char const*, unsigned int) at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 146
0x402053b4: esp_yield() at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 87
0x40201413: delay at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_wiring.c line 51
0x40203e0f: WiFiClientSecure::_connectSSL(char const*) at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 350
0x40203e64: WiFiClientSecure::connect(IPAddress, unsigned short) at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 506
0x402043d7: TelegramBOT::connectToTelegram(String) at /root/Arduino/libraries/ESP8266-TelegramBot-master/src/ESP8266TelegramBOT.cpp line 53
0x40204f47: String::reserve(unsigned int) at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/WString.cpp line 141
0x40205038: String::operator=(String const&) at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/WString.cpp line 215
0x402044fd: TelegramBOT::getUpdates(String) at /root/Arduino/libraries/ESP8266-TelegramBot-master/src/ESP8266TelegramBOT.cpp line 93
0x40204efb: String::changeBuffer(unsigned int) at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/WString.cpp line 151
0x40204f79: String::copy(char const*, unsigned int) at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/WString.cpp line 170
0x40205038: String::operator=(String const&) at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/WString.cpp line 215
0x4020141e: delay at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_wiring.c line 54
0x4020239e: loop() at /root/Arduino/sketch_mar29a/sketch_mar29a.ino line 108
0x402053d5: esp_schedule() at /root/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 91

```

Eszartek, 2018-04-01:
https://github.com/Gianbacchio/ESP8266-TelegramBot/issues may be a more appropriate place to report your issue. Looks like others there are having similar issues to you. Your MCVE Sketch is really a someone elses full project (that's buggy), not a minimal sketch that shows an example of a bug in the esp/arduino environment.

devyte, 2018-04-04:
The author of the telegram bot should investigate this first, or someone on his behalf. If an issue is found in the core code, then thst should be reported here with an MCVE sketch.
Closing due to third party lib.

closed 2018-04-04 by devyte

d-a-v, 2018-04-04:
To help debugging, you should activate the OOM debug option.
According to your stacktrace, you are running out of memory and WiFiClientSecure currently does not cope well with low memory.