esp8266/Arduino #4591: iOS does not see HAP service

levidhuyvetter, 2018-03-30:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-01]
- Core Version:			[1.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[512K]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I'm trying to setup an HAP (HomeKit Accessorie Protocol) service with mDNS to be discovered by HomeKit on iOS devices. However in normal circumstances I can't see the device in the Home app.

I have tried the things below with varying results:

- **Test:** Downloaded a Bonjour discovery app on my iPhone called "Radar".
- **Result:** When the mDNS service starts the app will show that there is some kind of HAP service available but when I click on it (which shows devices offering that service) I see nothing. It seems the iPhone knows there is a service but doesn't know which devices are offering it.

- **Test:** Changed the service offered from "hap" to something else and tested with the app above.
- **Result:** The service and the device both show up in the app so that is all working fine. It seems to be only a problem with "hap". (When I use "homekit" as the service it gives the same result as "hap".

- **Test:** Used "dns-sd" on a mac to try and discover the HAP service.
- **Result:** This works just fine and the service + device are discovered. After doing this iOS is also able to discover the device in both the "Radar" app and in the "Home" app. It seems the iOS device is somehow also receiving the mDNS response that was sent back to the home app.

- **Test:** Following up on the above, I programmed another esp8266-01 to continiously query the HAP service (as with dns-sd).
- **Result:** Everything works as expected but this is not really a solution as I would like to have the job done by only 1 esp module. This also makes me think that the library is not advertising the service properly or at least not to the expectations of the iOS device.

- **Test:** Enabled debug messages.
- **Result:** When using dns-sd on the mac, the requests come through correctly and are also responded to. When using the Radar app clicking (which sends a request) on "hap" I do not see any requests coming through, when clicking on any other service I see an ERR_NO_SERVICE (which is expected but shows the requests are coming through). Using the Home app and letting it search for devices does not give me any requests.

- **Test:** Used wireshark to monitor mDNS packets on the networks.
- **Result:** In normal circumstances the iOS devices does not seem to make any requests for HAP service when using either the Radar or Home app. This only starts happening after I "poke" a request from somewhere else such as dns-sd on the Mac.

- **Test:** Let the esp query it's own service after it has been set up.
- **Result:** This doesn't make a difference.

I have been spending 2 days trying to figure this out and I still haven't found a solution. From the above my guess is that somehow the mDNS library has to advertise this service in a slightly different way from everything else for the iOS device to discover it. Perhaps the service is not registered somewhere were it should and the iOS device is waiting for this to start sending it's requests. (This would then be fixed when making a request from dns-sd).

I really have no clue where to go from here and your help to fix this issue is appreciated!

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char* ssid = "WiFi";
const char* password = "xxxxx";

ESP8266WebServer server(80);

void handleNotFound(){
  server.send(404, "text/plain", "Not Found!");
}

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.hostname("ESP8266");
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  if (MDNS.begin("ESP8266")) {
    Serial.println("MDNS Started.");
  }
  server.onNotFound(handleNotFound);

  server.begin();

  MDNS.addService("hap", "tcp", 80);
  //TXT records go here and they are correct, left out for simplicity.
}

void loop() {
  server.handleClient();
}

```

### Debug Messages

```
SDK:2.2.1(cfd48f3)/Core:2.4.1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
scandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 

connected with WiFi, channel 11
dhcp client start...
....ip:10.0.1.116,mask:255.255.255.0,gw:10.0.1.1
.MDNS listening
MDNS Started.
ERR_NO_SERVICE: homekit
ERR_NO_SERVICE: homekit
ERR_NO_SERVICE: companion-link
ERR_NO_SERVICE: homekit
pm open,type:2 0
TX: service:hap, proto:tcp
Reading answers RX: REQ, ID:0, Q:0, A:17, NS:0, ADD:0
Not expecting any answers right now, returning
Reading answers RX: REQ, ID:0, Q:0, A:25, NS:0, ADD:0
Not expecting any answers right now, returning
ERR_NO_SERVICE: homekit
TX: service:hap, proto:tcp <----- Here I was sending a request for hap from the Radar app.
TX: service:hap, proto:tcp <----- Those messages seem to be random though and do not line
TX: service:hap, proto:tcp <----- up with me sending requests at all.
TX: service:hap, proto:tcp
TX: service:hap, proto:tcp
TX: service:hap, proto:tcp
TX: service:hap, proto:tcp
ERR_NO_SERVICE: http <------ Here I was sending a request for http from the Radar app.
ERR_NO_SERVICE: http
ERR_NO_SERVICE: homekit
ERR_NO_SERVICE: http
TX: service:hap, proto:tcp
Reading answers RX: REQ, ID:0, Q:0, A:1, NS:0, ADD:0
Not expecting any answers right now, returning
TX: service:hap, proto:tcp
```

levidhuyvetter, 2018-03-31:
I have don some further testing and gone through the code step by step and it seems that the problem is nothing to do with this library. I am closing this issue but for anyone having the same problem I have asked the [question on StackOverflow](https://stackoverflow.com/questions/49590546/how-does-homekit-find-nearby-devices).

levidhuyvetter, 2018-03-31:
I have done some more testing and installed Homebridge on a Mac and monitored it's mDNS packets using WireShark. On startup of the server it sends out a couple of "ANY" queries and a couple of responses. What it is doing here is as described in [section 8 of RFC 6762](https://tools.ietf.org/html/rfc6762) namely, "Probing and Announcing on Startup". I am hoping that any contributors will be able to implement this as ESP8266mDNS currently does not send out any mDNS packets on startup. I will attempt to implement this on my side and maybe make a pull request if I get things to work.

devyte, 2018-03-31:
Be aware that I am currently rewriting the mdns responder/query based on the idf implementation. It is very slow going, though, so it will take quite some time.

levidhuyvetter, 2018-03-31:
Thanks for the heads up! Does the idf implementation incorporate probing and announcing? In case it does I can close this issue as it will be already accounted for.

devyte, 2018-03-31:
The idf implementation is supposed to be complete enough to pass Apple's conformance tests, and I believe that is needed for HAP support, among other things.

levidhuyvetter, 2018-04-01:
Indeed, I couldn't get this to pass the conformance test. It failed before I could even get any useful info. Hoping this will be finished soon, in the meantime I'll have to figure out a way to get it to work temporarily. I'll be closing this issue for now as this will not be relevant in the new version. Thanks for your help!

closed 2018-04-01 by levidhuyvetter