esp8266/Arduino #4592: Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json

juztinfied, 2018-03-31:
Does anyone have any issues with the board manager for WE MOS D1 R2 ESP8266? The board now does not even appear on my IDE list of arduino boards

devyte, 2018-04-01:
Works for me.
Closing as can't reproduce.

closed 2018-04-01 by devyte

zcattacz, 2018-04-22:
Can't Open it Now.

devyte, 2018-04-22:
Still works for me.

zcattacz, 2018-04-22:
HTTP error code 416.   Arduino IDE's download manager is to blame. 

https://github.com/arduino/Arduino/issues/6628

MuhammadShoaibJahangir, 2019-07-19:
The issue with the boards disappearing is definitely a bug with the Arduino IDE. You can get the boards back by:
https://github.com/esp8266/Arduino/releases/download/2.3.0/package_esp8266com_index.json

to File > Preferences > Additional Boards Manager URLs
Tools > Board > Boards Manager
Wait for downloads to complete.
Click "Close" (don't install the ESP8266 boards, you already have them installed)
The ESP8266 boards will now appear in your Tools > Board menu again.