esp8266/Arduino #4596: ESP restart by himsafe after 1-2h | ip != 0 wrong@glue-esp/lwip-esp.c:757

TakiKroSmaN, 2018-03-31:
#### Platform

- Hardware:				[ESP-12]
- Core Version:			[2.4.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		        [4MB/1MB]
- lwip Variant:                  [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:              [115200]

### Problem Description

after one or two hours of working without any problem ESP restart by himself. I cannot decode stack I don't know why but decoder is doesn't work on my PC:

### Debug Messages
I turn on debug and I get this stack:

ip != 0 wrong@glue-esp/lwip-esp.c:757
Fatal exception 29(StoreProhibitedCause):
epc1=0x40219250, epc2=0x00000000, epc3=0x400004e3, excvaddr=0x00000000, depc=0x00000000

Exception (29):
epc1=0x40219250 epc2=0x00000000 epc3=0x400004e3 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc70 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe10:  3fff4ab8 00000090 3ffeff48 00000016  
3ffffe20:  3ffef8ac 3ffef7e8 3ffeceb8 00000014  
3ffffe30:  4022814c 3ffffed0 3ffefdf0 3fff4854  
3ffffe40:  3fff25cc 00000000 3fff25c8 40219301  
3ffffe50:  4021c672 3ffeb004 3fff25c8 4021c874  
3ffffe60:  00000000 3ffffe90 00000000 4000050c  
3ffffe70:  3ffeb004 3ffeb004 3fff25c8 4021c8a2  
3ffffe80:  3fff31a4 3fff25c8 3fff2680 4021ead3  
3ffffe90:  c0a80001 3fff2804 00000000 40228e43  
3ffffea0:  00000000 000000f9 3fff488e 3fff26a6  
3ffffeb0:  000000f5 3fff2680 000000f0 00000004  
3ffffec0:  00000004 00000004 00000000 3fff4854  
3ffffed0:  00000224 00000014 00000020 3fff4854  
3ffffee0:  3fff26b0 3fff4886 3fff2d2c 3fff4854  
3ffffef0:  3fff26b0 3fff4886 3fff2d2c 4021cb34  
3fffff00:  00000001 00000043 3fff25c8 401007ac  
3fffff10:  40220000 3ffecf1b 3fffff80 3fff4872  
3fffff20:  3fff25c8 3fff4854 3fff26b0 4021ff00  
3fffff30:  00000014 00000001 0000007f 3fff25c8  
3fffff40:  3fffdc80 3fff2bc4 3fff42d4 3fff2c44  
3fffff50:  00000008 3fff25c8 3fff4854 40219665  
3fffff60:  3fffdc80 3fff2bc4 3fff42dc 40104858  
3fffff70:  40236ae2 3fff2bc4 3fff42dc 40236af4  
3fffff80:  3fff4864 3fff4854 00000000 3fff1180  
3fffff90:  402319df 00000000 3fff42dc 40238a73  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
⸮H⸮SDK:2.2.1(cfd48f3)/Core:2.4.1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
STUB: dhcp_stop

I have found this line in core 2.4.1 ip != 0 wrong@glue-esp/lwip-esp.c:757 but I don't know for what is it.

for moment I go back to core 2.4.0 and testing the same program once again.

devyte, 2018-04-02:
@TakiKroSmaN without a MCVE sketch it is unlikely this will be looked at. This was required as art of the issue template. Can you provide it?

TakiKroSmaN, 2018-04-02:
@devyte Sure please find below **part** of the sketch. I removed SSID, password, and keys for api so adresses in HTTP functions are not correct :) 
I think my problem is solved. Solution was just to assign IP to MAC in router. Otherwise my router after 2h (settings of my router) restart IP assignet and my ESP from 192.168.0.120 goes to something like 192.168.0.10x by this time for a second ESP IP adress was 0.0.0.0 and trigger function IP!=0 which crash ESP 
I found it by using core 2.4.0 - at the end of previous post is an info about it.

[image]

sorry for language mix below :)
```

#include <TimeLib.h>
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPClient.h>
#define ENCODER_USE_INTERRUPTS
#include <Encoder.h>
#include <Wire.h>
#include <FS.h>
#include <EEPROM.h>

#define DEBUG   // zakomentować w gotowym projekcie
//#define debugCzasu
#define M1A 16  // silnik 1A
#define M1B 14  // silnik 1B
#define E1A 12  // encoder 1A
#define E1B 13  // encoder 1B
#define BLUE_LED 2  // LED na module ESP

Encoder Encoder1 (E1A, E1B);

WiFiClient client;
ESP8266WebServer server(80);

const char* ssid = "XXXX";
const char* password = "YYYYYY";

byte humidity, clouds, IPaddress = 120;

void getTime() {
	server.stop();
	#ifdef DEBUG
	Serial.println("getTime");
	#endif
	boolean bRetry = LOW;
	retry:
	HTTPClient http;  //Object of class HTTPClient
	http.begin("http://api.timezonedb.com/?zone=Europe");
	int httpCode = http.GET();
	//Check the returning code
	if (httpCode > 0) {
		// Parsing
		const size_t bufferSize = JSON_OBJECT_SIZE(2) + JSON_OBJECT_SIZE(3) + JSON_OBJECT_SIZE(5) + JSON_OBJECT_SIZE(8) + 370;
		DynamicJsonBuffer jsonBuffer(bufferSize);
		JsonObject& root = jsonBuffer.parseObject(http.getString());
		// Parameters
		gmtOffset = root["gmtOffset"];
		gmtOffset = gmtOffset / 3600;
		setTime(root["timestamp"]);
		bladNr[10] = 0;
	}
	else {
		bladNr[10] = 1;
		if (bRetry = LOW) {
			bRetry = HIGH;
			STOP
			delay(2000);
			goto retry;
		}
	}
[... 160 lines cut ...]
	else {
	bladNr[0] = 0;
	digitalWrite(BLUE_LED, HIGH);
	}

	// tryb automatyczny ******************************************************************
	if (!bManual && !bTrybReferencji) {
	trybAutomatyczny();
	}

	// odswiez dane godziny i pogody ******************************************************************
	odswiezDane();

	// jeśli roleta po za zasięgiem ****************************************************
	if (actPos < 0 - toleracjaPozycji && !bTrybReferencji) {
	STOP
	ObslugaBledu(31);
	}
	if (actPos > numPozycjaDocelowa1 + toleracjaPozycji && !bTrybReferencji) {
	STOP
	ObslugaBledu(32);
	}
	}

```

d-a-v, 2018-04-02:
These ```wrong@``` and ```STUB``` message are on me. At least IP static management needs revisiting in lwip2.
@TakiKroSmaN you can try with 2.4.1 or git version *and* lwip1.4 in the meantime.

Markusenz, 2018-04-17:
I have the exact same issue here. Problem is that in 2.4.1 there is an issue when configuring a static IP:
My ESP8266 is connecting with static IP, but in the background it makes a DHCP DISCOVER request to my router for a reason I do not understand. This takes place at the moment of WiFi.Begin().
my code:
  WiFi.mode(WIFI_STA);
  WiFi.config(ownip, gateway, subnet);    
  WiFi.begin(ssid, password);
The ESP8266 connects correctly with static IP 192.168.6.245, but at the same time it send DHCP DISCOVER. My router replies by  offering 192.168.6.117 with a lease time of 10 minutes.
The ESP then is responding to the static IP 192.168.6.245. However, after exactly 10 minutes (DHCP lease timeout), it crashes with debug message

ip != 0 wrong@glue-esp/lwip-esp.c:757
...

This needs to be fixed, should be a problem for many users !?

d-a-v, 2018-04-17:
Can you please try this, assuming you are using git:
```
cd tools/sdk/lwip2
make download
cd builder
git checkout master
cd ..
make install
```
and check again for this issue ?

lrmoreno007, 2018-04-18:
Fixed IP need less time than DHCP.

Insert this lines for wait connection result:
 while (WiFi.status() != WL_CONNECTED) { 
delay(500); 
//Serial.print(".");
 }

Regards

d-a-v, 2018-04-18:
@Markusenz 
While @lrmoreno007 is right, your sketch as-is is working here (no crash, static IP).
The DHCP request is indeed sent but that's because of your ```delay(1000)``` and [this interpretation](https://github.com/esp8266/Arduino/issues/4495#issuecomment-373539281).
Without your delay, the dhcp request is not sent.
To be sure, I even changed my dhcp server's lease time to 1mn. Not any one is coming without your initial delay.
Can you please double check ?

Markusenz, 2018-04-18:
Thank you! When removing the delay, DHCP seems completely turned off now, I do not see any requests now.

However, I could reproduce the crash of the above mentioned sketch on another Arduino installation, also with 2.4.1 core, however it still uses the original lwip2 version from the 2.4.1 installation. 
(code as above, but removed the delay now).The output:

```
SDK:2.2.1(cfd48f3)/Core:2.4.1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
mode : sta(2c:3a:e8:36:4a:0f) + softAP(2e:3a:e8:36:4a:0f)
add if0
STUB: dhcp_stop
bcn 0
del if1
mode : sta(2c:3a:e8:36:4a:0f)
⸮⸮*⸮P*V⸮⸮ET⸮⸮⸮⸮T⸮⸮K⸮⸮u⸮U+UZJ⸮U⸮TEK⸮⸮⸮⸮⸮
```
I don't know if this issue is relevant since the order of Wifi commands seems to be unusual, but its not documented anywhere I think. DHCP starts so I think the WiFi.mode() shall be the first command, to prevent DHCP coming up.

d-a-v, 2018-04-18:
Would you be able to try this other sketch/installation with the current git version of the core (and with the lwip2 update from above) ?

lrmoreno007, 2018-04-18:
First of all, are you using a static IP in the DCHP range of your router?
****
In the other hand, surely you are using "Erase Flash : All Flash Content", then this is what occurs:
- first start after upload the ESP start as AP
- with config() you set STA+ AP, config STA+AP  and disable DHCP
- with mode() kill STA+AP and set STA but without configured IPs (don't know how DHCP state is).
- with begin() you connect to WiFi.
- then a WDT reset occurs

After WDT reset, ESP start as STA and everything work fine, at least for me.

It happens more or less the same with the order that I told you (as we can see in your feedback), but it's the correct order.

You can check all this by inserting `WiFi.printDiag(Serial)` after each instruction:
```
  WiFi.printDiag(Serial);
  WiFi.config(ownip, gateway, subnet);
  WiFi.printDiag(Serial);
  WiFi.mode(WIFI_STA);
  WiFi.printDiag(Serial);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println(".");
  }
  WiFi.printDiag(Serial);
```

```
SDK:2.2.1(cfd48f3)/Core:win-2.5.0-dev/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
mode : sta(xx:xx:xx:xx:xx:xx) + softAP(xx:xx:xx:xx:xx:xx)
add if0
STUB: dhcp_stop
Mode: STA+AP
PHY mode: N
Channel: 1
AP id: 0
Status: 0
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
bcn 0
del wifi evt: 8
if1
mode : sta(xx:xx:xx:xx:xx:xx)
Mode: STA
PHY mode: N
Channel: 1
AP id: 0
Status: 0
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
wifi evt: 8
wifi evt: 2
.....scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with XXXXXXXX, channel 1
ip:192.168.1.101,mask:255.255.255.0,gw:192.168.1.1
ip:192.168.1.101,mask:255.255.255.0,gw:192.168.1.1
wifi evt: 0
wifi evt: 3
.
Mode: STA
PHY mode: N
Channel: 1
AP id: 0
Status: 5
Auto connect: 1
SSID (7): XXXXXXXX
Passphrase (9): XXXXXXXXX
BSSID set: 0
Fatal exception 0(IllegalInstructionCause):             

<-------------WDT-------------->

SDK:2.2.1(cfd48f3)/Core:win-2.5.0-dev/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
Mode: STA
PHY mode: N
Channel: 1
AP id: 0
Status: 1
Auto connect: 1
SSID (7): XXXXXXXXX
Passphrase (9): XXXXXXXXXXX
BSSID set: 0
Swifi evt: 2
TUB: dhcp_stop
Mode: STA
PHY mode: N
Channel: 1
AP id: 0
Status: 1
Auto connect: 1
SSID (7): XXXXXXXXXX
Passphrase (9): XXXXXXXXXXXXX
BSSID set: 0
Mode: STA
PHY mode: N
Channel: 1
AP id: 0
Status: 1
Auto connect: 1
SSID (7): XXXXXXXXXXX
Passphrase (9): XXXXXXXXXXXX
BSSID set: 0
scandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with XXXXXXXXXXX, channel 1
ip:192.168.1.101,mask:255.255.255.0,gw:192.168.1.1
ip:192.168.1.101,mask:255.255.255.0,gw:192.168.1.1
wifi evt: 0
wifi evt: 3
.
Mode: STA
PHY mode: N
Channel: 1
AP id: 0
Status: 5
Auto connect: 1
SSID (7): XXXXXXXXXXXXX
Passphrase (9): XXXXXXXXXXX
BSSID set: 0
pm open,type:2 0
```

Markusenz, 2018-04-19:
Thanks @lrmoreno007, now I understand whats going on with DHCP. No, the static IP is not taken from the DHCP server's range.

I could reproduce the crash on another machine with latest versions, however as mentioned the crash only happens when issuing WiFi.config() before WiFi.mode():

`SDK:2.2.1(cfd48f3)/Core:2.4.1-72-g5d5ea92a/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-4-gbc33a88)
mode : sta(2c:3a:e8:36:4a:0f) + softAP(2e:3a:e8:36:4a:0f)
add if0
bcn 0
del if1
mode : sta(2c:3a:e8:36:4a:0f)
wifi evt: 8
wifi evt: 8
wifi evt: 2
.....scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with BOWi, channel 2
ip:192.168.6.245,mask:255.255.255.0,gw:192.168.6.5
ip:192.168.6.245,mask:255.255.255.0,gw:192.168.6.5
wifi evt: 0
wifi evt: 3
.Fatal exception 0(IllegalInstructionCause):
epc1=0x596a7495, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (0):
epc1=0x596a7495 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3fffeb90 end: 3fffffb0 offset: 01a0

>>>stack>>>

Decoding stack results
0x4020eda3: ethernet_output at netif/ethernet.c line 305
0x4020edac: ethernet_output at netif/ethernet.c line 305
0x4020cee8: _svfprintf_r at ../../../.././newlib/libc/stdio/nano-vfprintf.c line 531
0x4020f6dc: mem_malloc at core/mem.c line 136
0x4020f6dc: mem_malloc at core/mem.c line 136
0x401007c6: malloc at /Users/markusenzinger/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1676
0x4020ea2a: new_linkoutput at glue-lwip/lwip-git.c line 259
0x4020eda3: ethernet_output at netif/ethernet.c line 305
0x4020edac: ethernet_output at netif/ethernet.c line 305
0x40214e21: etharp_output_LWIP2 at core/ipv4/etharp.c line 893
0x402159dc: ip4_output_if_opt_src at core/ipv4/ip4.c line 962
0x4020f6dc: mem_malloc at core/mem.c line 136
0x40215a1c: ip4_output_if_opt at core/ipv4/ip4.c line 788
0x402151ad: igmp_send at core/ipv4/igmp.c line 793
0x4021554c: igmp_tmr at core/ipv4/igmp.c line 653
0x40212f30: cyclic_timer at core/timeouts.c line 165
0x40212f3e: cyclic_timer at core/timeouts.c line 171
0x40212ff4: sys_check_timeouts at core/timeouts.c line 351
0x4020e6b7: glue2esp_linkoutput at glue-esp/lwip-esp.c line 305
0x4020ea2a: new_linkoutput at glue-lwip/lwip-git.c line 259
0x4020eda3: ethernet_output at netif/ethernet.c line 305
0x4020edac: ethernet_output at netif/ethernet.c line 305`

d-a-v, 2018-09-04:
Fix is released in core-2.4.2

closed 2018-09-04 by d-a-v