esp8266/Arduino #4598: [BUG] timer1 interrupt doesn't work after timer1 NMI interrupt

AlfonsMittelmeyer, 2018-04-01:
In core version 2.3.0 analogWrite was implemented by a normal timer1 interrupt. In core version 2.4.0 a timer1 NMI interrupt is used (core_esp8266_timer.c). Therefore in tools/sdk/include/ets_sys.h the function ETS_FRC_TIMER1_NMI_INTR_ATTACH(func) exists. The timer1 NMI interrupt may be detached by using a NULL pointer as parameter for this function. But a normal timer1 interrupt via ETS_FRC_TIMER1_INTR_ATTACH(func, arg) and ETS_FRC1_INTR_ENABLE() doesn't work anymore after. Looks like a bug of the espressif sdk.

devyte, 2018-04-04:
@igrr do you know anything about this?

igrr, 2018-04-04:
Just to clarify, what is the sequence of calls to Arduino functions which triggers this problem?

AlfonsMittelmeyer, 2018-04-04:
This is very simple. Just call:

> ETS_FRC_TIMER1_NMI_INTR_ATTACH(NULL);

After this a normal timer1 interrupt, which is initiated by:

>   ETS_FRC_TIMER1_INTR_ATTACH(some_timer_isr,NULL);
>   ETS_FRC1_INTR_ENABLE();

doesn't work anymore

igrr, 2018-04-04:
ETS_blah aren't Arduino APIs, but `analogWrite` and `tone` are. Probably a bug in analogWrite that we don't detach timer from the NMI interrupt. Will check.

On a separate note, it probably makes sense for `tone` implementation to be merged with `analogWrite` and also use NMI...

AlfonsMittelmeyer, 2018-04-04:
Why is now  a timer1 NMI interrupt used in PWM? A NMI interrupt has a reaction time of nearly 10 µs. A normal timer1 interrupt needs only 4 µs.

igrr, 2018-04-04:
`analogWrite` uses NMI in order to produce correct waveform even when interrupts are disabled (via INTLEVEL or INTENABLE).

AlfonsMittelmeyer, 2018-04-04:
Yes, this is a reason, but timer1 cannot be used thereafter anymore, because there is a bug in the SDK for ETS_FRC_TIMER1_NMI_INTR_ATTACH(NULL). And for small PWM values below 10 the NMI timer1 interrupt isn't accurate enough. But if somebody disables the interrupts, an NMI interrupt would be better. For an NMI interrupt for my shared_timer module additionally shared_timer_nmi_isr_init() should be called in the tone and pwm module.

devyte, 2018-04-04:
It's not necessarily the user who has disabled the interrupts, it could be e.g. he's calling an api with timing critical code under the hood, like OneWire, where the interrupts must be disabled for the generation of the protocol pulses, or it won't work.

AlfonsMittelmeyer, 2018-04-04:
@igrr I didn't measure a difference between NMI and non NMI timer ISR. But there is a bug in PWM. What would you think about values near maximum?

For a value of 1020 I measured these times in microseconds via pulseIn:

1000, 1998, 1001, 4007, 3006, 1001, 1001, 1001, 3000, 1998

Seems to be a lot of jitter, but of course, it shouldn't matter much, whether we have a full cycle without switchung some microseconds to LOW.

Another question is, why is it so important, that we don't have some jitter and what about the accuracy of pulseIn? Could also be, that pulseIn jitters. So we don't know, whether jitter measured by pulseIn is jitter of the signal or of pulseIn. Of course the values above are not jitter of pulseIn, but a missing LOW for values near maximum.

devyte, 2018-04-04:
@AlfonsMittelmeyer you probably should be aware that there is reported flickering in the current pwm implementation, reference: #836 #2592 #2621 #2675 . There are reports of a better implementation for pwm ( see discussion in #836 ), but that particular implementation can't be integrated here due to licensing incompatibility.
I mention this, because maybe some ideas from there are useful for your work here. I haven't looked at either code in detail, so I can't really tell myself.

AlfonsMittelmeyer, 2018-04-05:
@devyte I noticed such flickering, when I tested with blinking for cycles below 300 ticks when using non NMI timer isr and for cycles below 800 ticks when using NMI timer isr. We need to find out about the reason.

@igrr Because I didn't notice a difference between non NMI timer isr and NMI timer isr for PWM in the case of Wifi not running, there seems to be no reason, why we shouldn't use NMI. And my opinion now is also, that NMI should be used for PWM.

Of course, measuring via pulseIn produces jitter, if there is an ISR running. We cannot have PWM and PulseIn performing well at the same time, when using only one ESP8266. For exact measurements timer0 could be used in NMI mode and timer1 in non NMI mode for PWM, in the case, if some jitter would be acceptable. What the priorities are, depend what the user wants to do. I would think, that the best way would be, if the user could configure the priorities for his application.

I will set NMI mode as default for my timer_shared module und I will do measurements about ticks for implementing exact time spans. There are some questions to be answered like how many ticks does it take from timer becoming 0 until the timer isr called and some further similar questions.

AlfonsMittelmeyer, 2018-04-05:
@igrr currently the time spans for small PWM values are too long, because it wasn't considered, that it takes time until the ISR is called. I made some tests by using this sketch:

```
volatile uint32_t ccount2 = 0;
volatile bool isr_ready = false;

static void ICACHE_RAM_ATTR timer_nmi_isr(void) {
  ccount2 = asm_ccount();
  isr_ready = true;
}

static void ICACHE_RAM_ATTR timer_isr(void * para) {
  ccount2 = asm_ccount();
  isr_ready = true;
}

static inline uint32_t asm_ccount(void) {
    uint32_t r;
    asm volatile ("rsr %0, ccount" : "=r"(r));
    return r;
}

void setup() {
  Serial.begin(115200);
  ETS_FRC_TIMER1_NMI_INTR_ATTACH(timer_nmi_isr);
  //ETS_FRC_TIMER1_INTR_ATTACH(timer_isr,NULL);
  //ETS_FRC1_INTR_ENABLE();
  T1C = 0x80;
  TEIE |= TEIE1;
}

void loop() {
  if(Serial.available()) {

    while(Serial.available())
      Serial.read();

     isr_ready = false;
     uint32_t ccount0 = asm_ccount();
     T1L = 1;
     uint32_t ccount1 = asm_ccount();

     while(!isr_ready);
     Serial.print(ccount1-ccount0);
     Serial.print(" ");
     Serial.println(ccount2-ccount0);
  }
}
```
This is the result for non NMI isr for 80 MHz CPU clock:

`11 121`

1 tick for calling asm_ccount(), 10 ticks for calling T1L. Normally the code doesn't have asm_ccount(), so swe have to subtract 2 ticks for two times asm_ccount and one tick, for value 1 used in T1L = 1.

This means, that we have a delay of 118 ticks, when using non NMI isr.

The results for NMI isr were:

```
325 447
11 133
11 133
11 133
11 133
11 133
```

When T1L is used the first time for a NMI isr, it has an additional delay of 314 ticks. But this shouldn't matter. The delay for NMI isr is 130 ticks instead of 118 ticks for non NMI or 119 ticks, if we disable other interrupts via uint32_t savedPS = xt_rsil(15;

This behaviour I will consider for the tick calculation in my timer shared module. For smaller values an idle loop within the isr would make sense.

AlfonsMittelmeyer, 2018-04-05:
NMI and not NMI isr shall have the same timing. This we may do by:

```
volatile uint32_t time_consuming = 0;

static void ICACHE_RAM_ATTR timer_isr(void * para) {
  uint32_t savedPS = xt_rsil(15);
  time_consuming = asm_ccount() + 1; // consuming 11 ticks (80 MHz CPU clock) , so that the timing is the same as for NMI isr
  timer_nmi_isr();
  xt_wsr_ps(savedPS);
}
```

igrr, 2018-04-07:
@AlfonsMittelmeyer  I see you are sharing valuable progress here, but i ask you to keep discussion in issues on topic, for everyone's sanity. Specifically, this one tracks the bug in SDK that Timer can not be used with non-NMI interrupt once NMI has been used. For other issues (such as which interrupt to use, NMI vs non-NMI, and so on), please open separate tickets.

FWIW, I have reported the issue about `NmiTimSetFunc` to ESP8266 SDK maintainers, so we now wait for an update there.

earlephilhower, 2018-08-04:
@AlfonsMittelmeyer we've released 2.4.2 which doesn't use NMI and which lets tone/analogWrite run in any order or at the same time.  So this specific bug is fixed as far as I understand the initial problem.

I know you had a good start on optimizing things, so if you have a PR to submit that helps bring down jitter using some of the techniques you were discussing, we'd love to see it!

closed 2018-08-04 by earlephilhower