esp8266/Arduino #4600: Set Hostname but cant do ping Request and not able to communicate using UDP communication

akshaydudhagara, 2018-04-02:
Hello friends;

i want to communicate two esp8266 using udp communication with passing UDP.begin(hostname,port) this  
arguments, but i am not able to do this, i dont  know how..please help me if any one knows.

here is my both firmware for esp8266

for esp8266-1

#include <ESP8266WiFi.h>        // Include the Wi-Fi library
#include <WiFiUDP.h>
#include <ESP8266mDNS.h>

WiFiUDP UDP;
MDNSResponder mdns;

char* ssid = "ssid"; // put here your router's config(SSID)
char* pw = "password";

char packetBuffer[UDP_TX_PACKET_MAX_SIZE];
int localPort = 2370;
bool udpConnected;

extern "C" {
#include "user_interface.h"
}
   
void setup() {
  Serial.begin(74880);         // Start the Serial communication to send messages to the computer
  delay(10);
  Serial.println('\n');
    
  Serial.println("Connecting ...");
  int i = 0;
  WiFi.hostname("ABI3K101");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pw);
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.println("trying to connect");
    delay(2000);
  }
Serial.println(String(WiFi.hostname()));
  if(mdns.begin("ABI3K101",WiFi.localIP()));
  {
    Serial.println("MDNS is started successfully");
  }
  Serial.println('\n');
  Serial.print("Connected to ");
  Serial.println(WiFi.SSID());              // Tell us what network we're connected to
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());           // Send the IP address of the ESP8266 to the computer

  UDP.begin(localPort);
  udpConnected = connectUDP();
}

void loop() {
  //
  if (WiFi.status() == WL_CONNECTED && udpConnected == true)
  {
    Serial.println(WiFi.hostname());
    Serial.println("Sending msg");
    Serial.println(UDP.beginPacket("ABI1K101.local", localPort));
    Serial.println(UDP.write("hy"));
    Serial.println(UDP.endPacket());
  }
  udp_data();
  delay(2000);
}

boolean connectUDP()
{
  boolean state = false;
  Serial.println("");
  Serial.println("Connecting to UDP");

  if (UDP.begin(localPort) == 1) {
    Serial.println("Connection successful");
    state = true;
  }
  else {
    Serial.println("Connection failed");
  }
  return state;
}

void udp_data()
{
  int packetSize = 0;
  if (udpConnected) {
    packetSize = UDP.parsePacket();
    if (packetSize)
    {
      Serial.println("");
      Serial.print("Received packet of size ");
      Serial.println(packetSize);
      Serial.print("From ");
      IPAddress remote = UDP.remoteIP();
      int remoteport = UDP.remotePort();
      Serial.print(UDP.remoteIP());
      Serial.print(", port ");
      Serial.println(UDP.remotePort());

      UDP.read(packetBuffer, UDP_TX_PACKET_MAX_SIZE);
      Serial.println("Contents:");
      Serial.println(packetBuffer);
      String packetdata = String(packetBuffer);

    }
    for (int k = 0; k < UDP_TX_PACKET_MAX_SIZE; k++)
      packetBuffer[k] = NULL;
  }
}

for esp8266-2

#include <ESP8266WiFi.h>        // Include the Wi-Fi library
#include <WiFiUDP.h>
#include <ESP8266mDNS.h>

WiFiUDP UDP;
MDNSResponder mdns;

char* ssid = "ssid"; // put here your router's config(SSID)
char* pw = "password";

char packetBuffer[UDP_TX_PACKET_MAX_SIZE];
int localPort = 2370;
bool udpConnected;

extern "C" {
#include "user_interface.h"
}

void setup() {
  Serial.begin(74880);         // Start the Serial communication to send messages to the computer
  delay(10);
  Serial.println('\n');

  Serial.println("Connecting ...");
  int i = 0;

  WiFi.hostname("ABI1K101");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pw);
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.println("trying to connect");
    delay(2000);
  }
  Serial.println(WiFi.hostname());
  if(mdns.begin("ABI1K101",WiFi.localIP()));
  {
    Serial.println("MDNS is started successfully");
  }
  Serial.println("mDNS responder started");
  Serial.println('\n');
  Serial.print("Connected to ");
  Serial.println(WiFi.SSID());              // Tell us what network we're connected to
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());           // Send the IP address of the ESP8266 to the computer

  UDP.begin(localPort);
  udpConnected = connectUDP();
}

void loop() {

  if (WiFi.status() == WL_CONNECTED && udpConnected == true)
  {
    Serial.println(WiFi.hostname());
    Serial.println("Sending msg");
    Serial.println(UDP.beginPacket("ABI3K101.local", localPort));
    Serial.println(UDP.write("hello"));
    Serial.println(UDP.endPacket());
  }
  udp_data();
  delay(2000);
}

boolean connectUDP()
{
  boolean state = false;
  Serial.println("");
  Serial.println("Connecting to UDP");

  if (UDP.begin(localPort) == 1) {
    Serial.println("Connection successful");
    state = true;
  }
  else {
    Serial.println("Connection failed");
  }
  return state;
}

void udp_data()
{
  int packetSize = 0;
  if (udpConnected) {
    packetSize = UDP.parsePacket();
    if (packetSize)
    {
      Serial.println("");
      Serial.print("Received packet of size ");
      Serial.println(packetSize);
      Serial.print("From ");
      IPAddress remote = UDP.remoteIP();
      int remoteport = UDP.remotePort();
      Serial.print(UDP.remoteIP());
      Serial.print(", port ");
      Serial.println(UDP.remotePort());

      UDP.read(packetBuffer, UDP_TX_PACKET_MAX_SIZE);
      Serial.println("Contents:");
      Serial.println(packetBuffer);
      String packetdata = String(packetBuffer);

    }
    for (int k = 0; k < UDP_TX_PACKET_MAX_SIZE; k++)
      packetBuffer[k] = NULL;
  }
}

RudyFiero, 2018-04-03:
@akshaydudhagara  This is not the right place for your question. It is for issues affecting this repository. Bugs, or new feature requests regarding the ESP8266 Arduino framework code. Your request is neither of these. Your request is asking how to do something and not pointing out a problem with the code in this repository. You should be asking these kind of questions at places like www.esp8266.com

There already is a method to accomplish what you want. There are examples that you can use and adapt to your requirements. 
https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266mDNS/examples
mDNS-SD_Extended.ino  and  mDNS_Web_Server.ino are where you should start. It is for discovery of the web server but it is easy to change it to esp (or whatever you call it) on udp. I know because I just did it. I have the mDNS-SD_Extended device scanning for esp, udp. When it finds the esp device it uses the I.P. address and port from the discovery and then sends a message to the udp server. 

```
/*
  ESP8266 mDNS-SD responder and query sample

  This is an example of announcing and finding services.
  
  Instructions:
  - Update WiFi SSID and password as necessary.
  - Flash the sketch to two ESP8266 boards
  - The last one powered on should now find the other.
 */

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>

const char* WiFi_hostname = "mDNS scanner 4.106";

const char * ssid = " ";
const char * password = " ";

unsigned int localPort = 8888;

WiFiUDP udp;

//-----------------------------------------------------
void nMDScheck() {
  Serial.println("Sending mDNS query -  esp,udp");
  int n = MDNS.queryService("esp", "udp"); // Send out query for esp tcp services
  Serial.println("mDNS query done");
  if (n == 0) {
    Serial.println("no services found");
  }
  else {
    Serial.print(n);
    Serial.println(" service(s) found");
    for (int i = 0; i < n; ++i) {
      // Print details for each service found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(MDNS.hostname(i));
      Serial.print(" (");
      Serial.print(MDNS.IP(i));
      Serial.print(":");
      Serial.print(MDNS.port(i));
      Serial.println(")");
    }
  }
  Serial.println();
  
}
//========================================================
void setup() {
  Serial.begin(115200);
  Serial.println("");
  Serial.println("mDNS-SD_singleDiscovery.ino");
  Serial.print("Lookes for specific service - esp-device");
  Serial.print("");
  
  WiFi.hostname(WiFi_hostname);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(250);
[... 159 lines cut ...]
    Serial.print(udp.remoteIP());
    Serial.print(":");
    Serial.println(udp.remotePort());

    udp.beginPacket(udp.remoteIP(), udp.remotePort());
    udp.write("UDP packet was- ");
    udp.write(packetBuffer);
    //udp.write(temperature);
    //udp.write("\r\n");
    udp.endPacket();

    //-----
    clearLCDline(2);
    lcd.setCursor(0, 2);    // column, row
    lcd.print(udp.remoteIP());
    lcd.print(":");
    lcd.print(udp.remotePort());
    clearLCDline(3);
    lcd.setCursor(0, 3);    // column, row
    lcd.print(packetBuffer);

  }
  delay(500);
}
```

devyte, 2018-04-03:
Previous comment is correct. Closing as off topic, see issue POLICY doc for details.

closed 2018-04-03 by devyte