esp8266/Arduino #4603: HTTPClient.getString() randomly returns empty string when memory is low

teo1978, 2018-04-03:
----------------------------- Delete above -----------------------------

### Basic Infos

#### Platform

- Hardware:					ESP8266
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

At random times, when calling http.getString() after having done http.GET() and having obtained a code 200, getString() returns an empty string, even though the response from the server does have the Content-Header set properly.

I am under the strong impression that this happens when there is little RAM available, and/or when the size of the response is bigger. If I try to load urls that return bigger response bodies (and hence use more RAM), I get the issue systematically, if I reduce the size of the response enough, I manage to never get the issue; with intermediate response sizes I get the issue intermittently and randomly (meaning that if I execute the exact same sketch several times loading a url that returns always the same response, I sometimes get an empty string and sometimes get the expected result.

If there is an issue with insufficient RAM that prevents you from properly processing the response and return the string, you should throw an exception. A random unpredictable empty string is unacceptable.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {
  HTTPClient http;
  String url="http://some_url";
  int code=http.GET();
  if (code==200) {
    Serial.println(http.getString());
  }
}

void loop() {

}

```

### Debug Messages

```
Expected: the contents of the response body
Observed: sometimes as expected, sometimes empty.
```

devyte, 2018-04-04:
@teo1978 this is on purpose. There are no exceptions on the ESP,  they are disabled in the build process. The reason, IIRC, is that they require additional heap use and binary size, and given the current constraints we'd rather live without them.
So, when oom, an alloc will fail, and there have been different decisions made in different parts of the core code about how to handle that.
If you would like to investigate enabling exceptions, at least in a way that doesn't increase heap use, it would be a great contribution.
Closing.

devyte, 2018-04-04:
BTW, there is an oom debug option available in the IDE menu. It is there to detect the situation (serial msg).

teo1978, 2018-04-04:
It doesn't have to be literally an exception, but there MUST be some way to detect what's happened at runtime.
One may well get a true well-formed response with a legitimate empty body and that must be distinguishable from an error so dramatic as a failure to allocate memory.

d-a-v, 2018-04-04:
This is a very valid demand. Users should be warned about something going badly wrong.

teo1978, 2018-04-04:
> this is not always possible, e.g.: in managed object construction like...

There are a number of workarounds. You could add another "status"-like property or method that we (meaning the program we write) can check to know whether getting the string was successful.
I can't just check whether the obtained string equals "" because I wouldn't know whether I got a legitimate empty response from the server or I failed to store it as string.

> If by "let the user know" you mean print a message, isn't that covered by the oom debug option?

LOL no. By "letting the user know" we mean that the program we write must be able to detect it, and then take whatever action to let OUR user know (or even something else e.g. retry). There's not necessarily always something attached to the con port, or somebody reading its output

earlephilhower, 2019-01-16:
Exceptions can now be enabled, and you can use a catch{} block at the top of your app, or get a crash listing the uncaught OOM exception.

closed 2019-01-16 by earlephilhower