esp8266/Arduino #4604: compiler trouble

5chufti, 2018-04-04:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:			[none]
- Core Version:			[latest git]
- Development Env:		[Arduino IDE 1.8.5]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [other] (serial upload only)
- warnings: [all]

### Problem Description

1) lots of 
_<command-line>:0:23: warning: ISO C99 requires whitespace after the macro name [enabled by default]_
during compile of core components

2) after "Done compiling" without any changes whatsoever just hitting "compile" button again brings
_Build options changed, rebuilding all_
maybe related to 1)?
### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

yes, just an empty sketch is fastest and easiest to demonstrate

d-a-v, 2018-04-04:
Can you by chance make a copy-paste of the gcc command line with its error ?
you can get it by enabling some option in IDE's preference menu

About the always rebuilding problem, I had it too, not tried to debug the IDE, and was finally solved by @whyameye's findings in #3911.

5chufti, 2018-04-04:
Hi,
looking closer, it is the setup described as "Using git version" I just use the path like hardware/espressif/esp8266_git (or_241 etc) and modify the platform.txt (name=ESP8266_git) but this shouldn't make such difference?

d-a-v, 2018-04-04:
I personally use symbolic links without messing with conf files.

5chufti, 2018-04-04:
here is one of the "offending" commandlines
`Compiling core...
cmd.exe /c mkdir C:\Users\me\AppData\Local\Temp\arduino_build_561890\core & (echo #define ARDUINO_ESP8266_GIT_VER 0x00000000 & echo #define ARDUINO_ESP8266_GIT_DESC win-2.5.0 ) > C:\Users\me\AppData\Local\Temp\arduino_build_561890\core\core_version.h
"C:\Users\me\arduino\hardware\espressif\esp8266_2.5.0/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\me\arduino\hardware\espressif\esp8266_2.5.0/tools/sdk/include" "-IC:\Users\me\arduino\hardware\espressif\esp8266_2.5.0/tools/sdk/lwip2/include" "-IC:\Users\me\arduino\hardware\espressif\esp8266_2.5.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\me\AppData\Local\Temp\arduino_build_561890/core" -c -g -x assembler-with-cpp -MMD -mlongcalls -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10806 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266_2.5.0 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-IC:\Users\me\arduino\hardware\espressif\esp8266_2.5.0\cores\esp8266" "-IC:\Users\me\arduino\hardware\espressif\esp8266_2.5.0\variants\generic" "C:\Users\me\arduino\hardware\espressif\esp8266_2.5.0\cores\esp8266\cont.S" -o "C:\Users\me\AppData\Local\Temp\arduino_build_561890\core\cont.S.o"
<command-line>:0:23: warning: missing whitespace after the macro name [enabled by default]`

Changing the platform.txt is not "messing with conf files", it only changes the caption in boards menu.

5chufti, 2018-04-04:
ok, I found my way to have parallel install of various versions and git.
so if no one else is concerned we can close this issue ...

d-a-v, 2018-04-04:
Hard to guess.
There is this unknown define to me ```-DARDUINO_ARCH_ESP8266_2.5.0``` which seems wrong with these dots.
...
I think this comes from your changes for {build_arch} in platform.txt: no other chars than alphanum and ```_``` are allowed in it (since it comes to be part of a define).

5chufti, 2018-04-04:
Hi d-a-v,
your my man! It didn't like the "-" nor the "." so I changed the path to esp8266_241, no it works like a charm. 
thank you very much for making my esp and arduino experience more joyfull again!

closed 2018-04-04 by 5chufti