esp8266/Arduino #4605: Esp8266 2.4.1 update issue

jeesonraphael, 2018-04-04:
jeesonraphael
Hi,
     I have installed esp8266  v 2.4.1 by using board manager. During the time of compiling there showing an error -- functions are " not declared in this scope ". But if I moved that function to the top of  "void setup()" code is running smoothly. I am just a beginner in this.

. . . .  error code . . . . . 

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  Led();
}

void Led() {
  digitalWrite(LED_BUILTIN, LOW);   
  delay(1000);                      
  digitalWrite(LED_BUILTIN, HIGH);  
  delay(2000);
}

. . . . error message . . . . .  

Arduino: 1.6.9 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

WARNING: Spurious .github folder in 'IRremoteESP8266' library
C:\Users\jeesonraphael\Desktop\Blink\Blink.ino: In function 'void loop()':

Blink:8: error: 'Led' was not declared in this scope

   Led();

       ^

exit status 1
'Led' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

. . . . . running code . . . . .

void Led() {
 digitalWrite(LED_BUILTIN, LOW);
 delay(1000);
 digitalWrite(LED_BUILTIN, HIGH);
 delay(2000);
}

void setup() {
 pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
 Led();
}

d-a-v, 2018-04-04:
You indeed found that functions must be declared before using them, this is how C and C++ languages works.
(some would say that order does not matter inside C++ classes, but we are not there).

The spurious warning is just a warning from the Arduino IDE. It is up to you to (re)move that directory from that library if you think you can do it.

Anyway this is not an issue with the esp8266/arduino core. So closing.

igrr, 2018-04-04:
@d-a-v in Arduino, it should work. Arduino preprocessor should generate prototypes automatically so that users don't have to bother writing forward declarations. I can imagine what you think about this practice, but that's how it is supposed to work. If it doesn't, it means that we have pulled in some header file into Arduino.h which confuses the preprocessor, messing up prototype generation. Need to check this.

d-a-v, 2018-04-04:
Thanks @igrr 
@jeesonraphael sorry for my Arduino's basics ignorance :-]

devyte, 2018-04-04:
I can't reproduce this with Arduino IDE 1.8.2 and the provided code above, I get a clean build. 
Either there is an installation issue, or the issue is in IDE 1.6.9, which is ancient and not the version mentioned in our current docs, or there is a lib conflict or incompatible libs in OP's path.
I don't see a core issue here.
Closing as can't reproduce.

closed 2018-04-04 by devyte

lrmoreno007, 2018-04-04:
@igrr as you know it is the theory, but there are many cases in which the theory fails. An attempt has already been made to improve this on IDE Arduino 1.6.7 (17 december 2015)(see changelog https://www.arduino.cc/en/Main/ReleaseNotes) and many issues https://github.com/arduino/arduino-builder/issues/68 (in this one you collaborated ;-) ). But it still fails sometimes. 

An example of many other, fail with no-UTF path:
7 august 2016 --> https://github.com/arduino/Arduino/issues/5216

------------------------------

That is why it is good practice to declare the functions before using them, it costs nothing.

In any case, it is not a problem with the ESP8266-Arduino framework.

Regards.

d-a-v, 2018-04-04:
I have tested the failing sketch with arduino-1.9beta, works too.
Lots of surprises in arduino :)
@lrmoreno007 thanks for all these instructive pointers.

jeesonraphael, 2018-04-05:
I updated the arduino version 1.6.9 to 1.8.5.  Now it is working perfectly.
 Thank you everyone for your support. . .