esp8266/Arduino #4607: Serial Port Error

iemektar, 2018-04-04:
#### Platform

- Hardware:				ESP-01
- Core Version:			28.03.2018
- Development Env:		Arduino IDE 1.8.5
- Operating System:		Windows

### Settings in IDE

-Board: Generic ESP8266 Module
-Flash Mode: QIO
-Flash Size: 512K (no SPIFFS)
-Debug Port: Disabled
-Debug Level: None
-IwIP: v2 Lower Memory
-Reset Method: ck
-Crystal Frequency: 24 MHz
-Flash Frequency: 40 MHz
-CPU Frequency: 80 MHz
-Buildin Led: 2
-Upload Speed: 115200
-Erase Flash: Only sketch
-Port: COM3

### Problem Description

I have Arduino Mega 2560, esp8266-01, gps and bluetooth modules. I need to use 3 serial ports but when i try use them i'm taking " Serial2 was not declared in this scope" error.

I try to fix it but I couldn't. Can you help me?

```cpp

#include <ESP8266WiFi.h>

void setup() {

  Serial.begin(115200);
  Serial1.begin(115200);
  Serial2.begin(9600);
}

void loop() {

}

```

### Debug Messages

```
...
sketch_apr04a:7: error: 'Serial2' was not declared in this scope

   Serial2.begin(9600);

   ^

exit status 1
'Serial2' was not declared in this scope
```

Pablo2048, 2018-04-04:
Not a bug IMO - ESP8266 has only 2 hw serial ports (and the second one is TX only).

iemektar, 2018-04-04:
I dont know and want to learn. Which esp module i must use?

d-a-v, 2018-04-04:
Like @Pablo2048 tells, there are only 1½ Hardware serial ports on every tastes of esp8266.
Use the SoftwareSerial library to have more.

devyte, 2018-04-04:
@iemektar if you don't know, and need guidance, this is not the right place to ask for it. This is an issue tracker, meant to track issues in the core code. Please read the issue POLICY doc, as mentioned in the issue template that you saw when opening this issue.
Closing due to user error and off-topic.

closed 2018-04-04 by devyte