esp8266/Arduino #4608: Random websocket disconnects when using WIFI_STA mode

bachi76, 2018-04-05:
[Since the [original issue](https://github.com/esp8266/Arduino/issues/1384#issuecomment-378710481) was closed, but the discussion continued around a possible bug, I'll file the findings as a new issue here.

This behaviour was tested with the latest 2.4 version and confirmed by other users.]

----

If WIFI_STA was set I experienced constant websocket connection interruptions, usually 1-2 times / min. :
    
    WiFi.mode(WIFI_STA);
    WiFi.begin(..., ...);

Setting `WiFi.mode(WIFI_AP);` has completely solved this, _even though the ESP connects to the existing network as if were in STA mode_.. 

    WiFi.mode(WIFI_AP);
    WiFi.begin(..., ...);

(I need to connect to an existing network - so using WIFI_AP is actually wrong, but apparently still connecting if the specified network already exists). 

With WIFI_STA mode set I couldn't get a stable websocket connection. Tested with two devices to be sure. Looks like a bug to me. @igrr ?

WereCatf, 2018-04-06:
Maybe it's just the power-saving of WIFI_STA-mode that's causing this? There are plenty of access-points/routers that seemingly have random issues with the ESP for some reason and this seems like one very possible candidate to me. I mean, I have no trouble using websockets with WIFI_STA, but then again, I've got a router running OpenWRT. If you have the option, maybe try with another AP and see if it changes anything?

bachi76, 2018-04-06:
I've tried it with two different APs (home: Netgear, office: Ubiquiti), but it was the same effect. Could of course still be that power-saving you've mentioned - is there a way to turn it off for testing?

WereCatf, 2018-04-06:
@bachi76 `wifi_set_sleep_type(NONE_SLEEP_T);` should do the trick, as seen in cores/esp8266/Updater.cpp

devyte, 2018-04-09:
@bachi76 please edit your OP and add the info requested in the template you saw when opening this issue. Alternatively, close this and open a new issue with all the requested info.

devyte, 2018-11-14:
Closing due to lack of feedback and info.

closed 2018-11-14 by devyte

Slayer4477, 2020-02-18:
i have router which can hostspot 2 port in time and ,, i write 
    WiFiMulti.addAP("hostpot1 ", "pasword1 ");
  WiFiMulti.addAP("hostpot2 ", "pasword2 "); 
but both signal frequence same so esp try to recon ,, after i comment one of them socket works stable

d-a-v, 2020-02-18:
@Slayer4477 This is a closed issue. 
Please try with latest release, then open a new issue and fill the issue template if you think it is related with a problem in this core.