esp8266/Arduino #4624: MQTT with TLS fail

xnorbi, 2018-04-09:
core:2.4.1
ESP-12

Iam using MQTT with TLS. With core version 2.3.0 evrything is going fine. But after upgrade to 2.4.1 I get the following error:

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:			ESP-12
- Core Version:		2.4.1
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		        [4MB]
- lwip Variant:   [v1.4|v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[|160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [230600] (serial upload only)

### Problem Description

Iam using MQTT with TLS. With core version 2.3.0 everything is going fine. But after upgrade to 2.4.1 I get the following error:

State:	receiving Certificate (11)
=== CERTIFICATE ISSUED TO ===
Common Name (CN):		a
Organization (O):		a
Organizational Unit (OU):	a
Location (L):			aa
Country (C):			HU
State (ST):			a
=== CERTIFICATE ISSUED BY ===
Common Name (CN):		server
Organization (O):		Irrigation system
Organizational Unit (OU):	c.o
Location (L):			a
Country (C):			HU
State (ST):			a
Not Before:			Tue Sep  5 17:47:40 2017
Not After:			Fri Sep  4 17:47:40 2020
RSA bitsize:			2048
Sig Type:			SHA256
=== CERTIFICATE ISSUED TO ===
Common Name (CN):		server
Organization (O):		Irrigation system
Organizational Unit (OU):	c.o
Location (L):			a
Country (C):			HU
State (ST):			a
Basic Constraints:		CA:TRUE, pathlen:10000
=== CERTIFICATE ISSUED BY ===
Common Name (CN):		server
Organization (O):		Irrigation system
Organizational Unit (OU):	c.o
Location (L):			a
Country (C):			HU
State (ST):			a
Not Before:			Tue Sep  5 17:46:11 2017
Not After:			Fri Sep  4 17:46:11 2020
RSA bitsize:			2048
Sig Type:			SHA256
:rd 5, 365, 314
:rdi 51, 5
:rd 46, 365, 319
:rdi 46, 46
:c0 46, 365
State:	receiving Certificate Request (13)
State:	receiving Server Hello Done (14)
State:	sending Certificate (11)
:wr 868 868 0
:wrc 256 868 0
:wrc 256 612 0
:wrc 256 356 0
:wrc 100 100 0
State:	sending Client Key Exchange (16)
:wr 204 267 0
:wrc 204 204 0
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wr 0 63 204
:wtmo
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wtmo
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wtmo
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wtmo
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wtmo
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:wr 0 63 0
:rcl
:abort
Error: connection lost
:ur 2
Alert: close notify
The mqtt state is: -2

xnorbi, 2018-04-09:
In 2.3.0 is evrything going well without error. After upgrade to version 2.4.0 or later I get the issue.

This is the code:

```
#include <ESP8266WiFi.h>
#include <PubSubClient.h>

#define MQTT_SERVER                       "mytestmosquitto.dynamic-dns.net"

const char* ssid     = "wifi";
const char* password = "password";

WiFiClientSecure espClient;
PubSubClient client(espClient);

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.println(ESP.getResetReason());

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("IP  address: ");
  Serial.println(WiFi.localIP());

  espClient.setCertificate(certificates_esp8266_bin_crt, certificates_esp8266_bin_crt_len);
  espClient.setPrivateKey(certificates_esp8266_bin_key, certificates_esp8266_bin_key_len);
  client.setServer(MQTT_SERVER, 8883);
  client.setCallback(mqtt_callback);
}

void loop() {
  delay(1000);
  client.connect("client_name","username","pass" );
  Serial.print("The mqtt state is: "); Serial.println(client.state());

}

void mqtt_callback(char* topic, byte* payload, unsigned int length) { 

}

```

chkr1011, 2018-04-17:
I have a similar issue. I use _PubSubClient_ together with _CloudMQTT_ and TLS. It seems that it works while publishing messages slowly (1 per second). But when sending 10 messages in a row the ESP crashes. Even with _loop()_, _delay()_, _yield()_, _ESP.feedWdt()_ etc. 

The problem is gone when using the not encrypted WiFi client. Even with 10 messages in a row there is no crash. The other code remains the same.

dejanra, 2018-04-23:
It's working with v2.4.0 rc1!

xnorbi, 2018-05-06:
Yes I can confirm it is working with v2.4.0 rc1 and v2.4.0 rc2.

With 2.4.0 and 2.4.1 is NOT working.

But if you set in Tools --> lwIP variant to v1.4 Compile from source it will work again!

So, you have to set lwIP variant to v1.4 Compile from source, the default is not working for me (others not tested).

earlephilhower, 2019-01-16:
Appears to be memory and axtls specific, and seems to have been fixed by using a lower memory LWIP stack.  Closing since we're not supporting axtls anymore.

closed 2019-01-16 by earlephilhower