esp8266/Arduino #4630: multiple definition of `_putc_r'

trlafleur, 2018-04-10:
See issue defined below....
tested with 2.4.0 and 2.4.1
~~~
https://github.com/grzesl/ESPMail/issues/4
~~~

~~~
C:\Users\anonymous\AppData\Local\Temp\arduino_build_962164/arduino.ar(libc_replacements.c.o): In function `_putc_r:

C:\Users\anonymous\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/libc_replacements.c:96: multiple definition of `_putc_r'

C:\Users\anonymous\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/lib\libc.a(lib_a-putc.o):/Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/putc.c:97: first defined here

collect2.exe: error: ld returned 1 exit status

~~~

igrr, 2018-04-11:
Thanks for reporting this! Started investigating this and found another bug: `printf` returns `-1` instead of the number of characters written, when writing to the console.

Komse, 2018-07-01:
Hi, I had the same issue. I needed to remove the "#include <iostream>" to compile.

cristianCarrerasCastillo, 2018-07-08:
remover of what file? @Komse

Komse, 2018-07-09:
Oh yes sorry, to remove the "#include <iostream>" @cristianCarrerasCastillo

shineangelic, 2018-07-11:
+1 on this. I didn't get what "#include" I shall remove, thanks

Komse, 2018-07-11:
For me it was the iostream include @shineangelic

shineangelic, 2018-07-11:
thank you @Komse but that did not help. I have other weird linking errors as reported in #4663

shineangelic, 2018-07-11:
I was able to solve this commenting out _put_c method already defined in "tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\"

closed 2018-08-26 by devyte