esp8266/Arduino #4632: delayMicroseconds uncorrect, about 50 ticks too much

AlfonsMittelmeyer, 2018-04-11:
I measured the following delays in CPU ticks (80 MHz):

```
delayMicroseconds(1) : 130 ticks, 50 ticks too much
delayMicroseconds(2) : 202 ticks, 42 ticks too much
delayMicroseconds(3) : 286 ticks, 46 ticks too much
delayMicroseconds(4) : 370 ticks, 50 ticks too much
delayMicroseconds(5) : 442 ticks, 42 ticks too much
```

Couldn't a precise function be implemented?

pipi61, 2018-04-11:
Hi!
How to measure this time? Source?
Measure value/how much the call/return time of delayMicroseconds function?
Measure value/How much the call nmi execution time?

AlfonsMittelmeyer, 2018-04-12:
@pipi61 : The CPU has 256 general registers and some special registers. The general registers are used by the compiler for local variables, so that the code may be executed very fast. One very interesting special register is the register ccount. It counts the CPU ticks. There is an instruction "read special register" with assembler mnemonic rsr. With this instruction we may read the cpu ticks into a general register, which we declare as a local variable. For C++ we may also make this inline function:

```
// ===========================================================
//               Measuring time by CCOUNT register
// -----------------------------------------------------------
static inline volatile uint32_t asm_ccount(void) {
    uint32_t r;
    asm volatile ("rsr %0, ccount" : "=r"(r));
    return r;
}
// ===========================================================
```
Rewading the CPU ticks costs one tick tim,e, which we may subtract. The time for the NMI interrupt doesn't matter, because we measure during the NMI interrupt, so that nothing may influence the measurement.

By using this sketch, you may test:

```
// ===========================================================
//               Measuring time by CCOUNT register
// -----------------------------------------------------------
static inline volatile uint32_t asm_ccount(void) {
    uint32_t r;
    asm volatile ("rsr %0, ccount" : "=r"(r));
    return r;
}
// ===========================================================

volatile uint32_t time_diff01;
volatile bool isr_ready;

volatile int32_t input_value = 0;
char inputbuffer[80];

static void ICACHE_RAM_ATTR timer_nmi_isr() {

  int32_t value = input_value;
  int32_t time_0;
  int32_t time_1;

  time_0 = asm_ccount();
  delayMicroseconds(value);
  time_1 = asm_ccount();
  time_diff01 = time_1 - time_0 - 1; // one tick subtracted for measurement by asm_ccount
  
  isr_ready = true;
}

void setup() {
  Serial.begin(115200);
  while(!Serial);

  // initialize timer1 nmi interrupt
  ETS_FRC_TIMER1_NMI_INTR_ATTACH(timer_nmi_isr);
  T1C = 0x80;
  TEIE |= TEIE1;
}

void loop() {
  size_t charcount;
  if(Serial.available()) {

    charcount = Serial.readBytesUntil('\n',inputbuffer,79);
    inputbuffer[charcount] = 0;
    sscanf(inputbuffer,"%ld",&input_value);
    
    isr_ready = false;
    T1L = 1;
    while(!isr_ready)
        
    delay(1); // Serial.print doesn't work proper the first time after the interrupt
    Serial.print("delayMicroseconds(");
    Serial.print(input_value);
    Serial.print(") : ");
    Serial.print(time_diff01);
    Serial.println(" ticks");
  }
}
```
Just opem the serial monitor, choose baud rate 115200 and sending with newline and enter a value (microseconds)

Makuna, 2018-04-12:
BTW, CCOUNT was also exposed off the ESP object for easy consumption.

AlfonsMittelmeyer, 2018-04-12:
For exact time consumption CCOUNT isn't neccessary. But the advantage of using CCOUNT within a delay function is, that an interruption by an interrupt wouldn't effect the delay function, except that the interrupt would exceed the intended duration of the delay.

AlfonsMittelmeyer, 2018-04-12:
What can we do in this case?

We can use an own function delay_us, which works correct:

```
// ===========================================================
//               Measuring time by CCOUNT register
// -----------------------------------------------------------
static inline volatile uint32_t asm_ccount(void) {
    uint32_t r;
    asm volatile ("rsr %0, ccount" : "=r"(r));
    return r;
}
// ===========================================================

// ===========================================================
//               delayTicks for ticks >= 17
//               by Alfons Mittelmeyer 2018
// -----------------------------------------------------------

void ICACHE_RAM_ATTR delayTicks(int32_t ticks) {

  uint32_t  expire_ticks = asm_ccount() + ticks -13;

  do {
    ticks = expire_ticks - asm_ccount();
  } while(ticks >= 6 );

  asm volatile (        // delay of remaining ticks by branches
    "blti %0, 1 , 0f;"
    "beqi %0, 1 , 0f;"
    "beqi %0, 2 , 0f;"
    "beqi %0, 3 , 0f;"
    "beqi %0, 4 , 0f;"
    "bgei %0, 5 , 0f;"
    "0:          ;" : : "r"(ticks)
   );
}
// ===========================================================

// ===========================================================
//           delay_us, which uses delayTicks
//           (use instead of delayMicroseconds)
//           by Alfons Mittelmeyer 2018
// -----------------------------------------------------------
void ICACHE_RAM_ATTR delay_us(int32_t us) {
  delayTicks(80 * us - 13); // -13 for function call of delayTicks and multiplication and subtraction
}
// ===========================================================
```

devyte, 2019-11-10:
@AlfonsMittelmeyer whatever happened to your work? did you ever reach a conclusion?

Tech-TX, 2019-11-23:
It's quite a bit farther off than that, but who's counting?

When I do:
<pin wiggle>
delayMicroseconds(10)
<pin wiggle>
the measured time is 10.75us.  Subtracting the time for the pin wiggle I get 10.662us actual delay.
With delayMicroseconds(1) the actual is 1.675us, and with delayMicroseconds(1) it's ~100.7 (accuracy of my 'scope is only 4 digits).  I vaguely remember measuring the time it took to do a call(parameter) / return being in the realm of 560ns, which is close to the 660ns I see above.
Quick math: 1 'tick' is 12.5ns, and 46 of them (the mid-point of his error range) is 575ns.  It looks like the calculation for delayMicroseconds forgot to subtract the time it takes to do it's call/return, as it's a (roughly) fixed positive error value from 1 to 100 us.

Time is relative.  Reading ccount ticks will always have a variance compared to real time due to the accuracy of the crystal, so there's a point of diminishing returns for time tweaks.  The CPU clock isn't exactly 80MHz, especially with dirt cheap crystals for a $2 module.  I just checked one of my boards that doesn't have the RF can and the crystal is running 25.9986MHz.  My freq counter is accurate to 8 digits, was designed for RF circuits, and doesn't load the crystal/cap network.  I checked at the output of the inverter. :-)