esp8266/Arduino #4635: Is it possible to change SPI Pin out?

nardev, 2018-04-11:
I have a device which i want to hook in following order:

SDA --> GPIO04
SCK --> GPIO05
MOSI --> GPIO08
MISO --> GPIO06

IRQ --> NC
GND --> GND
RST --> GPIO0
3.3V --> 3.3V

Us that possible???

nardev, 2018-04-12:
@devyte you didn't understand my question.

devyte, 2018-04-12:
@nardev entirely possible, I doubt I spent more than a minute on it.
But you obviously didn't read this repo's issue POLICY doc, or the issue template that shows up when you open an issue. 
This is an issue tracker, not a community forum, and like I said, not the right place to ask. Strictly speaking, I should have just closed the issue and not even have attempted an answer.

nardev, 2018-04-12:
@devyte i apologize, looks like you are right.