esp8266/Arduino #4638: HTTPUpdateServer with TLS not working with WiFiClientSecure

donnm, 2018-04-12:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[Wemos D1 R2]
- Core Version:			[4c23e66bbae61693b377172d61583ac356dc82e2]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[?]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[?]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I am using ESP8266HTTPUpdateServer over TLS together with an MQTT client (https://github.com/256dpi/arduino-mqtt) also using TLS for secure OTA updates. The basic idea is to publish an MQTT message putting the ESP into firmware update mode, which starts the update server (WebServerSecure on port 443 with a TLS certificate).

The update works fine as long as there is no MQTT client class (WiFiClientSecure) connected. It also works if HTTPUpdateServer uses an insecure HTTP webserver instead of the TLS enabled equivalent WebServerSecure.

The failure happens when the firmware is uploaded (clicking the upload button). This means the update server can serve a TLS encrypted upload form, but as soon as I click upload with a new .bin file, it goes into a loop of TLS handshaking, and never starts the upload. The normal serial message of "sleep disabled" and then the series of upload dots "." are not printed.

I have tried a bit of debugging, and it seems that _currentClientSecure.available() under the case HC_WAIT_READ is never returning a nonzero number (ESP8266WebServerSecure.cpp:103 master branch). My suspicion is that somehow the WiFiClientSecure is somehow interfering with the client in WebServerSecure.

If I comment out the mqttconnect() block in the loop() of the sketch below, everything works fine. Note that even if I call mqttclient.disconnect() before starting the server, the upload still does not work. It is something about the initially connected instantiation of the WiFiClientSecure class inside of MQTTClient.

Also note I have tried dynamically instantiating MQTTClient and deleting it before starting the update server in case it had to do with a memory issue, but the upload fails also in this case.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
/*
  SecureHTTPSUpdater - SSL encrypted, password-protected firmware update

  This example starts a HTTPS server on the ESP8266 to allow firmware updates
  to be performed.  All communication, including the username and password,
  is encrypted via SSL.  Be sure to update the SSID and PASSWORD before running
  to allow connection to your WiFi network.

  IMPORTANT NOTES ABOUT SSL CERTIFICATES

  1. USE/GENERATE YOUR OWN CERTIFICATES
    While a sample, self-signed certificate is included in this example,
    it is ABSOLUTELY VITAL that you use your own SSL certificate in any
    real-world deployment.  Anyone with the certificate and key may be
    able to decrypt your traffic, so your own keys should be kept in a
    safe manner, not accessible on any public network.

  2. HOW TO GENERATE YOUR OWN CERTIFICATE/KEY PAIR
    A sample script, "make-self-signed-cert.sh" is provided in the
    ESP8266WiFi/examples/WiFiHTTPSServer directory.  This script can be
    modified (replace "your-name-here" with your Organization name).  Note
    that this will be a *self-signed certificate* and will *NOT* be accepted
    by default by most modern browsers.  They'll display something like,
    "This certificate is from an untrusted source," or "Your connection is
    not secure," or "Your connection is not private," and the user will
[... 190 lines cut ...]
### Debug Messages

Serial output (DEBUG_ESP_HTTP_SERVER defined, each `New secure client` is a successful TLS handshake but nothing received after that):

```
Booting Sketch...
setup()
HTTPSUpdateServer ready!
Open https://esp8266-webupdate.local/firmware in your browser and login with username 'admin' and password 'admin'
Connecting to MQTT server...
Connected to MQTT server.
Subscribed.
New secure client
New secure client
New secure client
New secure client
New secure client
New secure client
New secure client
New secure client
New secure client
New secure client
New secure client

```

donnm, 2018-04-12:
I attach an example using only WiFiClientSecure giving the same behaviour. However, if WiFiClientSecure is statically declared in setup(), so that it destructs when setup returns, the update works.

```
/*
  SecureHTTPSUpdater - SSL encrypted, password-protected firmware update

  This example starts a HTTPS server on the ESP8266 to allow firmware updates
  to be performed.  All communication, including the username and password,
  is encrypted via SSL.  Be sure to update the SSID and PASSWORD before running
  to allow connection to your WiFi network.

  IMPORTANT NOTES ABOUT SSL CERTIFICATES

  1. USE/GENERATE YOUR OWN CERTIFICATES
    While a sample, self-signed certificate is included in this example,
    it is ABSOLUTELY VITAL that you use your own SSL certificate in any
    real-world deployment.  Anyone with the certificate and key may be
    able to decrypt your traffic, so your own keys should be kept in a
    safe manner, not accessible on any public network.

  2. HOW TO GENERATE YOUR OWN CERTIFICATE/KEY PAIR
    A sample script, "make-self-signed-cert.sh" is provided in the
    ESP8266WiFi/examples/WiFiHTTPSServer directory.  This script can be
    modified (replace "your-name-here" with your Organization name).  Note
    that this will be a *self-signed certificate* and will *NOT* be accepted
    by default by most modern browsers.  They'll display something like,
    "This certificate is from an untrusted source," or "Your connection is
    not secure," or "Your connection is not private," and the user will
    have to manully allow the browser to continue by using the
    "Advanced/Add Exception" (FireFox) or "Advanced/Proceed" (Chrome) link.

    You may also, of course, use a commercial, trusted SSL provider to
    generate your certificate.  When requesting the certificate, you'll
    need to specify that it use SHA256 and 1024 or 512 bits in order to
    function with the axTLS implementation in the ESP8266.

  Interactive usage:
    Go to https://esp8266-webupdate.local/firmware, enter the username
    and password, and the select a new BIN to upload.

  To upload through terminal you can use:
  curl -u admin:admin -F "image=@firmware.bin" esp8266-webupdate.local/firmware

  Adapted by Earle F. Philhower, III, from the SecureWebUpdater.ino example.
  This example is released into the public domain.
*/

#include <ESP8266WiFi.h>
#include <ESP8266WebServerSecure.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPUpdateServer.h>
#include <WiFiClientSecure.h>

const char* host = "esp8266-webupdate";
const char* update_path = "/firmware";
const char* update_username = "admin";
const char* update_password = "admin";

bool started = false;
const char* server = "api.github.com";
const int httpsPort = 443;
const char* fingerprint = "35 85 74 EF 67 35 A7 CE 40 69 50 F3 C0 F6 80 CF 80 3B 2E 19";

WiFiClientSecure client;
ESP8266WebServerSecure httpServer(443);
ESP8266HTTPUpdateServer httpUpdater(true);

// The certificate is stored in PMEM 
static const uint8_t x509[] PROGMEM = {
  0x30, 0x82, 0x01, 0xc9, 0x30, 0x82, 0x01, 0x32, 0x02, 0x09, 0x00, 0xe6,
[... 128 lines cut ...]
    } else {
      Serial.println("certificate doesn't match");
    }
    verified = true;
  }

  if(millis() > 10000 && !started) {
  
    MDNS.begin(host);
  
    httpServer.setServerKeyAndCert_P(rsakey, sizeof(rsakey), x509, sizeof(x509));
    httpUpdater.setup(&httpServer, update_path, update_username, update_password);
    httpServer.begin();
  
  
    MDNS.addService("https", "tcp", 443);
    Serial.printf("HTTPSUpdateServer ready!\nOpen https://%s.local%s in "\
                  "your browser and login with username '%s' and password "\
                  "'%s'\n", host, update_path, update_username, update_password);
    httpServer.begin();

    started = true;
  }
}
```

donnm, 2018-04-12:
OK, I narrowed it down and found a solution. But I still consider this a bug.

Below is another sketch, this time a modification of the original using the heap for MQTTClient and WiFiClientSecure. Before starting the update server I delete these objects, and the upload succeeds. The difference from my first attempt at using the heap is that was still statically declaring WiFiClientSecure. This is a sufficient workaround for my use case, but why is this happening?

```
/*
  SecureHTTPSUpdater - SSL encrypted, password-protected firmware update

  This example starts a HTTPS server on the ESP8266 to allow firmware updates
  to be performed.  All communication, including the username and password,
  is encrypted via SSL.  Be sure to update the SSID and PASSWORD before running
  to allow connection to your WiFi network.

  IMPORTANT NOTES ABOUT SSL CERTIFICATES

  1. USE/GENERATE YOUR OWN CERTIFICATES
    While a sample, self-signed certificate is included in this example,
    it is ABSOLUTELY VITAL that you use your own SSL certificate in any
    real-world deployment.  Anyone with the certificate and key may be
    able to decrypt your traffic, so your own keys should be kept in a
    safe manner, not accessible on any public network.

  2. HOW TO GENERATE YOUR OWN CERTIFICATE/KEY PAIR
    A sample script, "make-self-signed-cert.sh" is provided in the
    ESP8266WiFi/examples/WiFiHTTPSServer directory.  This script can be
    modified (replace "your-name-here" with your Organization name).  Note
    that this will be a *self-signed certificate* and will *NOT* be accepted
    by default by most modern browsers.  They'll display something like,
    "This certificate is from an untrusted source," or "Your connection is
    not secure," or "Your connection is not private," and the user will
    have to manully allow the browser to continue by using the
    "Advanced/Add Exception" (FireFox) or "Advanced/Proceed" (Chrome) link.

    You may also, of course, use a commercial, trusted SSL provider to
    generate your certificate.  When requesting the certificate, you'll
    need to specify that it use SHA256 and 1024 or 512 bits in order to
    function with the axTLS implementation in the ESP8266.

  Interactive usage:
    Go to https://esp8266-webupdate.local/firmware, enter the username
    and password, and the select a new BIN to upload.

  To upload through terminal you can use:
  curl -u admin:admin -F "image=@firmware.bin" esp8266-webupdate.local/firmware

  Adapted by Earle F. Philhower, III, from the SecureWebUpdater.ino example.
  This example is released into the public domain.
*/

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServerSecure.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPUpdateServer.h>
#include <MQTTClient.h>

const char* host = "esp8266-webupdate";
const char* update_path = "/firmware";
const char* update_username = "admin";
const char* update_password = "admin";

#define MQTT_SERVER "mqttserver"
#define MQTT_PORT 8883

ESP8266WebServerSecure httpServer(443);
ESP8266HTTPUpdateServer httpUpdater(true);

WiFiClientSecure* net = NULL;
MQTTClient* mqttclient = NULL;

[... 139 lines cut ...]
    if(!mqttclient->connected())
      mqttconnect();
  }

  if(millis() > 20000 && !started) {
    mqttclient->disconnect();
    delete mqttclient;
    delete net;  
  
    MDNS.begin(host);
  
    httpServer.setServerKeyAndCert_P(rsakey, sizeof(rsakey), x509, sizeof(x509));
    httpUpdater.setup(&httpServer, update_path, update_username, update_password);
    httpServer.begin();
    
    MDNS.addService("https", "tcp", 443);
    Serial.printf("HTTPSUpdateServer ready!\nOpen https://%s.local%s in "\
                  "your browser and login with username '%s' and password "\
                  "'%s'\n", host, update_path, update_username, update_password);
    httpServer.begin();

    started = true;
  }
}
```

devyte, 2018-05-29:
BearSSL is merged in #4273 , with alternate BearSSL::WiFi* classes. Although axtls-based classes are still available and even the default, they are planned for deprecation and then retirement, hence won't be fixed. Any issues with BearSSL-based classes should be reported in new issues.
Closing.

closed 2018-05-29 by devyte