esp8266/Arduino #4639: Incorrect expansion of ,##__VA_ARGS__ in macro

wmacevoy, 2018-04-12:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ?] I have tested that the issue is present in current master branch (aka latest git).
         Latest build from README docs build failed (Mac OS 10.12.6) on python get.py with:
         > cd esp8266/tools
         > python get.py
         IOError: [Errno socket error] [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)
- [ X] I have searched the issue tracker for a similar issue.
- [ X] If there is a stack dump, I have decoded it.
- [X ] I have filled out all fields below.

#### Platform

- Hardware:				[ESP8285 device]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[512MB]
- lwip Variant:   [v2mss536]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

g++ supports variadic macros which cleanly reduces to 0 arguments with the notation

```
#define va_macro(...) va(begin, ##__VA_ARGS__, end)
```
The `,##__VA_ARGS__`  notation expands to nothing if no arguments are given (the leading comma) is removed, while the comma remains for 1 or more arguments.

This works correctly on uno, mega and esp32 boards, but fails to expand correctly on esp8266.  This notation is important for variable parameter macros.  Here is a sketch that compiles under uno, mega, esp32, but has a compile-time-error for esp8266, with log files for uno, mega, esp32 (correct) and esp8266 (compile time error):

https://github.com/mmurdoch/arduinounit/tree/pr65/va_args

```cpp

#define va_macro(...) va_sub(-9,##__VA_ARGS__,9)

// Unimportant, just to have something the va_macro can compile to
void va_sub(int first, int last) { }
void va_sub(int first, int va1, int last) { }
void va_sub(int first, int va1, int va2, int last) {}

void setup() {
   va_macro(1,2);  // ok: expands to va_sub(-9,1,2,9);
   va_macro(1); // ok: expands to va_sub(-9,1,9); 
   va_macro(); // fail: expands to va_sub(-9,,9) on ESP 8266, but should be va_sub(-9,9);
}

void loop() {

}
```

### Debug Messages

```
None.  Does not compile.
```

igrr, 2018-04-13:
IIRc, this is a GNU extension. In this project C++ code is compiled with `-std=c++11`, so GNU extensions are disabled.

earlephilhower, 2018-04-13:
Yup, that'd be the cause.  Here is G++ 5.4 under Linux, same behavior.

````
earle@server:~/Arduino/ss1$ g++ -c /tmp/a.c 
<no error>

earle@server:~/Arduino/ss1$ g++ -c --std=c++11 /tmp/a.c 
/tmp/a.c: In function ‘void setup()’:
/tmp/a.c:1:46: error: expected primary-expression before ‘,’ token
 #define va_macro(...) va_sub(-9,##__VA_ARGS__,9)
                                              ^
/tmp/a.c:14:4: note: in expansion of macro ‘va_macro’
    va_macro(); // compile time error just on ESP 8266
    ^
````

wmacevoy, 2018-04-13:
It is a GNU extension, but AVR and ESP32 boards are compiled with those extensions enabled.  And I can find no resolution to this without those options.  This breaks ArduinoUnit on the ESP 8266.  c++20 seems to finally have a fix; but until then the standard is broken. (see __VA_OPT__ in https://en.cppreference.com/w/cpp/compiler_support).  How horrible is it to compile with -std=gnu++11, which is what the other systems ArduinoUnit supports does?  See https://github.com/mmurdoch/arduinounit/blob/pr65/README.md#known-bugs

PR #6339 (2019-07-25): Move to -std=g++11 from -std=c++11
Looking at other projects, they're all building using gnu extensions
(-std=gnu++11).  We're already embedding gnu-specific pragmas for
compile options, so I see no reason not to fall in line w/others on
this.

Fixes #4639
commit 17ee3807 (2019-07-25): Move to -std=g++11 from -std=c++11 (#6339)

Looking at other projects, they're all building using gnu extensions
(-std=gnu++11).  We're already embedding gnu-specific pragmas for
compile options, so I see no reason not to fall in line w/others on
this.

Fixes #4639
closed 2019-07-25 by d-a-v