esp8266/Arduino #4644: ESP-07 hanged after ESP.restart()

Rasoul-Khajavi, 2018-04-14:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:			ESP-07
- Core Version:			2.4.1
- Development Env:		Arduino 1.8.5
- Operating System:		Windows 10

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 		       1MB
- lwip Variant:                 v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	       40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:              921600

### Problem Description

ESP-07 hanged after  ``` ESP.restart(); ```  it can't be flashed And it doesn't reset with hard reset button(even after cutting the power supply and connecting it) ... 
is it dead ? can i flash it again ? 
i'm not a beginner so i have observed amateure tips .  
Serial port is not working at all so i don't have any debug code

RudyFiero, 2018-04-28:
Flash on the ESP-07 normally has been 512K and not 1M. How did you confirm that your ESP-07 has 1M? 
Are you sure that you had the configuration correct?

I made a sketch to try the esp.restart(); on one of my ESP-07. 

```
void loop()
{
  delay(2000);
  Serial.println("5");
  delay(2000);
  Serial.println("4");
  delay(2000);
  Serial.println("3");
  delay(2000);
  Serial.println("2");
  delay(2000);
  Serial.println("1");
  delay(2000);
  Serial.println("restart");
  ESP.restart();
  delay(7000);
  Serial.println("Huh?");
}
```

The output was the following.  And my ESP-07 still works and I have re-programmed it.

```
5
4
3
2
1
restart

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

5
4
3
2
1
restart

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

5
4
3
2
1
restart

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

5
4
3
2
1
restart

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
```

@Rasoul-Khajavi   What makes you think your problem is related to the code in this repository? Did you get multiple ESP-07 devices fail? Did you try it with 512K setting? Could you provide any more details.

devyte, 2018-04-28:
In addition to the previous, upload speed of 921600 is just asking for trouble.
Given the previous comment, closing as can't reproduce, and also not a core issue.

closed 2018-04-28 by devyte