esp8266/Arduino #4646: Always recompiling all?

NdK73, 2018-04-15:
After updating to latest GIT, I noticed that every time I upload a sketch I get "options changed, recompiling everything" (well, actually the Italian equivalent) even if I try re-uploading the same code because the previous upload failed.
Did I miss something obvious?

devyte, 2018-04-15:
@NdK73 maybe: #3911 .
if works for you, it's the same IDE issue, in which case please close this.

5chufti, 2018-04-20:
Hi,
actually it is not easy to accept it as an IDE problem since the same IDE does not need to recompile with earlier versions of the esp arduino core. Would really be interested what causes this behaviour, I suspect boards.txt as it was the culprit in earlier similar issue.