esp8266/Arduino #4648: mDNS library fails to find mqtt services on the network

thunderace, 2018-04-17:
----------------------------- Delete below -----------------------------

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12
- Core Version:			7820fb73a54137ca66be51184ddd76cd60c4d54d
- Development Env:		Make
- Operating System:		Debian

### Settings in Makefile

- Module:  			Nodemcuv2
- Flash Mode:			dio
- Flash Size: 		        4MB
- lwip Variant:                  v1.4
- Reset Method:		nodemcu
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:              115200

### Problem Description

mDNS library fails to find mqtt DNS-SD services on the network (avahi-daemon 0.6.31 on Linux debianServer 3.16.0-4-amd64 # 1 SMP Debian 3.16.51-2 (2017-12-03) x86_64 GNU/Linux).
The error is 'Data len too long! 4500'
Work fine with stable 2.4.1.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>

const char* ssid     = "main";
const char* password = "*************************";
char hostString[16] = {0};

void setup() {
  Serial.begin(115200);
  Serial.println("\r\nsetup()");

  sprintf(hostString, "ESP_%06X", ESP.getChipId());
  Serial.print("Hostname: ");
  Serial.println(hostString);
  WiFi.hostname(hostString);

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(250);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

 if (!MDNS.begin(hostString)) {
    Serial.println("Error setting up MDNS responder!");
  }
  Serial.println("mDNS responder started");

  Serial.println("Sending mDNS query");
  int n = MDNS.queryService("mqtt", "tcp"); 
  Serial.println("mDNS query done");
  if (n == 0) {
    Serial.println("no services found");
  }
  else {
    Serial.print(n);
    Serial.println(" service(s) found");
    for (int i = 0; i < n; ++i) {
      // Print details for each service found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(MDNS.hostname(i));
      Serial.print(" (");
      Serial.print(MDNS.IP(i));
      Serial.print(":");
      Serial.print(MDNS.port(i));
      Serial.println(")");
    }
  }
  Serial.println();
}

void loop() {
}
```

### Debug Messages with git version

```
setup()
Hostname: ESP_E0440F
............
Connected to main
IP address: 192.168.1.188
MDNS listening
mDNS responder started
Sending mDNS query
Reading answers RX: REQ, ID:0, Q:0, A:5, NS:0, ADD:0
 5 _mqtt
found matching service: mqtt
 4 _tcp
 5 local
type: 000c rdlength: 40
PTR 40 %Mosquitto MQTT server on debianServer�
Compressed pointer, jumping from 82 to 40
 37 Mosquitto MQTT server on debianServer
Compressed pointer, jumping from 80 to 12
 5 _mqtt
found matching service: mqtt
 4 _tcp
 5 local
Compressed pointer, jumping back to 80
Data len too long! 4500
mDNS query done
no services found

```
### Debug Messages with stable (2.4.1)

```
setup()
Hostname: ESP_E0440F
....
Connected to main
IP address: 192.168.1.188
MDNS listening
mDNS responder started
Sending mDNS query
Reading answers RX: REQ, ID:0, Q:0, A:5, NS:0, ADD:0
 5 _mqtt
found matching service: mqtt
 4 _tcp
 5 local
type: 000c rdlength: 40
PTR 40 %Mosquitto MQTT server on debianServer�
type: 0010 rdlength: 1
TXT 1 
type: 0021 rdlength: 21
SRV 12 64 65 62 69 61 6e 53 65 72 76 65 72 
debianServer
type: 001c rdlength: 16
Ignoring unsupported type 6f
type: 0001 rdlength: 4
All answers parsed, adding to _answers list..
mDNS query done
1 service(s) found
1: debianServer (192.168.1.230:1883)

```

bikemike, 2018-05-22:
I had the same issue and fixed it. I'll make a pull request.

thunderace, 2018-05-23:
Nice!! thx

devyte, 2018-12-05:
Closing in view of #5442 with a full rewrite of mdns.

closed 2018-12-05 by devyte