esp8266/Arduino #4649: Nodemcu exception 29

alexlab963, 2018-04-18:
### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [X ] If there is a stack dump, I have decoded it.
- [X ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			2.4.1
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Nodemcu|other
- Flash Mode:			[qio|dio|other]
- Flash Size: 		4MB
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:	SERIAL
- Upload Speed:  921600

### Problem Description

I am running a webserver, it seems like whenever page /set/c/ gets loaded the error happens

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <functional>

#include "switch.h"
#include "UpnpBroadcastResponder.h"
#include "CallbackFunction.h"

// prototypes
boolean connectWifi();

// the IP address for the shield:
IPAddress ip(192, 168, 1, 213); 
IPAddress gateway(192,168,1,1);
IPAddress subnet(255,255,255,0);

// Single devices on/off callbacks
void bookshelfOn();
void bookshelfOff();

// Change this before you flash
const char* ssid = "LAB";
const char* password = "2392063048";

boolean wifiConnected = false;

UpnpBroadcastResponder upnpBroadcastResponder;
Switch *bookshelf = NULL;

int status1;

int rv = 255;
int gv = 255;
[... 325 lines cut ...]
  return 0;
}

```

### Debug Messages

```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 13 results
0x40216480: tcp_setprio at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp.c line 1412
0x40216218: inet_chksum_pseudo at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/inet_chksum.c line 542
0x4020457c: ClientContext at C:\Users\Alex.ALEX_JR\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi\src/WiFiServer.cpp line 168
:  (inlined by) WiFiServer::_accept(tcp_pcb*, long) at C:\Users\Alex.ALEX_JR\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi\src/WiFiServer.cpp line 162
0x40204608: WiFiServer::_s_accept(void*, tcp_pcb*, long) at C:\Users\Alex.ALEX_JR\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi\src/WiFiServer.cpp line 168
0x402114b0: tcp_listen_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_in.c line 599
:  (inlined by) tcp_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_in.c line 322
0x40215949: ip4_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 685
0x4020ef9d: ethernet_input_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c line 182
0x4010453c: esp2glue_ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 433
0x4022c52e: ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 357
0x4022c540: ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 365
0x4022e483: ets_snprintf at ?? line ?

```

alexlab963, 2018-04-18:
Update: [Error 2](https://pastebin.com/KdWFLG5s)

earlephilhower, 2019-01-22:
Closing as it looks to be OOM related, includes many 3rd party libraries, doesn't have a MCVE we can use, and the stack dumps look incomplete (missing the actual PC of the error, only containing the call stack).

closed 2019-01-22 by earlephilhower