esp8266/Arduino #4651: Huge compilatio issue after swich from 2.3.0 to 2.4.1

miky2k, 2018-04-18:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[Generic ESP8266]
- Core Version:			[2.4.1]
- Development Env:		[Slober Eclipse Plugin]
- Operating System:		[Mint]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [all tested]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description
Huge compilation issue on package ESP8266Wifi and ESP8622WebServer may be some 
flag or switch in libraries ?

Fresh Eclipse+Slober install does not solve.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
Sketch work flawless in 2.3.0 version.

### Debug Messages

```
"/home/user/bin/cpp-oxygen/eclipse//arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/user/bin/cpp-oxygen/eclipse//arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/include" "-I/home/user/bin/cpp-oxygen/eclipse//arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/lwip/include" "-I/home/user/bin/cpp-oxygen/eclipse//arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-I/media/user/74a50ba9-aa9c-4031-8341-b77f685e75f6/cvs/progetti/esp8266/esp8266-WebServices/Release/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC  -DARDUINO=10802 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 '-DARDUINO_BOARD="ESP8266_GENERIC"' -DLED_BUILTIN=0 -DF_CRYSTAL=40000000 -DESP8266  -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/libraries/WiFi/1.2.6" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/libraries/WiFi/1.2.6/src" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/Ethernet" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/Ethernet/src" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/variants/generic" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ArduinoOTA" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/EEPROM" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266HTTPClient" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266HTTPClient/src" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266mDNS" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/SPI" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/Ticker" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/Wire" -I"/home/user/Arduino/libraries/configure" -I"/home/user/Arduino/libraries/cron" -I"/home/user/Arduino/libraries/DS1631" -I"/home/user/Arduino/libraries/GOFi2cOLED" -I"/home/user/Arduino/libraries/INA219" -I"/home/user/Arduino/libraries/IniFile" -I"/home/user/Arduino/libraries/picIO" -I"/home/user/Arduino/libraries/Rtc" -I"/home/user/Arduino/libraries/Rtc/src" -I"/home/user/Arduino/libraries/SI7021" -I"/home/user/Arduino/libraries/SMTPClient" -I"/home/user/Arduino/libraries/TSL2561" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/SD" -I"/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/SD/src" -MMD -MP -MF"libraries/ESP8266WebServer/src/ESP8266WebServer.cpp.d" -MT"libraries/ESP8266WebServer/src/ESP8266WebServer.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp"  -o  "libraries/ESP8266WebServer/src/ESP8266WebServer.cpp.o"
In file included from /home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src/ESP8266WiFi.h:42:0,
                 from /home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/ESP8266WebServer.h:29,
                 from /home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp:28:
/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src/WiFiClientSecure.h:85:33: error: expected ')' before '*' token
   WiFiClientSecure(ClientContext* client, bool usePMEM, const uint8_t *rsakey, int rsakeyLen, const uint8_t *cert, int certLen);
                                 ^
/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src/WiFiClientSecure.h:34:3: error: 'WiFiClientSecure::~WiFiClientSecure()' marked override, but does not override
   ~WiFiClientSecure() override;
   ^
/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src/WiFiClientSecure.h:37:7: error: 'int WiFiClientSecure::connect(String, uint16_t)' marked override, but does not override
   int connect(const String host, uint16_t port) override;
       ^
/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src/WiFiClientSecure.h:45:10: error: 'size_t WiFiClientSecure::write_P(const char*, size_t)' marked override, but does not override
   size_t write_P(PGM_P buf, size_t size) override;
          ^
/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src/WiFiClientSecure.h:50:10: error: 'size_t WiFiClientSecure::peekBytes(uint8_t*, size_t)' marked override, but does not override
   size_t peekBytes(uint8_t *buffer, size_t length) override;
          ^
In file included from /home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp:28:0:
/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/ESP8266WebServer.h: In member function 'virtual size_t ESP8266WebServer::_currentClientWrite_P(const char*, size_t)':
/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/ESP8266WebServer.h:140:83: error: 'class WiFiClient' has no member named 'write_P'
   virtual size_t _currentClientWrite_P(PGM_P b, size_t l) { return _currentClient.write_P( b, l ); }
                                                                                   ^
/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp: In constructor 'ESP8266WebServer::ESP8266WebServer(IPAddress, int)':
/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp:59:17: error: no matching function for call to 'WiFiServer::WiFiServer(IPAddress&, [... cut ...]
/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp: In member function 'virtual void ESP8266WebServer::close()':
/home/user/bin/cpp-oxygen/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp:337:11: error: 'class WiFiServer' has no member named 'close'
   _server.close();
           ^
make: *** [libraries/ESP8266WebServer/src/ESP8266WebServer.cpp.o] Errore 1
libraries/ESP8266WebServer/src/subdir.mk:24: set di istruzioni per l'obiettivo "libraries/ESP8266WebServer/src/ESP8266WebServer.cpp.o" non riuscito
```

miky2k, 2018-04-19:
For reference.
https://github.com/Sloeber/arduino-eclipse-plugin/issues/967

jandebleser, 2018-06-14:
I had this error `'class WiFiClient' has no member named 'write_P'`
That was due to having this in my code: 
`#include <WiFi.h>`
Removing it fixed my issues.

devyte, 2018-12-05:
WiFiClientSecure has changed drastically since this was opened due to migration to bearssl. Closing. If the issue is still present after migration to the new WiFiClientSecure, please open a new issue and fill out the requested info.

closed 2018-12-05 by devyte