esp8266/Arduino #4653: Incorporating new boards to the boards.txt

lazyzero, 2018-04-19:
Hello I am writing here to ask how to get a board to the boards.txt?

I ask because I like to have the Kniwwelino (kniwwelino.lu) to be included into the official list of boards.

Best Christian

d-a-v, 2018-04-19:
You need to update the generator in ```tools/boards.txt.py```.

devyte, 2018-04-22:
Is the boars distinct in its hw from all other boards? I.e.: can you select nodemcu and it will work?
If the board is essentially a clone of another board, then it probably shouldn't be added. If it has differences from other boards, e.g. different flash mode, or different pinouts, etc, then you can make a pr with the modifications to boards.txt.py and the new generated boards.txt.
Closing.

closed 2018-04-22 by devyte