esp8266/Arduino #4655: ESP8266httpUpdate issue with AWS S3 URLs

yknivag, 2018-04-19:
#### Platform

- Hardware:				[Wemos D1 Mini]
- Core Version:			[2.4.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Linux]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Size: 		[4MB]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL/OTA]
- Upload Speed:   [921600]

### Problem Description

I'm trying to create a sketch that checks GitHub for a new release and updates accordingly.  I have the GutHub API part working and, by attaching the .bin file to the GitHub release, am able to obtain a URL for the new binary from the GitHub API.

This URL doesn't work with ESP8266httpUpdate as the code doesn't handle redirects, so I have a custom function [found in this Gist]( https://gist.github.com/yknivag/2707e27787abeafa69b33c4e9d043c3a) which resolves the redirecting URL to a valid final destination URL.

The "true" URL is valid only for 5 minutes so I cannot provide an example, but below is the way to get it.

This "true" URL is a time-bound, pre-signed share URL to the correct file in GitHub's AWS S3 bucket.  When feeding this into ESP8266httpUpdate I get the error "400 Bad Request" from AWS.

I can replicate the Error 400 on the command line using: `curl -v <url_here>` but if I use `curl -v "<url_here>"` it works on the CLI.  I have though no idea how to make the necessary change to ESP8266httpUpdate to make it work there.

### Procedure to create URL

`curl -v https://github.com/yknivag/test/releases/download/0.0.0/temp.ino.d1_mini.bin` will provide the time limited URL in the `Location` field.

Or it is possible to use the partial sketch in the gist above feeding the URL `https://github.com/yknivag/test/releases/download/0.0.0/temp.ino.d1_mini.bin` into it.

torntrousers, 2018-04-19:
I don't think ESP8266httpUpdate works with https. Try getting the URL from the Location and just changing the https to be http.

yknivag, 2018-04-20:
Hi, thank you for your reply.

The ESP8266httpUpdate library examples certainly suggest it works with https and it supports https specific function calls also.  I'm not sure why that should be the case if it doesn't support https.

However testing this morning has revealed that the I have exactly the same problem when replacing the `https://` with `http://`.

Incidentally the same behaviour is observed in curl too (it works as `curl -v "http://url/file.bin"` but not as `curl -v http://url/file.bin`) which suggests that it has something to do with how AWS is expecting the URL to be presented and that without the `""` curl is misinterpreting the URL.  The behaviour is replicated also with wget in both `https://` and `http://`.

This leads me to believe that there is a potentially similar fix for the ESP if only I knew where to look.

torntrousers, 2018-04-20:
Yes looking in the code it does look like its supposed to work with https now, it didn't used to.

yknivag, 2018-04-20:
It does make the connection on https - it must as I get back an HTTP header but that aside it doesn't work with http either.

Looking more deeply into the code of `ESP8266httpUpdate` it seems to rely on the `ESP8266HTTPClient`.

Funnily enough if I test using the same http URL in the `BasicHttpClient` example from the `ESP8266HTTPClient` library directly the link works perfectly well.

Given that the link works perfectly well in `ESP8266HTTPClient` but not in `ESP8266httpUpdate` that would lead me even more strongly to believe that this is an issue with `ESP8266httpUpdate`.

devyte, 2019-11-09:
@yknivag what is the status with release 2.6.0?

yknivag, 2019-11-09:
Hi @devyte I haven't upgraded to 2.6.0 yet and am mobile atm. Will test asap and update.

yknivag, 2019-11-12:
All is now resolved.  Many thanks!

It would be really nice if ESP8266HTTPUpdate also handled URL re-directions internally too, but that's a feature request and unrelated to this so I've closed this.

Thank you for resolving.

closed 2019-11-12 by yknivag