esp8266/Arduino #4656: After serial programming, when given static config WiFi.config() still requests DHCP address

andrewradke, 2018-04-20:
### Platform

- Hardware:				ESP-12|ESP-07
- Core Version:			2.4.1
- Development Env:		Arduino IDE 1.8.5
- Operating System:		MacOS

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			dio
- Flash Size: 			1MB
- lwip Variant:		v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:		any

### Problem Description

When programming the ESP8266 over a serial link it will request an IP address via DHCP even if successfully given a static IP address with `WiFi.config`.

It will replace the static IP address with the DHCP supplied address when it receives the DHCP Offer packet. It responds to the static one before between the time of setting the static address and receiving the DHCP Offer. Once the DHCP Offer has been received and the address has been changed to that it is possible to call `WiFi.config` again and change the IP back again. It will continue to periodically request or renew the DHCP address after this but it appears not to make use of it.

This behaviour continues after a reset or reboot but removal of power and restarting the ESP8266 corrects the behaviour.

Below is a packet capture of a request following serial programming and immediately following `WiFi.config(local_ip, gateway, subnet, dns1, dns2);` Between the first and third packet it will respond to the static `local_ip` given.
```
    1 0.000000000      0.0.0.0 → 255.255.255.255 DHCP 350 DHCP Discover - Transaction ID 0x48ad2bd3
    2 0.000513333 192.168.26.30 → 192.168.26.164 ICMP 62 Echo (ping) request  id=0xa8fa, seq=0/0, ttl=64
    3 1.001605162 192.168.26.30 → 192.168.26.164 DHCP 342 DHCP Offer    - Transaction ID 0x48ad2bd3
    4 1.007302656      0.0.0.0 → 255.255.255.255 DHCP 350 DHCP Request  - Transaction ID 0x48ad2bd3
    5 1.007925333 192.168.26.30 → 192.168.26.164 DHCP 342 DHCP ACK      - Transaction ID 0x48ad2bd3
```

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
The following is extracted from the full program here: https://github.com/andrewradke/ESP8266-Homie-sensors

```cpp
#include <ESP8266WiFi.h>

bool use_staticip = true;
IPAddress local_ip(192, 168, 26, 58);
IPAddress dns1(192, 168, 26, 30);
IPAddress dns2(192, 168, 26, 20);
IPAddress subnet(255, 255, 255, 0);
IPAddress gateway(192, 168, 26, 1);
String ssid  = "";
String psk   = "";

void setup() {
  Serial.begin(74880);        // Open serial monitor at SERIALSPEED baud to see results.
  delay(200);                 // Give the serial port time to setup

  WiFi.mode(WIFI_STA);

  ssid = WiFi.SSID();
  psk  = WiFi.psk();

  if (use_staticip) {
    bool configstatus = WiFi.config(local_ip, gateway, subnet, dns1, dns2);
      Serial.print("IP config ");
    if (configstatus) {
      Serial.print("succeeded");
    } else {
      Serial.print("succeeded");
    }
  }
}

void loop() {

}
```

d-a-v, 2018-04-20:
Can you please try this lwip2 update and report back ? [comment](https://github.com/esp8266/Arduino/issues/4596#issuecomment-382177555)

d-a-v, 2018-04-21:
It is indeed. Please try the above fix.

TD-er, 2018-05-15:
In ESPeasy I made a work-around for this, to be able to keep using the (a bit older) PlatformIO version of the core library:

https://github.com/letscontrolit/ESPEasy/blob/648353b21648b00571410296ff18d1ff232f8d9e/src/ESPEasyWiFiEvent.h#L1-L32

closed 2018-08-01 by devyte