esp8266/Arduino #4660: scanf / printf : Enabling 8-bit types

GiovanniCmpaner, 2018-04-20:
Hello everyone!

I need 8-bit types ("hh" length specifiers) when using scanf / printf, but it isn't enabled by default.

I could conclude it by looking at the headers:
```
inttypes.h
stdint.h
stdio.h
```
and noted the macro`_WANT_IO_C99_FORMATS` which must be defined to enable them.

After reading the docs [https://sourceware.org/newlib/README](https://sourceware.org/newlib/README), before I compile the newlib, I must configure it with the flag `--enable-newlib-io-c99-formats`, but I don't know how to proceed, where I should do it, how to compile it, I couldn't easily find information about this.

Is there a "simpler"/"easier" way to enable 8-bit types?

earlephilhower, 2019-08-21:
We don't compile with the full-fat IO types due to code and RAM usage limits, but it may be possible to hack in half-width types at some point

GiovanniCmpaner, 2019-08-21:
To add some context, it would be especially useful when receiving IPv4 address, as I can't simply switch to int because the (ethernet, wifi) libraries expect byte vectors.
Right now, the simplest way is to convert the int vector to byte vector in a few lines, not that bad.
But thanks for answering after such a long time.

closed 2019-12-10 by d-a-v

GiovanniCmpaner, 2019-12-11:
I didn't think it would be done. Thanks for the improvement.

PR #6896 (2019-12-10): Add hh (byte) format for printf and scanf
Fixes #4660

Add in support for byte-wide scanf and printf (%hh..) in printf and
scanf.  Allows reading into a byte array one byte at a time.

Simple test:
````
void setup() {
  Serial.begin(115200);
  Serial.printf("%hhx\n", 123+256);
  char b[4];
  int i=999;
  int r= sscanf("1 2 3 4 5", "%hhd %hhd %hhd %hhd %d", b,b+1,b+2,b+3, &i);
  Serial.printf("r=%d  %d %d %d %d %d\n", r, b[0], b[1], b[2], b[3], i);
}

void loop() {
}
````

Gives expected output
````
7b
r=5  1 2 3 4 5
````
commit d012adc9 (2019-12-10): Add hh (byte) format for printf and scanf (#6896)

Fixes #4660

Add in support for byte-wide scanf and printf (%hh..) in printf and
scanf.  Allows reading into a byte array one byte at a time.